libfive/libfive #7: Crash with while trying to watch a file

amirouche, 2016-08-19:
Under Ubuntu with Guile 2.1.3 and the gcc-5 patch (cf. pull requests) I can't watch for files inside `ao-guile`.

Here is my run

```bash
 amirouche@ubuaout  ~/src/guile/ao/bin   master  GUILE_LOAD_PATH=`pwd` gdb guile 
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.04) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from guile...done.
(gdb) r
Starting program: /usr/bin/guile 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff5e87700 (LWP 3622)]
[New Thread 0x7ffff5686700 (LWP 3623)]
[New Thread 0x7ffff4e85700 (LWP 3624)]
[New Thread 0x7ffff4684700 (LWP 3625)]
[New Thread 0x7ffff3e83700 (LWP 3626)]
[New Thread 0x7ffff3682700 (LWP 3627)]
[New Thread 0x7ffff2e81700 (LWP 3628)]
[New Thread 0x7ffff247d700 (LWP 3629)]
[New Thread 0x7ffff0f5c700 (LWP 3630)]
GNU Guile 2.1.3
Copyright (C) 1995-2016 Free Software Foundation, Inc.

Guile comes with ABSOLUTELY NO WARRANTY; for details type `,show w'.
This program is free software, and you are welcome to redistribute it
under certain conditions; type `,show c' for details.

Enter `,help' for help.
scheme@(guile-user)> (load "ao-guile")
[New Thread 0x7fffec9d4700 (LWP 3631)]

         .8.
        .888.
       :88888.          ,o8888o.
      . `88888.      . 8888   `88.
     .8. `88888.    ,8 8888     `8b
    .8`8. `88888.   88 8888      `8b
   .8' `8. `88888.  88 8888      ,8P
  .8'   `8. `88888. `8 8888     ,8P
 .888888888. `88888. ` 8888   ,88'
.8'       `8. `88888.   `88888P'
       (c) 2015 Matt Keeter

REPL is provided by Ao [1]> (ao-watch "../examples/menger.scm")
Watching /home/amirouche/src/guile/ao/bin/../examples/menger.scm
[New Thread 0x7fffe53b6700 (LWP 3634)]

Thread 11 "guile" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffec9d4700 (LWP 3631)]
0x00007ffff7afac60 in scm_make_dynamic_state (parent=parent@entry=0x0) at fluids.c:430
430	  SCM_ASSERT (IS_DYNAMIC_STATE (parent), parent, SCM_ARG1, FUNC_NAME);
(gdb) bt
```

Here is the full backtrace

```
#0  0x00007ffff7afac60 in scm_make_dynamic_state (parent=parent@entry=0x0) at fluids.c:430
#1  0x00007fffecab0d38 in guilify_self_2 (parent=0x0) at threads.c:617
#2  0x00007fffecab13ec in scm_i_init_thread_for_guile (base=0x7fffec9d34c0, parent=0x0) at threads.c:855
#3  0x00007fffecab1810 in scm_init_guile () at threads.c:867
#4  0x00007fffefe8800c in window_watch_callback_ (s="/home/amirouche/src/guile/ao/bin/../examples/menger.scm")
    at ../bind/api.cpp:273
#5  0x00007fffefe880fb in std::_Function_handler<void (std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >), void (*)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)>::_M_invoke(std::_Any_data const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&) (
    __functor=..., __args#0=<optimized out>) at /usr/include/c++/5/functional:1871
#6  0x00007fffefec8a45 in std::function<void (std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)>::operator()(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) const (
    __args#0="", this=0x7fffec9d3620) at /usr/include/c++/5/functional:2267
#7  ScriptWatcher::ScriptWatcher(Window*, std::function<void (std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) (this=0x7fffe8008f40, window=<optimized out>, 
    callback=..., directory="", filename=...) at ../ui/src/watcher.cpp:40
#8  0x00007fffefe86e4e in window_watch_file (dir=<optimized out>, file=<optimized out>) at ../bind/api.cpp:288
#9  0x00007ffff704fe40 in ffi_call_unix64 () from /usr/lib/x86_64-linux-gnu/libffi.so.6
#10 0x00007ffff704f8ab in ffi_call () from /usr/lib/x86_64-linux-gnu/libffi.so.6
#11 0x00007ffff7afd2e7 in scm_i_foreign_call (foreign=<optimized out>, argv=0x7fffefe68ce8) at foreign.c:1030
#12 0x00007ffff7b65db8 in vm_debug_engine (thread=0x0, vp=0x980c80, registers=0xba2e90, resume=-325237128)
    at vm-engine.c:847
#13 0x00007ffff7b70f88 in scm_call_n (proc=proc@entry=0x7ffff01395c0, argv=argv@entry=0x0, nargs=nargs@entry=0)
    at vm.c:1238
#14 0x00007ffff7af0859 in scm_call_0 (proc=proc@entry=0x7ffff01395c0) at eval.c:473
#15 0x00007ffff7b5e300 in really_launch (d=0x7fffffffdba0) at threads.c:885
#16 0x00007ffff7aeb83a in c_body (d=0x7fffec9d3e60) at continuations.c:429
#17 0x00007ffff7b65ce9 in vm_debug_engine (thread=0x0, vp=0x980c80, registers=0xba2e90, resume=-325237128)
    at vm-engine.c:815
#18 0x00007ffff7b70f88 in scm_call_n (proc=proc@entry=0x8063c0, argv=argv@entry=0x0, nargs=nargs@entry=0)
    at vm.c:1238
#19 0x00007ffff7af0859 in scm_call_0 (proc=proc@entry=0x8063c0) at eval.c:473
#20 0x00007ffff7b5fb8a in catch (tag=tag@entry=0x404, thunk=0x8063c0, handler=0x8063a0, pre_unwind_handler=0x829fe0)
    at throw.c:135
#21 0x00007ffff7b5ffe5 in scm_catch_with_pre_unwind_handler (key=key@entry=0x404, thunk=<optimized out>, 
    handler=<optimized out>, pre_unwind_handler=<optimized out>) at throw.c:249
#22 0x00007ffff7b6009f in scm_c_catch (tag=tag@entry=0x404, body=body@entry=0x7ffff7aeb830 <c_body>, 
    body_data=body_data@entry=0x7fffec9d3e60, handler=handler@entry=0x7ffff7aebb00 <c_handler>, 
    handler_data=handler_data@entry=0x7fffec9d3e60, 
    pre_unwind_handler=pre_unwind_handler@entry=0x7ffff7aeb960 <pre_unwind_handler>, 
    pre_unwind_handler_data=0x6ecbe0) at throw.c:370
#23 0x00007ffff7aebe60 in scm_i_with_continuation_barrier (body=body@entry=0x7ffff7aeb830 <c_body>, 
    body_data=body_data@entry=0x7fffec9d3e60, handler=handler@entry=0x7ffff7aebb00 <c_handler>, 
    handler_data=handler_data@entry=0x7fffec9d3e60, 
    pre_unwind_handler=pre_unwind_handler@entry=0x7ffff7aeb960 <pre_unwind_handler>, 
    pre_unwind_handler_data=0x6ecbe0) at continuations.c:367
#24 0x00007ffff7aebf45 in scm_c_with_continuation_barrier (func=<optimized out>, data=<optimized out>)
    at continuations.c:463
#25 0x00007ffff7b5da3c in with_guile_and_parent (base=0x7fffec9d3ec0, data=0x7fffec9d3ef0) at threads.c:786
#26 0x00007ffff726d952 in GC_call_with_stack_base () from /usr/lib/x86_64-linux-gnu/libgc.so.1
#27 0x00007ffff7b5d4ac in scm_i_with_guile_and_parent (parent=<optimized out>, data=0x7fffffffdba0, 
    func=0x7ffff7b5e280 <really_launch>) at threads.c:829
#28 launch_thread (d=0x7fffffffdba0) at threads.c:897
#29 0x00007ffff78936fa in start_thread (arg=0x7fffec9d4700) at pthread_create.c:333
#30 0x00007ffff75c9b5d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
```

If I paste the content of `charm.scm` or `menger.scm` inside ao-guile, it works.

sethalves, 2016-08-21:
I see this on my machine, as well (Ubuntu 16.04.1 with guile 2.0.11 from ubuntu's repo).

    -seth

mkeeter, 2016-11-30:
Can you see if PR #23 fixes this?

sethalves, 2016-11-30:
I still get a segfault...
```
Thread 10 "guile" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffee14c700 (LWP 4112)]
0x00007fffed6f0e49 in glClearColor () from /usr/lib/nvidia-367/libGL.so.1
(gdb) bt
#0  0x00007fffed6f0e49 in glClearColor () from /usr/lib/nvidia-367/libGL.so.1
#1  0x00007ffff196b3c6 in Window::draw (this=this@entry=0x6157f0) at ../ui/src/window.cpp:231
#2  0x00007ffff191fc81 in ao_run () at ../bind/libao.cpp:310
#3  0x00007ffff6ff5e40 in ffi_call_unix64 () from /usr/lib/x86_64-linux-gnu/libffi.so.6
#4  0x00007ffff6ff58ab in ffi_call () from /usr/lib/x86_64-linux-gnu/libffi.so.6
#5  0x00007ffff7ab67fd in ?? () from /usr/lib/x86_64-linux-gnu/libguile-2.0.so.22
#6  0x00007ffff7b2ed71 in ?? () from /usr/lib/x86_64-linux-gnu/libguile-2.0.so.22
#7  0x00007ffff7b3401e in ?? () from /usr/lib/x86_64-linux-gnu/libguile-2.0.so.22
#8  0x00007ffff7b224a0 in ?? () from /usr/lib/x86_64-linux-gnu/libguile-2.0.so.22
#9  0x00007ffff7aa158a in ?? () from /usr/lib/x86_64-linux-gnu/libguile-2.0.so.22
#10 0x00007ffff7b34093 in ?? () from /usr/lib/x86_64-linux-gnu/libguile-2.0.so.22
#11 0x00007ffff7aab283 in scm_call_4 () from /usr/lib/x86_64-linux-gnu/libguile-2.0.so.22
#12 0x00007ffff7aa1d21 in ?? () from /usr/lib/x86_64-linux-gnu/libguile-2.0.so.22
#13 0x00007ffff7aa1e05 in scm_c_with_continuation_barrier () from /usr/lib/x86_64-linux-gnu/libguile-2.0.so.22
#14 0x00007ffff7b21bbc in ?? () from /usr/lib/x86_64-linux-gnu/libguile-2.0.so.22
#15 0x00007ffff7213952 in GC_call_with_stack_base () from /usr/lib/x86_64-linux-gnu/libgc.so.1
#16 0x00007ffff7b2154c in ?? () from /usr/lib/x86_64-linux-gnu/libguile-2.0.so.22
#17 0x00007ffff7218a31 in GC_inner_start_routine () from /usr/lib/x86_64-linux-gnu/libgc.so.1
#18 0x00007ffff7213952 in GC_call_with_stack_base () from /usr/lib/x86_64-linux-gnu/libgc.so.1
#19 0x00007ffff78396ba in start_thread (arg=0x7fffee14c700) at pthread_create.c:333
#20 0x00007ffff756f82d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
```

mkeeter, 2016-11-30:
Interesting.  This is unrelated to the FFI bug that I just fixed, since it's in the graphics stack.

Can you try running the test suite?

It should be in `build/test/ao-test`, and will (among other things) create an OpenGL context and verify that it's happy.

sethalves, 2016-11-30:
```seth@seth:~/src/ao$ ./build/test/ao-test

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
ao-test is a Catch v1.1 b3 host application.
Run with -? for options

-------------------------------------------------------------------------------
Evaluator::derivs
  X + Z
-------------------------------------------------------------------------------
../kernel/test/eval.cpp:235
...............................................................................

../kernel/test/eval.cpp:261: FAILED:
due to a fatal error condition:
  SIGSEGV - Segmentation violation signal

===============================================================================
test cases: 12 | 11 passed | 1 failed
assertions: 47 | 46 passed | 1 failed
```

mkeeter, 2016-11-30:
Hah, and that's entirely unrelated to OpenGL.  Sounds like there are a few interacting issues here.

Can you do a clean `git pull` + rebuild, then get a backtrace on that crash?
(i.e. run in `gdb` / `lldb` and call `bt` when it segfaults)

sethalves, 2016-11-30:
okay, here's a clean checkout with build and test:
```
seth@seth:~/src/ao$ mkdir build
seth@seth:~/src/ao$ cd build
seth@seth:~/src/ao/build$ cmake -G Ninja ..
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.8") 
-- Found PNG: /usr/lib/x86_64-linux-gnu/libpng.so (found version "1.2.54") 
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for one of the modules 'glfw3'
-- Configuring done
-- Generating done
-- Build files have been written to: /home/seth/src/ao/build
seth@seth:~/src/ao/build$ ninja
[3/67] Building CXX object kernel/CMakeFiles/ao-kernel.dir/src/tree/opcode.cpp.o
../kernel/src/tree/opcode.cpp: In function ‘size_t Opcode::args(Opcode::Opcode)’:
../kernel/src/tree/opcode.cpp:66:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
[67/67] Linking CXX executable test/ao-test
seth@seth:~/src/ao/build$ cd ..
seth@seth:~/src/ao$ ./build/test/ao-test 

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
ao-test is a Catch v1.1 b3 host application.
Run with -? for options

-------------------------------------------------------------------------------
Evaluator::derivs
  X + Z
-------------------------------------------------------------------------------
../kernel/test/eval.cpp:235
...............................................................................

../kernel/test/eval.cpp:261: FAILED:
due to a fatal error condition:
  SIGSEGV - Segmentation violation signal

===============================================================================
test cases: 12 | 11 passed | 1 failed
assertions: 47 | 46 passed | 1 failed

```

and here's that crash:
```
seth@seth:~/src/ao$ gdb ./build/test/ao-test 
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.04) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./build/test/ao-test...done.
(gdb) r
Starting program: /home/seth/src/ao/build/test/ao-test 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
eval_clause_derivs (count=1, odz=0x6f7080, ody=0x6f5c70, odx=<optimized out>, ov=0x70ca60, bdz=0x6f7480, bdy=0x6f6070, bdx=0x6f4c60, bv=0x70ce60, adz=<optimized out>, ady=<optimized out>, 
    adx=0x6f5060, av=0x70d260, op=Opcode::ADD) at ../kernel/src/eval/evaluator.cpp:960
960	                ody[i] = _mm256_add_ps(ady[i], bdy[i]);
(gdb) bt
#0  eval_clause_derivs (count=1, odz=0x6f7080, ody=0x6f5c70, odx=<optimized out>, ov=0x70ca60, bdz=0x6f7480, bdy=0x6f6070, bdx=0x6f4c60, bv=0x70ce60, adz=<optimized out>, ady=<optimized out>, 
    adx=0x6f5060, av=0x70d260, op=Opcode::ADD) at ../kernel/src/eval/evaluator.cpp:960
#1  Evaluator::derivs (this=this@entry=0x7fffffffc4e0, count=count@entry=2, vectorize=vectorize@entry=true) at ../kernel/src/eval/evaluator.cpp:1244
#2  0x000000000042c771 in ____C_A_T_C_H____T_E_S_T____235 () at ../kernel/test/eval.cpp:267
#3  0x00000000004b5b11 in Catch::RunContext::invokeActiveTestCase (this=0x7fffffffd120) at ../vendor/catch/catch.hpp:5473
#4  Catch::RunContext::runCurrentTest (redirectedCerr="", redirectedCout="", this=0x7fffffffd120) at ../vendor/catch/catch.hpp:5445
#5  Catch::RunContext::runTest (testCase=..., this=0x7fffffffd120) at ../vendor/catch/catch.hpp:5284
#6  Catch::Runner::runTests (this=this@entry=0x7fffffffd850) at ../vendor/catch/catch.hpp:5603
#7  0x00000000004b63f6 in Catch::Session::run (this=this@entry=0x7fffffffdbc0) at ../vendor/catch/catch.hpp:5734
#8  0x0000000000427809 in Catch::Session::run (argv=0x7fffffffde58, argc=<optimized out>, this=0x7fffffffdbc0) at ../vendor/catch/catch.hpp:5714
#9  main (argc=<optimized out>, argv=0x7fffffffde58) at ../test/main.cpp:6
(gdb) 
```

mkeeter, 2016-12-01:
I wonder if it's an AVX issue?

`evaluator.cpp` is using the AVX codepaths; can you check in `/proc/cpuinfo` to make sure those instructions are valid?

More things to check:
- Change `RELEASE` to `DEBUG` [here](https://github.com/mkeeter/ao/blob/master/CMakeLists.txt#L3) (and recompile) to get a more useful backtrace
- Remove `-march=native` [here](https://github.com/mkeeter/ao/blob/master/CMakeLists.txt#L7) (and recompile) to disable AVX entirely

sethalves, 2016-12-01:
Here's gdb of ao-test with a debug build:
```
Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7a7e1ae in eval_clause_derivs (op=Opcode::ADD, av=0x81d260, adx=0x805060, ady=0x806470, adz=0x807880, bv=0x81ce60, bdx=0x804c60, bdy=0x806070, bdz=0x807480, ov=0x81ca60, odx=0x804860, 
    ody=0x805c70, odz=0x807080, count=1) at ../kernel/src/eval/evaluator.cpp:960
960	                ody[i] = _mm256_add_ps(ady[i], bdy[i]);
(gdb) bt
#0  0x00007ffff7a7e1ae in eval_clause_derivs (op=Opcode::ADD, av=0x81d260, adx=0x805060, ady=0x806470, adz=0x807880, bv=0x81ce60, bdx=0x804c60, bdy=0x806070, bdz=0x807480, ov=0x81ca60, 
    odx=0x804860, ody=0x805c70, odz=0x807080, count=1) at ../kernel/src/eval/evaluator.cpp:960
#1  0x00007ffff7a80264 in Evaluator::derivs (this=0x7fffffffc660, count=2, vectorize=true) at ../kernel/src/eval/evaluator.cpp:1244
#2  0x00000000004ff422 in ____C_A_T_C_H____T_E_S_T____235 () at ../kernel/test/eval.cpp:267
#3  0x00000000005642c8 in Catch::FreeFunctionTestCase::invoke (this=0x7f89c0) at ../vendor/catch/catch.hpp:5874
#4  0x000000000054f697 in Catch::TestCase::invoke (this=0x821448) at ../vendor/catch/catch.hpp:6779
#5  0x00000000005620c0 in Catch::RunContext::invokeActiveTestCase (this=0x7fffffffd6e0) at ../vendor/catch/catch.hpp:5473
#6  0x0000000000561d58 in Catch::RunContext::runCurrentTest (this=0x7fffffffd6e0, redirectedCout="", redirectedCerr="") at ../vendor/catch/catch.hpp:5445
#7  0x0000000000560696 in Catch::RunContext::runTest (this=0x7fffffffd6e0, testCase=...) at ../vendor/catch/catch.hpp:5284
#8  0x00000000005627f6 in Catch::Runner::runTests (this=0x7fffffffd9d0) at ../vendor/catch/catch.hpp:5603
#9  0x0000000000563965 in Catch::Session::run (this=0x7fffffffdc90) at ../vendor/catch/catch.hpp:5734
#10 0x00000000005637fc in Catch::Session::run (this=0x7fffffffdc90, argc=1, argv=0x7fffffffded8) at ../vendor/catch/catch.hpp:5714
#11 0x0000000000554d50 in main (argc=1, argv=0x7fffffffded8) at ../test/main.cpp:6
```

i is 0

Here's a run of ao-test with -march=native removed.  It doesn't crash, but there are more failures:
```
seth@seth:~/src/ao$ ./build/test/ao-test

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
ao-test is a Catch v1.1 b3 host application.
Run with -? for options

-------------------------------------------------------------------------------
Vertex positioning
-------------------------------------------------------------------------------
../kernel/test/octree.cpp:69
...............................................................................

../kernel/test/octree.cpp:106: FAILED:
  REQUIRE( rmax < radius*1.1 )
with expansion:
  7.34172f < 0.55

-------------------------------------------------------------------------------
Performance
-------------------------------------------------------------------------------
../kernel/test/heightmap.cpp:271
...............................................................................

../kernel/test/heightmap.cpp:313: 
warning:
  Rendered sphere in 1.213630 sec
  Rendered sponge in 2.598057 sec

-------------------------------------------------------------------------------
Mesh performance
-------------------------------------------------------------------------------
../kernel/test/mesh.cpp:59
...............................................................................

../kernel/test/mesh.cpp:80: 
warning:
  
  Made sponge mesh in 18.154638 sec

-------------------------------------------------------------------------------
Face normals
  Positive
-------------------------------------------------------------------------------
../kernel/test/dc.cpp:42
...............................................................................

../kernel/test/dc.cpp:55: FAILED:
  REQUIRE( m.norm(j) == norm[i] )
with expansion:
  vec3(-nan, -nan, -nan)
  ==
  vec3(1.000000, 0.000000, 0.000000)

-------------------------------------------------------------------------------
Face normals
  Negative
-------------------------------------------------------------------------------
../kernel/test/dc.cpp:42
...............................................................................

../kernel/test/dc.cpp:71: FAILED:
  REQUIRE( m.norm(j) == -norm[i] )
with expansion:
  vec3(-nan, -nan, -nan)
  ==
  vec3(-1.000000, -0.000000, -0.000000)

-------------------------------------------------------------------------------
Simple 2D contouring
-------------------------------------------------------------------------------
../kernel/test/dc.cpp:77
...............................................................................

../kernel/test/dc.cpp:84: FAILED:
  REQUIRE( m.contours.size() == 1 )
with expansion:
  2 == 1

-------------------------------------------------------------------------------
2D contour tracking
-------------------------------------------------------------------------------
../kernel/test/dc.cpp:87
...............................................................................

../kernel/test/dc.cpp:94: FAILED:
  REQUIRE( m.contours.size() == 1 )
with expansion:
  5 == 1

-------------------------------------------------------------------------------
struct contours
-------------------------------------------------------------------------------
../bind/test/contours.cpp:25
...............................................................................

../bind/test/contours.cpp:33: FAILED:
  REQUIRE( cs->size == 1 )
with expansion:
  4 == 1

===============================================================================
test cases:  63 |  58 passed | 5 failed
assertions: 330 | 324 passed | 6 failed

seth@seth:~/src/ao$ 
```

sethalves, 2016-12-01:
here is /proc/cpu
```
seth@seth:~/src/ao$ cat /proc/cpuinfo
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 70
model name	: Intel(R) Core(TM) i7-4870HQ CPU @ 2.50GHz
stepping	: 1
microcode	: 0x12
cpu MHz		: 2507.128
cache size	: 6144 KB
physical id	: 0
siblings	: 8
core id		: 0
cpu cores	: 4
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm epb tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid xsaveopt dtherm ida arat pln pts
bugs		:
bogomips	: 4988.61
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:

processor	: 1
vendor_id	: GenuineIntel
cpu family	: 6
model		: 70
model name	: Intel(R) Core(TM) i7-4870HQ CPU @ 2.50GHz
stepping	: 1
microcode	: 0x12
cpu MHz		: 2500.000
cache size	: 6144 KB
physical id	: 0
siblings	: 8
core id		: 1
cpu cores	: 4
apicid		: 2
initial apicid	: 2
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm epb tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid xsaveopt dtherm ida arat pln pts
bugs		:
bogomips	: 4988.61
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:

processor	: 2
vendor_id	: GenuineIntel
cpu family	: 6
model		: 70
model name	: Intel(R) Core(TM) i7-4870HQ CPU @ 2.50GHz
stepping	: 1
microcode	: 0x12
cpu MHz		: 2324.511
cache size	: 6144 KB
physical id	: 0
siblings	: 8
core id		: 2
cpu cores	: 4
[... 124 lines cut ...]
cpu family	: 6
model		: 70
model name	: Intel(R) Core(TM) i7-4870HQ CPU @ 2.50GHz
stepping	: 1
microcode	: 0x12
cpu MHz		: 2669.335
cache size	: 6144 KB
physical id	: 0
siblings	: 8
core id		: 3
cpu cores	: 4
apicid		: 7
initial apicid	: 7
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm epb tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid xsaveopt dtherm ida arat pln pts
bugs		:
bogomips	: 4988.61
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:
```

sethalves, 2016-12-01:
Well... this is terrible, but if I enable asan like this:
```
seth@seth:~/src/ao$ git diff CMakeLists.txt
diff --git a/CMakeLists.txt b/CMakeLists.txt
index a7d4da4..d8cf51a 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -1,12 +1,13 @@
 cmake_minimum_required(VERSION 2.6)
 project(ao)
-set(CMAKE_BUILD_TYPE RELEASE)
+set(CMAKE_BUILD_TYPE DEBUG)
 
 ################################################################################
 
-set(CMAKE_CXX_FLAGS "-Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native")
+set(CMAKE_CXX_FLAGS "-Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -fsanitize=address")
 set(CMAKE_CXX_FLAGS_RELEASE  "-O3 -DRELEASE")
 set(CMAKE_CXX_FLAGS_DEBUG    "-O0")
+SET(CMAKE_EXE_LINKER_FLAGS "-lasan ${CMAKE_EXE_LINKER_FLAGS} -fsanitize=address")
 
 include_directories(SYSTEM vendor)
```
then ao-test says `All tests passed (346 assertions in 66 test cases)`

sethalves, 2016-12-01:
Okay, here's a different backtrace.  I got this by running:
```
cd ~/src/ao/bin
GUILE_LOAD_PATH=`pwd` gdb guile
handle SIGXCPU nostop
handle SIGPWR nostop
handle SIGTTOU nostop
r
(load "ao-guile")
(ao-watch "../examples/menger.scm")
```

```
Thread 13 "guile" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffe5ac5700 (LWP 12655)]
0x00007ffff194a0bd in Evaluator::applyTransform (this=this@entry=0x1b618a0, count=count@entry=200) at ../kernel/src/eval/evaluator.cpp:1350
1350	        __m256 y = result.mf[Y][i];
(gdb) bt
#0  0x00007ffff194a0bd in Evaluator::applyTransform (this=this@entry=0x1b618a0, count=count@entry=200) at ../kernel/src/eval/evaluator.cpp:1350
#1  0x00007ffff19329cb in Heightmap::pixels (e=e@entry=0x1b618a0, r=..., depth=..., norm=...) at ../kernel/src/render/heightmap.cpp:131
#2  0x00007ffff1933251 in Heightmap::recurse (e=e@entry=0x1b618a0, r=..., depth=..., norm=..., abort=...) at ../kernel/src/render/heightmap.cpp:226
#3  0x00007ffff193321d in Heightmap::recurse (e=e@entry=0x1b618a0, r=..., depth=..., norm=..., abort=...) at ../kernel/src/render/heightmap.cpp:251
#4  0x00007ffff193321d in Heightmap::recurse (e=e@entry=0x1b618a0, r=..., depth=..., norm=..., abort=...) at ../kernel/src/render/heightmap.cpp:251
#5  0x00007ffff193321d in Heightmap::recurse (e=e@entry=0x1b618a0, r=..., depth=..., norm=..., abort=...) at ../kernel/src/render/heightmap.cpp:251
#6  0x00007ffff193321d in Heightmap::recurse (e=e@entry=0x1b618a0, r=..., depth=..., norm=..., abort=...) at ../kernel/src/render/heightmap.cpp:251
#7  0x00007ffff193321d in Heightmap::recurse (e=e@entry=0x1b618a0, r=..., depth=..., norm=..., abort=...) at ../kernel/src/render/heightmap.cpp:251
#8  0x00007ffff193321d in Heightmap::recurse (e=e@entry=0x1b618a0, r=..., depth=..., norm=..., abort=...) at ../kernel/src/render/heightmap.cpp:251
#9  0x00007ffff193321d in Heightmap::recurse (e=e@entry=0x1b618a0, r=..., depth=..., norm=..., abort=...) at ../kernel/src/render/heightmap.cpp:251
#10 0x00007ffff193321d in Heightmap::recurse (e=0x1b618a0, r=..., depth=..., norm=..., abort=...) at ../kernel/src/render/heightmap.cpp:251
#11 0x00007ffff193331f in Heightmap::<lambda()>::operator() (__closure=<optimized out>) at ../kernel/src/render/heightmap.cpp:308
#12 std::_Bind_simple<Heightmap::Render(const std::vector<Evaluator*>&, Region, const atomic_bool&, glm::mat4)::<lambda()>()>::_M_invoke<> (this=<optimized out>)
    at /usr/include/c++/5/functional:1531
#13 std::_Bind_simple<Heightmap::Render(const std::vector<Evaluator*>&, Region, const atomic_bool&, glm::mat4)::<lambda()>()>::operator() (this=<optimized out>) at /usr/include/c++/5/functional:1520
#14 std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::_Bind_simple<Heightmap::Render(const std::vector<Evaluator*>&, Region, const atomic_bool&, glm::mat4)::<lambda()>()>, void>::operator() (this=0x7fffe5ac4ea0) at /usr/include/c++/5/future:1342
#15 std::_Function_handler<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter>(), std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::_Bind_simple<Heightmap::Render(const std::vector<Evaluator*>&, Region, const atomic_bool&, glm::mat4)::<lambda()>()>, void> >::_M_invoke(const std::_Any_data &) (__functor=...) at /usr/include/c++/5/functional:1857
#16 0x00007ffff1935c89 in std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>::operator()() const (this=<optimized out>)
    at /usr/include/c++/5/functional:2267
#17 std::__future_base::_State_baseV2::_M_do_set(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*) (this=0x7fffe000a660, 
    __f=<optimized out>, __did_set=0x7fffe5ac4e5f) at /usr/include/c++/5/future:527
#18 0x00007ffff7840a99 in __pthread_once_slow (once_control=0x7fffe000a678, init_routine=0x7ffff089bac0 <__once_proxy>) at pthread_once.c:116
#19 0x00007ffff193349d in __gthread_once (__func=<optimized out>, __once=0x7fffe000a678) at /usr/include/x86_64-linux-gnu/c++/5/bits/gthr-default.h:699
#20 std::call_once<void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*>(std::once_flag&, void (std::__future_base::_State_baseV2::*&&)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*&&, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*&&, bool*&&) (__f=<optimized out>, __once=...) at /usr/include/c++/5/mutex:738
#21 std::__future_base::_State_baseV2::_M_set_result(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>, bool) (__ignore_failure=false, 
    __res=..., this=0x7fffe000a660) at /usr/include/c++/5/future:387
#22 std::__future_base::_Async_state_impl<std::_Bind_simple<Heightmap::Render(const std::vector<Evaluator*>&, Region, const atomic_bool&, glm::mat4)::<lambda()>()>, void>::<lambda()>::operator() (
[... cut ...]
#25 std::thread::_Impl<std::_Bind_simple<std::__future_base::_Async_state_impl<_BoundFn, <template-parameter-1-2> >::_Async_state_impl(_BoundFn&&) [with _BoundFn = std::_Bind_simple<Heightmap::Render(const std::vector<Evaluator*>&, Region, const atomic_bool&, glm::mat4)::<lambda()>()>; _Res = void]::<lambda()>()> >::_M_run(void) (this=0x7fffe000a750) at /usr/include/c++/5/thread:115
#26 0x00007ffff089cc80 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#27 0x00007ffff78396ba in start_thread (arg=0x7fffe5ac5700) at pthread_create.c:333
#28 0x00007ffff756f82d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
```

mkeeter, 2016-12-02:
In the interest of fixing one bug at a time, try out the latest commit in `master` (43720740329ed6d0271ed3d27dc3a1c655e12d96) and let me know how it goes.

I'm hopeful that the weirdness you're seeing has the same root cause (addressed in PR #25 to fix issue #24).

sethalves, 2016-12-02:
Yes, all the crashes I was experiencing here seem to be fixed.

mkeeter, 2016-12-02:
Great, I'll close this ticket now (as it's strayed a bit from the original issue).

@amirouche, feel free to open a new issue if you try again and still have problems.

closed 2016-12-02 by mkeeter