libfive/libfive #11: Build crashes while linking libao

poppingtonic, 2016-08-20:
While rebuilding after fixing #9, I missed `GL/glcorearb.h`:

> /usr/local/include/GLFW/glfw3.h:131:28: fatal error: GL/glcorearb.h: No such file or directory
>    #include <GL/glcorearb.h>
>                             ^
> compilation terminated.
> [49/71] Building CXX object ui/CMakeFiles/ao-ui.dir/src/task.cpp.o
> FAILED: ui/CMakeFiles/ao-ui.dir/src/task.cpp.o 
> /usr/bin/c++   -DGLFW_INCLUDE_GLCOREARB -Wall -Wextra -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -isystem ../vendor -isystem /usr/local/include -isystem /usr/inclu
> de/libdrm -I../ui/include -I../ui/vendor -I../kernel/include -MMD -MT ui/CMakeFiles/ao-ui.dir/src/task.cpp.o -MF ui/CMakeFiles/ao-ui.dir/src/task.cpp.o.d -o ui/CMakeFiles/ao-ui.dir/src/task.cpp
> .o -c ../ui/src/task.cpp
> In file included from ../ui/include/ao/ui/gl/core.hpp:28:0,
>                  from ../ui/include/ao/ui/worker.hpp:24,
>                  from ../ui/src/task.cpp:20:
> /usr/local/include/GLFW/glfw3.h:131:28: fatal error: GL/glcorearb.h: No such file or directory
>    #include <GL/glcorearb.h>
>                             ^
> compilation terminated.
> 

So I downloaded it from [here](https://www.opengl.org/registry/api/GL/glcorearb.h) and placed it locally as `/usr/include/GL/glcorearb.h`.  Rebuilding after that brings this:

> 28/28] Linking CXX executable test/ao-test
> FAILED: test/ao-test 
> : && /usr/bin/c++   -Wall -Wextra -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE   test/CMakeFiles/ao-test.dir/__/ui/test/gl.cpp.o test/CMakeFiles/ao-test.dir/__/kernel/test/heightmap.cpp.o test/CMakeFiles/ao-test.dir/__/kernel/test/region.cpp.o test/CMakeFiles/ao-test.dir/__/kernel/test/shapes.cpp.o test/CMakeFiles/ao-test.dir/__/kernel/test/dc.cpp.o test/CMakeFiles/ao-test.dir/__/kernel/test/octree.cpp.o test/CMakeFiles/ao-test.dir/__/kernel/test/eval.cpp.o test/CMakeFiles/ao-test.dir/__/kernel/test/mesh.cpp.o test/CMakeFiles/ao-test.dir/__/kernel/test/jit.cpp.o test/CMakeFiles/ao-test.dir/__/kernel/test/matrix.cpp.o test/CMakeFiles/ao-tes
> t.dir/__/kernel/test/affine.cpp.o test/CMakeFiles/ao-test.dir/__/kernel/test/subregion.cpp.o test/CMakeFiles/ao-test.dir/__/kernel/test/avx.cpp.o test/CMakeFiles/ao-test.dir/main.cpp.o test/CMakeFiles/ao-test.dir/api.cpp.o  -o test/ao-test  -rdynamic bind/libao.so ui/libao-ui.a kernel/libao-kernel.a -lpng -lz -lglfw3 -lepoxy ui/vendor/efsw/libefsw.so -lpthread -Wl,-rpath,/home/brian/git/ao/build/bind:/home/brian/git/ao/build/ui/vendor/efsw && :
> bind/libao.so: undefined reference to `XConvertSelection'
> bind/libao.so: undefined reference to `XUnsetICFocus'
> bind/libao.so: undefined reference to `XSetScreenSaver'
> bind/libao.so: undefined reference to `XRRSetCrtcGamma'
> bind/libao.so: undefined reference to `XRRGetOutputInfo'
> bind/libao.so: undefined reference to `XrmUniqueQuark'
> bind/libao.so: undefined reference to `XAllocSizeHints'
> bind/libao.so: undefined reference to `XFreeColormap'
> bind/libao.so: undefined reference to `XcursorImageLoadCursor'
> bind/libao.so: undefined reference to `XRRGetScreenResourcesCurrent'
> bind/libao.so: undefined reference to `XRRAllocGamma'
> bind/libao.so: undefined reference to `XRRQueryVersion'
> bind/libao.so: undefined reference to `XSendEvent'
> bind/libao.so: undefined reference to `XGetWindowProperty'
> bind/libao.so: undefined reference to `XGetIMValues'
> bind/libao.so: undefined reference to `XGetScreenSaver'
> bind/libao.so: undefined reference to `XSetErrorHandler'
> bind/libao.so: undefined reference to `Xutf8SetWMProperties'
> bind/libao.so: undefined reference to `XQueryExtension'
> bind/libao.so: undefined reference to `XUndefineCursor'
> bind/libao.so: undefined reference to `XFreeEventData'
> bind/libao.so: undefined reference to `XGrabPointer'
> bind/libao.so: undefined reference to `XRRFreeOutputInfo'
> bind/libao.so: undefined reference to `XRRFreeScreenResources'
> bind/libao.so: undefined reference to `XFindContext'
> bind/libao.so: undefined reference to `XPeekEvent'
> bind/libao.so: undefined reference to `XkbKeycodeToKeysym'
> bind/libao.so: undefined reference to `XSetWMHints'
> bind/libao.so: undefined reference to `XSetWMNormalHints'
> bind/libao.so: undefined reference to `XFilterEvent'
> bind/libao.so: undefined reference to `XIconifyWindow'
> bind/libao.so: undefined reference to `XRRGetOutputPrimary'
> bind/libao.so: undefined reference to `XSetWMProtocols'
> bind/libao.so: undefined reference to `XCreateFontCursor'
> bind/libao.so: undefined reference to `XRRFreeGamma'
> bind/libao.so: undefined reference to `XSetClassHint'
> bind/libao.so: undefined reference to `XineramaQueryScreens'
> bind/libao.so: undefined reference to `XCreateColormap'
> bind/libao.so: undefined reference to `Xutf8LookupString'
> bind/libao.so: undefined reference to `XF86VidModeSetGammaRamp'
> bind/libao.so: undefined reference to `XInternAtom'
> bind/libao.so: undefined reference to `XSetInputFocus'
> bind/libao.so: undefined reference to `XRRQueryExtension'
> bind/libao.so: undefined reference to `XAllocWMHints'
[... 48 lines cut ...]
> bind/libao.so: undefined reference to `XFlush'
> bind/libao.so: undefined reference to `XCreateWindow'
> bind/libao.so: undefined reference to `XDestroyWindow'
> bind/libao.so: undefined reference to `XChangeWindowAttributes'
> bind/libao.so: undefined reference to `XTranslateCoordinates'
> bind/libao.so: undefined reference to `XkbGetNames'
> bind/libao.so: undefined reference to `XDeleteProperty'
> bind/libao.so: undefined reference to `XSetICFocus'
> bind/libao.so: undefined reference to `XRRSelectInput'
> bind/libao.so: undefined reference to `XOpenIM'
> bind/libao.so: undefined reference to `XGetInputFocus'
> bind/libao.so: undefined reference to `XCheckTypedEvent'
> bind/libao.so: undefined reference to `XDefineCursor'
> bind/libao.so: undefined reference to `XGetWindowAttributes'
> bind/libao.so: undefined reference to `XkbSetDetectableAutoRepeat'
> bind/libao.so: undefined reference to `XGetKeyboardMapping'
> bind/libao.so: undefined reference to `XRRGetCrtcGammaSize'
> bind/libao.so: undefined reference to `XGetSelectionOwner'
> bind/libao.so: undefined reference to `XCloseDisplay'
> bind/libao.so: undefined reference to `XSync'
> bind/libao.so: undefined reference to `XMapRaised'
> bind/libao.so: undefined reference to `XCheckIfEvent'
> collect2: error: ld returned 1 exit status
> ninja: build stopped: subcommand failed.
>

sethalves, 2016-08-20:
on ubuntu 16, I have that file from having run something like:

apt-get install mesa-common-dev

poppingtonic, 2016-08-22:
I already had `mesa-common-dev`, but not the file. In any case, I no longer see the first error, just the next list of `undefined reference` problems.

sethalves, 2016-08-22:
Well... I remain suspicious.  If you had mesa-common-dev installed, you should have had that header file.  Here are some other files I see attributed to that package:
```
$ dpkg -L mesa-common-dev | grep -v /usr/share
/.
/usr
/usr/lib
/usr/lib/x86_64-linux-gnu
/usr/lib/x86_64-linux-gnu/pkgconfig
/usr/lib/x86_64-linux-gnu/pkgconfig/dri.pc
/usr/include
/usr/include/GL
/usr/include/GL/glcorearb.h
/usr/include/GL/glxext.h
/usr/include/GL/gl.h
/usr/include/GL/glx_mangle.h
/usr/include/GL/gl_mangle.h
/usr/include/GL/glext.h
/usr/include/GL/glx.h
/usr/include/GL/internal
/usr/include/GL/internal/dri_interface.h
```
I would try reinstalling that package:

    sudo apt-get --reinstall install mesa-common-dev

and then removing the ao build directory and rerunning cmake, etc.

poppingtonic, 2016-08-24:
Hmm... I still don't see them. See:

Before Reinstall:
```
$ dpkg -L mesa-common-dev | grep -v /usr/share 
/.
/usr
/usr/lib
/usr/lib/x86_64-linux-gnu
/usr/lib/x86_64-linux-gnu/pkgconfig
/usr/lib/x86_64-linux-gnu/pkgconfig/dri.pc
/usr/include
/usr/include/GL
/usr/include/GL/internal
/usr/include/GL/internal/dri_interface.h
/usr/include/GL/glext.h
/usr/include/GL/glxext.h
/usr/include/GL/glx.h
/usr/include/GL/gl_mangle.h
/usr/include/GL/glx_mangle.h
/usr/include/GL/gl.h
```

After Reinstall:
```
$ sudo apt-get --reinstall install mesa-common-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  erlang-nox racket-doc
Use 'apt-get autoremove' to remove them.
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 25 not upgraded.
Need to get 261 kB of archives.
After this operation, 0 B of additional disk space will be used.
WARNING: The following packages cannot be authenticated!
  mesa-common-dev
Install these packages without verification? [y/N] y
Get:1 http://archive.ubuntu.com/ubuntu/ trusty-updates/main mesa-common-dev amd64 10.1.3-0ubuntu0.6 [261 kB]
Fetched 261 kB in 1s (229 kB/s)          
(Reading database ... 752309 files and directories currently installed.)
Preparing to unpack .../mesa-common-dev_10.1.3-0ubuntu0.6_amd64.deb ...
Unpacking mesa-common-dev (10.1.3-0ubuntu0.6) over (10.1.3-0ubuntu0.6) ...
Setting up mesa-common-dev (10.1.3-0ubuntu0.6) ...

$ dpkg -L mesa-common-dev | grep -v /usr/share
/.
/usr
/usr/lib
/usr/lib/x86_64-linux-gnu
/usr/lib/x86_64-linux-gnu/pkgconfig
/usr/lib/x86_64-linux-gnu/pkgconfig/dri.pc
/usr/include
/usr/include/GL
/usr/include/GL/internal
/usr/include/GL/internal/dri_interface.h
/usr/include/GL/glext.h
/usr/include/GL/glxext.h
/usr/include/GL/glx.h
/usr/include/GL/gl_mangle.h
/usr/include/GL/glx_mangle.h
/usr/include/GL/gl.h

```

sethalves, 2016-08-24:
I think those missing symbols should be provided by libglfw... but this is weird.  In your link line, I see -lglfw3 and in mine I see -lglfw

try:

sudo apt-get install libglfw

and then do a clean rebuild of ao.

PuercoPop, 2016-09-01:
I get the same error as the OP, besides the trying the steps above, I installed libglfw2-dev (I had libglfw3-dev) and rerun the call to c++ with -lglfw just in canse, ninja -lglfw3 for me as well. Made no difference. Any pointers?

sethalves, 2016-09-01:
Here's my entire build:
```seth@seth:~/src/ao/build$ ninja -v
[1/71] /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/tree/token.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/tree/token.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/tree/token.cpp.o -c ../kernel/src/tree/token.cpp
[2/71] /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/tree/atom.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/tree/atom.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/tree/atom.cpp.o -c ../kernel/src/tree/atom.cpp
[3/71] /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/render/region.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/render/region.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/render/region.cpp.o -c ../kernel/src/render/region.cpp
[4/71] /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/tree/tree.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/tree/tree.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/tree/tree.cpp.o -c ../kernel/src/tree/tree.cpp
[5/71] /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/tree/store.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/tree/store.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/tree/store.cpp.o -c ../kernel/src/tree/store.cpp
[6/71] /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/format/image.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/format/image.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/format/image.cpp.o -c ../kernel/src/format/image.cpp
[7/71] /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/format/contours.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/format/contours.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/format/contours.cpp.o -c ../kernel/src/format/contours.cpp
[8/71] /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/eval/row.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/eval/row.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/eval/row.cpp.o -c ../kernel/src/eval/row.cpp
[9/71] /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/eval/result.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/eval/result.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/eval/result.cpp.o -c ../kernel/src/eval/result.cpp
[10/71] /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/format/mesh.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/format/mesh.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/format/mesh.cpp.o -c ../kernel/src/format/mesh.cpp
[11/71] /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/eval/clause.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/eval/clause.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/eval/clause.cpp.o -c ../kernel/src/eval/clause.cpp
[12/71] /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/render/subregion.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/render/subregion.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/render/subregion.cpp.o -c ../kernel/src/render/subregion.cpp
[13/71] /usr/bin/c++   -DGLFW_INCLUDE_GLCOREARB -Defsw_EXPORTS -isystem ../vendor -I../ui/vendor/efsw/include -I../ui/vendor/efsw/src -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -fPIC   -Wall -Wno-long-long -w -MMD -MT ui/vendor/efsw/CMakeFiles/efsw.dir/src/efsw/Debug.cpp.o -MF ui/vendor/efsw/CMakeFiles/efsw.dir/src/efsw/Debug.cpp.o.d -o ui/vendor/efsw/CMakeFiles/efsw.dir/src/efsw/Debug.cpp.o -c ../ui/vendor/efsw/src/efsw/Debug.cpp
[14/71] /usr/bin/c++   -DGLFW_INCLUDE_GLCOREARB -Defsw_EXPORTS -isystem ../vendor -I../ui/vendor/efsw/include -I../ui/vendor/efsw/src -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -fPIC   -Wall -Wno-long-long -w -MMD -MT ui/vendor/efsw/CMakeFiles/efsw.dir/src/efsw/DirectorySnapshotDiff.cpp.o -MF ui/vendor/efsw/CMakeFiles/efsw.dir/src/efsw/DirectorySnapshotDiff.cpp.o.d -o ui/ [... cut ...] o-test.dir/__/kernel/test/affine.cpp.o test/CMakeFiles/ao-test.dir/__/kernel/test/dc.cpp.o test/CMakeFiles/ao-test.dir/__/kernel/test/jit.cpp.o test/CMakeFiles/ao-test.dir/__/kernel/test/avx.cpp.o test/CMakeFiles/ao-test.dir/main.cpp.o test/CMakeFiles/ao-test.dir/api.cpp.o  -o test/ao-test  -rdynamic bind/libao.so ui/libao-ui.a kernel/libao-kernel.a -lpng -lz -lglfw -lepoxy ui/vendor/efsw/libefsw.so -lpthread -Wl,-rpath,/home/seth/src/ao/build/bind:/home/seth/src/ao/build/ui/vendor/efsw && :
```

PuercoPop, 2016-09-01:
It seems similar to my build. I'm inexperienced with C++ but most of the missing functions are from X11, shouldn't -lX11 be including in the options passed ti /usr/bin/c++?

```
~/Projects/ao/build $ ninja -v
[1/71] /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/tree/store.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/tree/store.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/tree/store.cpp.o -c ../kernel/src/tree/store.cpp
[2/71] /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/tree/tree.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/tree/tree.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/tree/tree.cpp.o -c ../kernel/src/tree/tree.cpp
[3/71] /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/tree/token.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/tree/token.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/tree/token.cpp.o -c ../kernel/src/tree/token.cpp
[4/71] /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/tree/atom.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/tree/atom.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/tree/atom.cpp.o -c ../kernel/src/tree/atom.cpp
[5/71] /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/format/image.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/format/image.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/format/image.cpp.o -c ../kernel/src/format/image.cpp
[6/71] /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/format/mesh.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/format/mesh.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/format/mesh.cpp.o -c ../kernel/src/format/mesh.cpp
[7/71] /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/format/contours.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/format/contours.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/format/contours.cpp.o -c ../kernel/src/format/contours.cpp
[8/71] /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/render/quadtree.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/render/quadtree.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/render/quadtree.cpp.o -c ../kernel/src/render/quadtree.cpp
[9/71] /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/render/dc2d.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/render/dc2d.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/render/dc2d.cpp.o -c ../kernel/src/render/dc2d.cpp
[10/71] /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/render/subregion.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/render/subregion.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/render/subregion.cpp.o -c ../kernel/src/render/subregion.cpp
[11/71] /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/render/octree.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/render/octree.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/render/octree.cpp.o -c ../kernel/src/render/octree.cpp
[12/71] /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/render/dc.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/render/dc.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/render/dc.cpp.o -c ../kernel/src/render/dc.cpp
[13/71] /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/render/heightmap.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/render/heightmap.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/render/heightmap.cpp.o -c ../kernel/src/render/heightmap.cpp
[14/71] /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/render/region.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/render/region.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/render/region.cpp.o -c ../kerne [... cut ...]
bind/libao.so: undefined reference to `glXSwapBuffers'
bind/libao.so: undefined reference to `XRRGetCrtcGammaSize'
bind/libao.so: undefined reference to `XGetSelectionOwner'
bind/libao.so: undefined reference to `XCloseDisplay'
bind/libao.so: undefined reference to `XSync'
bind/libao.so: undefined reference to `XMapRaised'
bind/libao.so: undefined reference to `XkbGetKeyboard'
bind/libao.so: undefined reference to `XCheckIfEvent'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

sethalves, 2016-09-01:
In both your cases, the only important difference I can see is that my link line has -lglfw and yours have -lglfw3.  I have /usr/lib/x86_64-linux-gnu/libglfw.so which is provided by the ubuntu package libglfw3-dev.  In ui/CMakeLists.txt I see this line:

    pkg_search_module(GLFW REQUIRED glfw3)

I've got /usr/lib/x86_64-linux-gnu/pkgconfig/glfw3.pc which contains
```
prefix=/usr
exec_prefix=${prefix}
includedir=${prefix}/include
libdir=${exec_prefix}/lib/x86_64-linux-gnu

Name: GLFW
Description: A multi-platform library for OpenGL, window and input
Version: 3.1.2
URL: http://www.glfw.org/
Requires.private:  x11 xrandr xinerama xi xxf86vm xcursor gl
Libs: -L${libdir} -lglfw
Libs.private:  -lrt -lm -ldl
Cflags: -I${includedir}
```

PuercoPop, 2016-09-01:
After I chage the variable `GLFW_LIBRARIES:INTERNAL`(from the file `build/CMakeCache.txt`) from `glfw3` to `glfw` the build succeded. :tada: 

Thanks!

sethalves, 2016-09-01:
@PuercoPop would you find running `find /usr -name glfw3.pc` and pasting the contents of that file here?
I would also be interested in the output of `dpkg --list libglfw3-dev` and `cat /etc/issue`

PuercoPop, 2016-09-01:
```
$ find /usr/ -name glfw3.pc
/usr/local/lib/pkgconfig/glfw3.pc
/usr/lib/x86_64-linux-gnu/pkgconfig/glfw3.pc
```

```
$ dpkg --list libglfw3-dev
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
ii  libglfw3-dev:a 3.1.2-3      amd64        portable library for OpenGL, wind
```

```
$ cat /etc/issue
Ubuntu 16.04.1 LTS \n \l

```

sethalves, 2016-09-01:
Could you also run `cat /usr/local/lib/pkgconfig/glfw3.pc` and `cat /usr/lib/x86_64-linux-gnu/pkgconfig/glfw3.pc` and paste those in here also?  I suspect the problem is that you've built and installed a version of glfw3 (the one in /usr/local) and that one is interfering with the ao build.

PuercoPop, 2016-09-01:
Yeah, that would make sense, I always install to /usr/local/ when installing from source. Although I don't remember doing so in this case it is possible that I may have forgotten.

```
$ cat /usr/local/lib/pkgconfig/glfw3.pc
prefix=/usr/local
exec_prefix=${prefix}
includedir=${prefix}/include
libdir=${exec_prefix}/lib

Name: GLFW
Description: A multi-platform library for OpenGL, window and input
Version: 3.1.1
URL: http://www.glfw.org/
Requires.private:  x11 xrandr xinerama xi xxf86vm xcursor gl
Libs: -L${libdir} -lglfw3
Libs.private:  -lrt -lm
Cflags: -I${includedir}
```

```
$ cat /usr/lib/x86_64-linux-gnu/pkgconfig/glfw3.pc
prefix=/usr
exec_prefix=${prefix}
includedir=${prefix}/include
libdir=${exec_prefix}/lib/x86_64-linux-gnu

Name: GLFW
Description: A multi-platform library for OpenGL, window and input
Version: 3.1.2
URL: http://www.glfw.org/
Requires.private:  x11 xrandr xinerama xi xxf86vm xcursor gl
Libs: -L${libdir} -lglfw
Libs.private:  -lrt -lm -ldl
Cflags: -I${includedir}
```

mkeeter, 2017-09-10:
I just pushed a history-changing update to `master` that changes a bunch of the code and includes a more featureful GUI.  It removes the dependency on GLFW, since the minimal GUI is removed, so I'm going to close this issue for now.  If you have trouble building the new `master`, let me know and I'll see if I can help.

closed 2017-09-10 by mkeeter