libfive/libfive #9: Build fails at src/eval/evaluator.cpp

poppingtonic, 2016-08-20:
I'm using Ubuntu trusty, and installed the dependencies as:
```bash
sudo add-apt-repository -y ppa:keithw/glfw3
sudo apt-get update
sudo apt-get install -y libglfw3 libglfw3-dev libepoxy0 libepoxy-dev
cmake -G Ninja ..
```

> FAILED: kernel/CMakeFiles/ao-kernel.dir/src/eval/evaluator.cpp.o 
> /usr/bin/c++    -Wall -Wextra -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -isystem ../vendor -I../kernel/include -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/eval/evaluator.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/eval/evaluator.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/eval/evaluator.cpp.o -c ../
> kernel/src/eval/evaluator.cpp
> In file included from ../kernel/include/ao/kernel/tree/tree.hpp:26:0,
>                  from ../kernel/src/eval/evaluator.cpp:24:
> ../kernel/include/ao/kernel/tree/opcode.hpp:54:5: warning: multi-line comment [-Wcomment]
>      //             /      \
>      ^
> ../kernel/include/ao/kernel/tree/opcode.hpp:56:5: warning: multi-line comment [-Wcomment]
>      //        /    \      /      \
>      ^
> ../kernel/include/ao/kernel/tree/opcode.hpp:58:5: warning: multi-line comment [-Wcomment]
>      //    /   \   /  \  /   \
>      ^
> ../kernel/src/eval/evaluator.cpp: In constructor ‘Evaluator::Evaluator(const Tree*, const mat4&)’:
> ../kernel/src/eval/evaluator.cpp:49:17: error: ‘align’ is not a member of ‘std’
>                  std::align(alignment, sizeof(Clause) * count, buf, bytes));
>                  ^
> [22/71] Building CXX object kernel/CMakeFiles/ao-kernel.dir/src/render/quadtree.cpp.o
> In file included from ../kernel/include/ao/kernel/eval/clause.hpp:23:0,
>                  from ../kernel/include/ao/kernel/eval/evaluator.hpp:29,
>                  from ../kernel/include/ao/kernel/render/xtree.hpp:28,
>                  from ../kernel/include/ao/kernel/render/quadtree.hpp:21,
>                  from ../kernel/src/render/quadtree.cpp:20:
> ../kernel/include/ao/kernel/tree/opcode.hpp:54:5: warning: multi-line comment [-Wcomment]
>      //             /      \
>      ^
> ../kernel/include/ao/kernel/tree/opcode.hpp:56:5: warning: multi-line comment [-Wcomment]
>      //        /    \      /      \
>      ^
> ../kernel/include/ao/kernel/tree/opcode.hpp:58:5: warning: multi-line comment [-Wcomment]
>      //    /   \   /  \  /   \
>      ^
> [23/71] Building CXX object kernel/CMakeFiles/ao-kernel.dir/src/render/octree.cpp.o
> In file included from ../kernel/include/ao/kernel/eval/clause.hpp:23:0,
>                  from ../kernel/include/ao/kernel/eval/evaluator.hpp:29,
>                  from ../kernel/include/ao/kernel/render/xtree.hpp:28,
>                  from ../kernel/include/ao/kernel/render/octree.hpp:21,
>                  from ../kernel/src/render/octree.cpp:21:
> ../kernel/include/ao/kernel/tree/opcode.hpp:54:5: warning: multi-line comment [-Wcomment]
>      //             /      \
>      ^
> ../kernel/include/ao/kernel/tree/opcode.hpp:56:5: warning: multi-line comment [-Wcomment]
>      //        /    \      /      \
>      ^
> ../kernel/include/ao/kernel/tree/opcode.hpp:58:5: warning: multi-line comment [-Wcomment]
>      //    /   \   /  \  /   \
>      ^
> ninja: build stopped: subcommand failed.
>

tck42, 2016-08-20:
Attached patch got it compiling for me, though I'm having GL issues (old graphics card) so I can't tell if this is a proper fix (or I'd just PR it). Can you please test it and let me know?

[isnan.patch.txt](https://github.com/mkeeter/ao/files/428090/isnan.patch.txt)

tck42, 2016-08-20:
Disregard, sorry. Different issue.

iffsid, 2016-08-20:
I got the following two errors for the same file (`evaluator.cpp`).

```
../kernel/src/eval/evaluator.cpp: In constructor ‘Evaluator::Evaluator(const Tree*, const mat4&)’:
../kernel/src/eval/evaluator.cpp:49:17: error: ‘align’ is not a member of ‘std’
                 std::align(alignment, sizeof(Clause) * count, buf, bytes));
                 ^
../kernel/src/eval/evaluator.cpp: In function ‘void clause(Opcode, const float*, const float*, float*, size_t)’:
../kernel/src/eval/evaluator.cpp:197:32: error: ‘isnan’ was not declared in this scope
             out[i] = isnan(a[i]) ? b[i] : a[i];
                                ^
```

It turns out `gcc-4.9` has a bug in that it's missing `std::align` (issue [here](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=57350)), so I replaced the call with the fix supplied in the end of that thread.

As for the `isnan` error, you can just replace it with `std::isnan`.

The patch for both changes is available [here](http://lpaste.net/1936906778776174592).

mkeeter, 2016-08-20:
I've brought in @sethalves 's patches and it looks like @poppingtonic has gotten past this stage (and is having different issues in #11).

closed 2016-08-20 by mkeeter