libfive/libfive #14: exception after changing watched file

kbaldor, 2016-08-21:
OSX 10.11.6

I can watch a file and its contents will be displayed, but after saving the file ao-guile quits after printing

guile: uncaught throw to wrong-type-arg: (scm_i_foreign_call Wrong type argument in position ~A (expecting ~A): ~S (1 POINTER_P #<unspecified>) (#<unspecified>))

The watched file contains

(define (neg x) (* -1 x))

(define (chop inner outer)
  (difference
      (cube (map neg (list outer outer outer))
                     (list outer outer outer))
      (union
          (cube (map neg (list inner outer outer))
                         (list inner outer outer))
          (cube (map neg (list outer inner outer))
                         (list outer inner outer))
          (cube (map neg (list outer outer inner))
                         (list outer outer inner)))))

(define m (difference
              (difference 
                  (sphere '(0 0 0) 1)  
                  (sphere '(0 0 0) .8) )
              (chop .1 1)))

(ao-show m)

mkeeter, 2016-08-21:
Does this happen repeatable?  I tried dropping that script into a file, `ao-watch`ing it, and changing the file, but I didn't get that crash.

kbaldor, 2016-08-21:
Yes. Every time.

I had some difficulty compiling (I had to explicitly add paths to a few of the CMakeLists.txt), so I may have built it incorrectly. Perhaps that should be a separate issue.

mkeeter, 2016-08-21:
Can you paste in the diff from `master`, in case any of the build changes could explain this?

kbaldor, 2016-08-22:
index a7d4da4..2c7531a 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -9,6 +9,7 @@ set(CMAKE_CXX_FLAGS_RELEASE  "-O3 -DRELEASE")
 set(CMAKE_CXX_FLAGS_DEBUG    "-O0")
 
 include_directories(SYSTEM vendor)
+link_directories(/usr/local/lib)
 
 ################################################################################
 
diff --git a/bind/CMakeLists.txt b/bind/CMakeLists.txt
index f7661f2..2f2d895 100644
--- a/bind/CMakeLists.txt
+++ b/bind/CMakeLists.txt
@@ -1,7 +1,11 @@
 add_library(ao SHARED api.cpp)
 target_link_libraries(ao ao-kernel ao-ui ${GUILE_LIBRARY})
+link_directories(ao /usr/local/lib)
 target_include_directories(ao SYSTEM PRIVATE ${GUILE_INCLUDE_DIR})
 target_include_directories(ao PRIVATE .)
+target_include_directories(ao PRIVATE /opt/local/include)
+target_include_directories(ao PRIVATE /usr/local/Cellar/libepoxy/1.3.1/include)
+target_include_directories(ao PRIVATE /usr/local/Cellar/glfw3/3.2.1/include)
 
 add_custom_command(
     TARGET ao POST_BUILD
diff --git a/test/CMakeLists.txt b/test/CMakeLists.txt
index 3b77cd9..4e0b04b 100644
--- a/test/CMakeLists.txt
+++ b/test/CMakeLists.txt
@@ -2,4 +2,7 @@ file(GLOB SRC_FILES ../ui/test/*.cpp ../kernel/test/*.cpp)
 
 add_executable(ao-test ${SRC_FILES} main.cpp api.cpp)
 target_include_directories(ao-test PUBLIC ../bind)
+target_include_directories(ao-test PRIVATE /opt/local/include)
+target_include_directories(ao-test PRIVATE /usr/local/Cellar/libepoxy/1.3.1/include)
+target_include_directories(ao-test PRIVATE /usr/local/Cellar/glfw3/3.2.1/include)
 target_link_libraries(ao-test ao)

afriggeri, 2016-09-05:
OSX 10.11.6

I'm having the exact same issue, but I didn't have any problem building (nor did I have to modify CMakeLists.txt).

```
Ao> (ao-watch "examples/menger.scm")
Watching /Users/frg/Sources/ao/examples/menger.scm
ERROR: In procedure #<procedure 1102d7280 (_ _)>:
ERROR: In procedure scm_i_foreign_call: Wrong type argument in position 1 (expecting POINTER_P): #<unspecified>

Entering a new prompt.  Type `,bt' for a backtrace or `,q' to continue.
Ao [1]>
```

And then if I save `menger.scm` it crashes with the same error as op:

    guile: uncaught throw to wrong-type-arg: (scm_i_foreign_call Wrong type argument in position ~A (expecting ~A): ~S (1 POINTER_P #<unspecified>) (#<unspecified>))

mkeeter, 2016-09-05:
@afriggeri Can you get a backtrace from the first error message (by typing `,bt`)?

afriggeri, 2016-09-05:
Sure:

```
Ao> (ao-watch "/Users/frg/Sources/ao/examples/menger.scm")
Watching /Users/frg/Sources/ao/examples/menger.scm
ERROR: In procedure #<procedure 10db03800 (_ _)>:
ERROR: In procedure scm_i_foreign_call: Wrong type argument in position 1 (expecting POINTER_P): #<unspecified>

Entering a new prompt.  Type `,bt' for a backtrace or `,q' to continue.
Ao [1]> ,bt
           0 (#<procedure 10db03800 (_ _)> #<pointer 0x7fa8b3c18150> #<pointer 0x7fa8b3c178c0>)
Ao [1]> 
```

mkeeter, 2016-09-06:
Do you have multiple instances of Guile or `libguile-2.0` installed?

Check what ```otool -L `which guile `|grep guile``` reports: this is what the Guile command-line executable is linked again.

Then look in `/usr/local/lib` (or wherever it was installed) for `libguile-2.0`.  Are these libraries different?

mkeeter, 2016-11-30:
Can you see if PR #23 fixes this?  I managed to make this reproducible while changing other things, then made it stop happening, so maybe it's gone...

mkeeter, 2017-09-10:
(closing due to inaction and because the new `master` uses a completely different strategy to embed Guile).

closed 2017-09-10 by mkeeter