libfive/libfive #18: Running issue: `guile: Couldn't find current GLX or EGL context.`

galex-713, 2016-11-15:
Hi, I tried to exactely follow the procedure to install from source, and I’m under Debian GNU/Linux testing, and when running, it gives me this:

> galex-713@PC713:~/src/ao/bin$ guile ao-guile 
> ;;; note: auto-compilation is enabled, set GUILE_AUTO_COMPILE=0
> ;;;       or pass the --no-auto-compile argument to disable.
> ;;; compiling /home/galex-713/src/ao/bin/ao-guile
> ;;; compiling /home/galex-713/src/ao/bin/../bind/guile/ao/bind.scm
> ;;; compiled /home/galex-713/.cache/guile/ccache/2.0-LE-4-2.0/home/galex-713/src/ao/bind/guile/ao/bind.scm.go
> ;;; compiled /home/galex-713/.cache/guile/ccache/2.0-LE-4-2.0/home/galex-713/src/ao/bin/ao-guile.go
> ;;; compiling /home/galex-713/src/ao/bin/../bind/guile/ao/operators.scm
> ;;; compiling /home/galex-713/src/ao/bin/../bind/guile/ao/jit.scm
> ;;; compiled /home/galex-713/.cache/guile/ccache/2.0-LE-4-2.0/home/galex-713/src/ao/bind/guile/ao/jit.scm.go
> guile: Couldn't find current GLX or EGL context.

It also seems a certain Ralf Mattes encountered similar problem as he said on the 21 August on guile-user mailing-list…

mkeeter, 2016-11-29:
Okay, looks like an OpenGL issue.  Can you show me the output of running `glxinfo`?

galex-713, 2016-11-29:
Here:

> name of display: :0
> display: :0  screen: 0
> direct rendering: Yes
> server glx vendor string: SGI
> server glx version string: 1.4
> server glx extensions:
>     GLX_ARB_create_context, GLX_ARB_create_context_profile, 
>     GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, GLX_ARB_multisample, 
>     GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
>     GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
>     GLX_EXT_import_context, GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, 
>     GLX_EXT_visual_rating, GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, 
>     GLX_OML_swap_method, GLX_SGIS_multisample, GLX_SGIX_fbconfig, 
>     GLX_SGIX_pbuffer, GLX_SGIX_visual_select_group, GLX_SGI_swap_control
> client glx vendor string: Mesa Project and SGI
> client glx version string: 1.4
> client glx extensions:
>     GLX_ARB_create_context, GLX_ARB_create_context_profile, 
>     GLX_ARB_create_context_robustness, GLX_ARB_fbconfig_float, 
>     GLX_ARB_framebuffer_sRGB, GLX_ARB_get_proc_address, GLX_ARB_multisample, 
>     GLX_EXT_buffer_age, GLX_EXT_create_context_es2_profile, 
>     GLX_EXT_create_context_es_profile, GLX_EXT_fbconfig_packed_float, 
>     GLX_EXT_framebuffer_sRGB, GLX_EXT_import_context, 
>     GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, GLX_EXT_visual_rating, 
>     GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, 
>     GLX_MESA_multithread_makecurrent, GLX_MESA_query_renderer, 
>     GLX_MESA_swap_control, GLX_OML_swap_method, GLX_OML_sync_control, 
>     GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
>     GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
>     GLX_SGI_swap_control, GLX_SGI_video_sync
> GLX version: 1.4
> GLX extensions:
>     GLX_ARB_create_context, GLX_ARB_create_context_profile, 
>     GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, 
>     GLX_ARB_get_proc_address, GLX_ARB_multisample, 
>     GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
>     GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
>     GLX_EXT_import_context, GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, 
>     GLX_EXT_visual_rating, GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, 
>     GLX_MESA_multithread_makecurrent, GLX_MESA_query_renderer, 
>     GLX_MESA_swap_control, GLX_OML_swap_method, GLX_OML_sync_control, 
>     GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
>     GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
>     GLX_SGI_swap_control, GLX_SGI_video_sync
> Extended renderer info (GLX_MESA_query_renderer):
>     Vendor: Intel Open Source Technology Center (0x8086)
>     Device: Mesa DRI Intel(R) 945GM x86/MMX/SSE2 (0x27a2)
>     Version: 12.0.4
>     Accelerated: yes
>     Video memory: 192MB
>     Unified memory: yes
>     Preferred profile: compat (0x2)
>     Max core profile version: 0.0
>     Max compat profile version: 2.1
>     Max GLES1 profile version: 1.1
>     Max GLES[23] profile version: 2.0
> OpenGL vendor string: Intel Open Source Technology Center
> OpenGL renderer string: Mesa DRI Intel(R) 945GM x86/MMX/SSE2
> OpenGL version string: 2.1 Mesa 12.0.4
> OpenGL shading language version string: 1.20
> OpenGL extensions:
>     GL_3DFX_texture_compression_FXT1, GL_AMD_shader_trinary_minmax, 
>     GL_ANGLE_texture_compression_dxt3, GL_ANGLE_texture_compression_dxt5, 
>     GL_APPLE_object_purgeable, GL_APPLE_packed_pixels, 
>     GL_APPLE_vertex_array_object, GL_ARB_ES2_compatibility, 
>     GL_ARB_clear_buffer_object, GL_ARB_compressed_texture_pixel_storage, 
>     GL_ARB_copy_buffer, GL_ARB_debug_output, GL_ARB_depth_texture, 
>     GL_ARB_draw_buffers, GL_ARB_draw_elements_base_vertex, 
[... 90 lines cut ...]
> ----------------------------------------------------------------------------
> 0x05c  0 tc  0  16  0 r  y .   5  6  5  0 .  .  0  0  0  0  0  0  0  0 0 None
> 0x05d  0 tc  0  16  0 r  . .   5  6  5  0 .  .  0  0  0  0  0  0  0  0 0 None
> 0x05e  0 tc  0  16  0 r  y .   5  6  5  0 .  .  0 16  0  0  0  0  0  0 0 None
> 0x05f  0 tc  0  16  0 r  . .   5  6  5  0 .  .  0 16  0  0  0  0  0  0 0 None
> 0x060 24 tc  0  32  0 r  y .   8  8  8  8 .  .  0  0  0  0  0  0  0  0 0 None
> 0x061 24 tc  0  32  0 r  . .   8  8  8  8 .  .  0  0  0  0  0  0  0  0 0 None
> 0x062 24 tc  0  32  0 r  y .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
> 0x063 24 tc  0  32  0 r  . .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
> 0x064  0 tc  0  16  0 r  y .   5  6  5  0 .  .  0 16  0  0  0  0  0  0 0 None
> 0x065  0 tc  0  16  0 r  y .   5  6  5  0 .  .  0 16  0 16 16 16  0  0 0 Slow
> 0x066 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
> 0x067 24 tc  0  32  0 r  y .   8  8  8  8 .  .  0 24  8 16 16 16 16  0 0 Slow
> 0x068  0 dc  0  16  0 r  y .   5  6  5  0 .  .  0  0  0  0  0  0  0  0 0 None
> 0x069  0 dc  0  16  0 r  . .   5  6  5  0 .  .  0  0  0  0  0  0  0  0 0 None
> 0x06a  0 dc  0  16  0 r  y .   5  6  5  0 .  .  0 16  0  0  0  0  0  0 0 None
> 0x06b  0 dc  0  16  0 r  . .   5  6  5  0 .  .  0 16  0  0  0  0  0  0 0 None
> 0x06c 24 dc  0  32  0 r  y .   8  8  8  8 .  .  0  0  0  0  0  0  0  0 0 None
> 0x06d 24 dc  0  32  0 r  . .   8  8  8  8 .  .  0  0  0  0  0  0  0  0 0 None
> 0x06e 24 dc  0  32  0 r  y .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
> 0x06f 24 dc  0  32  0 r  . .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
> 0x070  0 dc  0  16  0 r  y .   5  6  5  0 .  .  0 16  0  0  0  0  0  0 0 None
> 0x071  0 dc  0  16  0 r  y .   5  6  5  0 .  .  0 16  0 16 16 16  0  0 0 Slow
> 0x072 24 dc  0  32  0 r  y .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
> 0x073 24 dc  0  32  0 r  y .   8  8  8  8 .  .  0 24  8 16 16 16 16  0 0 Slow

mkeeter, 2016-11-29:
It looks like the problem is in driver support – your OpenGL drivers only support 2.1, and Ao uses [3.3](https://github.com/mkeeter/ao/blob/master/ui/include/ao/ui/gl/core.hpp#L21-L22) or later.

Are there binary drivers for your GPU that would give a more recent OpenGL version?

galex-713, 2016-11-29:
I didn’t hear of it anywhere :/ I’m using a thinkpad X60T (and it features a quite old gpu: Intel Corporation Mobile 945GM/GMS, 943/940GML Express Integrated Graphics Controller (rev 03) (according lspci)) and actually it’s one of the few PC that supports fully free software, so it’s unfortunate to learn that most stuff involving OpenGL in the GNU Guile project (and GNU Guile in general, like Sly) doesn’t work on any PC supporting fully-free-software OSes…

mkeeter, 2016-11-30:
I don't think this is a Guile / Ao problem at all, since you'll see it whenever you try to get an OpenGL context newer than 2.1.

That chipset seems quite old.  The only thing I can find is [this link](https://downloadcenter.intel.com/product/81510/Graphics-Drivers-for-Mobile-Intel-915GM-GMS-910GML-Express-Chipset-Family), where Intel offers drivers.

I'm closing this for now, as I don't have the resources to make Ao work with OpenGL versions predating 3.3.  Feel free to comment / re-open if you get it working or want to try making an 2.1 port.

closed 2016-11-30 by mkeeter

galex-713, 2016-11-30:
I mean the “guile/ao” problem is that they’re designed for opengl versions newer than 2.1/2.3 that makes them require computers whose working require non-free software, unfortunately making these programs indirectly require non-free software…

mkeeter, 2016-11-30:
OpenGL 3.3 was released in 2010, so I'm comfortable making it a requirement.

If a particular set of hardware doesn't support it, I'll encourage you to take it up with Debian / Intel / upstream – I don't have the resources to make Ao support 10-year old integrated graphics chipsets, especially with an insistence on entirely free software.

> Are these drivers free software and do they support recent-enough versions of OpenGL?
I'm not sure, you'll have to check it out for yourself.

mkeeter, 2016-11-30:
That being said, [this thread](https://ubuntuforums.org/showthread.php?t=1741783) may also be of use – it looks like folks talking about the same chipset.

galex-713, 2016-11-30:
Why Debian? they didn’t make ao nor my GPU, and afaik the opengl version of the driver depends of the features supported by a GPU, and none recent GPU is supported by computers old enough not to be stuck with proprietary BIOSes/UEFIs… Because around 2006 Intel and AMD decided that CPUs needed non-disablable stuff that provide full remote control of PCs and prevent free software initialization… So Intel is not gonna help since they don’t care about free software…