libfive/libfive #21: Error while building: multithreading namespace errors in cache.cpp

galex-713, 2016-11-22:
I’ve noticed that some commits ago ao stopped building correctly, giving me this:
galex-713@PC713:~/src/ao/build$ ninja 

> [1/67] Building CXX object kernel/CMakeFiles/ao-kernel.dir/src/tree/opcode.cpp.o
> ../kernel/src/tree/opcode.cpp: In function ‘size_t Opcode::args(Opcode::Opcode)’:
> ../kernel/src/tree/opcode.cpp:65:1: warning: control reaches end of non-void function [-Wreturn-type]
>  }
>  ^
> ../kernel/src/tree/opcode.cpp: In function ‘std::__cxx11::string Opcode::to_str(Opcode::Opcode)’:
> ../kernel/src/tree/opcode.cpp:103:1: warning: control reaches end of non-void function [-Wreturn-type]
>  }
>  ^
> [2/67] Building CXX object kernel/CMakeFiles/ao-kernel.dir/src/tree/cache.cpp.o
> FAILED: kernel/CMakeFiles/ao-kernel.dir/src/tree/cache.cpp.o 
> /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MD -MT kernel/CMakeFiles/ao-kernel.dir/src/tree/cache.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/tree/cache.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/tree/cache.cpp.o -c ../kernel/src/tree/cache.cpp
> In file included from ../kernel/src/tree/cache.cpp:22:0:
> ../kernel/include/ao/kernel/tree/cache.hpp:177:17: error: ‘mutex’ in namespace ‘std’ does not name a type
>      static std::mutex instance_lock;
>                  ^~~~~
> ../kernel/src/tree/cache.cpp:29:6: error: ‘mutex’ in namespace ‘std’ does not name a type
>  std::mutex Cache::instance_lock;
>       ^~~~~
> ../kernel/src/tree/cache.cpp: In static member function ‘static std::shared_ptr<Cache> Cache::instance()’:
> ../kernel/src/tree/cache.cpp:35:5: error: ‘lock_guard’ is not a member of ‘std’
>      std::lock_guard<std::mutex> g(instance_lock);
>      ^~~
> ../kernel/src/tree/cache.cpp:35:21: error: ‘mutex’ is not a member of ‘std’
>      std::lock_guard<std::mutex> g(instance_lock);
>                      ^~~
> ../kernel/src/tree/cache.cpp:35:35: error: ‘instance_lock’ was not declared in this scope
>      std::lock_guard<std::mutex> g(instance_lock);
>                                    ^~~~~~~~~~~~~
> ../kernel/src/tree/cache.cpp:35:48: error: ‘g’ was not declared in this scope
>      std::lock_guard<std::mutex> g(instance_lock);
>                                                 ^
> ../kernel/src/tree/cache.cpp: In static member function ‘static void Cache::reset()’:
> ../kernel/src/tree/cache.cpp:47:5: error: ‘lock_guard’ is not a member of ‘std’
>      std::lock_guard<std::mutex> g(instance_lock);
>      ^~~
> ../kernel/src/tree/cache.cpp:47:21: error: ‘mutex’ is not a member of ‘std’
>      std::lock_guard<std::mutex> g(instance_lock);
>                      ^~~
> ../kernel/src/tree/cache.cpp:47:35: error: ‘instance_lock’ was not declared in this scope
>      std::lock_guard<std::mutex> g(instance_lock);
>                                    ^~~~~~~~~~~~~
> ../kernel/src/tree/cache.cpp:47:48: error: ‘g’ was not declared in this scope
>      std::lock_guard<std::mutex> g(instance_lock);
>                                                 ^
> [4/67] Building CXX object kernel/CMakeFiles/ao-kernel.dir/src/format/contours.cpp.o
> FAILED: kernel/CMakeFiles/ao-kernel.dir/src/format/contours.cpp.o 
> /usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MD -MT kernel/CMakeFiles/ao-kernel.dir/src/format/contours.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/format/contours.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/format/contours.cpp.o -c ../kernel/src/format/contours.cpp
> In file included from ../kernel/include/ao/kernel/tree/tree.hpp:27:0,
>                  from ../kernel/include/ao/kernel/eval/evaluator.hpp:30,
>                  from ../kernel/include/ao/kernel/render/xtree.hpp:30,
>                  from ../kernel/include/ao/kernel/render/quadtree.hpp:21,
>                  from ../kernel/include/ao/kernel/format/contours.hpp:24,
>                  from ../kernel/src/format/contours.cpp:22:
> ../kernel/include/ao/kernel/tree/cache.hpp:177:17: error: ‘mutex’ in namespace ‘std’ does not name a type
>      static std::mutex instance_lock;
>                  ^~~~~
> ninja: build stopped: subcommand failed.
> galex-713@PC713:~/src/ao/build$ 

Please notice me if I forgot to give any useful information (I really have the impression to have…)

mkeeter, 2016-11-29:
Sorry, I wasn't watching this project (thanks, Github?) so I didn't get a notification about this!

I've added `#include <mutex>` to the relevant header file – can you pull `master` and try building again?

galex-713, 2016-11-29:
Ok cool ^^ no more mutex namespace problems :D but more other compilation errors that prevent building… should I open another issue and close this one? or is continuing on this one ok?

Just in case, I actually get that if I retry building again:

```
galex-713@PC713:~/src/ao/build$ ninja
[12/54] Building CXX object kernel/CMakeFiles/ao-kernel.dir/src/eval/evaluator.cpp.o
FAILED: kernel/CMakeFiles/ao-kernel.dir/src/eval/evaluator.cpp.o 
/usr/bin/c++    -isystem ../vendor -I../kernel/include -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -MD -MT kernel/CMakeFiles/ao-kernel.dir/src/eval/evaluator.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/eval/evaluator.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/eval/evaluator.cpp.o -c /home/galex-713/src/ao/kernel/src/eval/evaluator.cpp
/home/galex-713/src/ao/kernel/src/eval/evaluator.cpp: In member function ‘const float* Evaluator::values(Result::Index)’:
/home/galex-713/src/ao/kernel/src/eval/evaluator.cpp:1217:45: error: cannot convert ‘__gnu_cxx::__alloc_traits<std::allocator<std::array<float, 256u> > >::value_type {aka std::array<float, 256u>}’ to ‘const float*’ for argument ‘2’ to ‘void eval_clause_values(Opcode::Opcode, const float*, const float*, float*, Result::Index)’
                     result.f[itr->id], count);
                                             ^
/home/galex-713/src/ao/kernel/src/eval/evaluator.cpp:1221:22: error: cannot convert ‘__gnu_cxx::__alloc_traits<std::allocator<std::array<float, 256u> > >::value_type {aka std::array<float, 256u>}’ to ‘const float*’ in return
     return result.f[0];
                      ^
/home/galex-713/src/ao/kernel/src/eval/evaluator.cpp: In member function ‘std::tuple<const float*, const float*, const float*, const float*> Evaluator::derivs(size_t)’:
/home/galex-713/src/ao/kernel/src/eval/evaluator.cpp:1265:25: error: cannot convert ‘__gnu_cxx::__alloc_traits<std::allocator<std::array<float, 256u> > >::value_type {aka std::array<float, 256u>}’ to ‘const float*’ for argument ‘2’ to ‘void eval_clause_derivs(Opcode::Opcode, const float*, const float*, const float*, const float*, const float*, const float*, const float*, const float*, float*, float*, float*, float*, Result::Index)’
                    count);
                         ^
/home/galex-713/src/ao/kernel/src/eval/evaluator.cpp:1284:5: error: no matching function for call to ‘std::tuple<const float*, const float*, const float*, const float*>::tuple(<brace-enclosed initializer list>)’
     };
     ^
In file included from /usr/include/c++/6/functional:55:0,
                 from /usr/include/c++/6/memory:79,
                 from /home/galex-713/src/ao/kernel/src/eval/evaluator.cpp:20:
/usr/include/c++/6/tuple:808:11: note: candidate: template<class _Alloc, class ... _UElements, typename std::enable_if<(std::_TC<(4u == sizeof... (_UElements)), const float*, const float*, const float*, const float*>::_MoveConstructibleTuple<_UElements ...>() && (! std::_TC<(4u == sizeof... (_UElements)), const float*, const float*, const float*, const float*>::_ImplicitlyMoveConvertibleTuple<_UElements ...>())), bool>::type <anonymous> > std::tuple< <template-parameter-1-1> >::tuple(std::allocator_arg_t, const _Alloc&, std::tuple<_Args2 ...>&&)
  explicit tuple(allocator_arg_t __tag, const _Alloc& __a,
           ^~~~~
/usr/include/c++/6/tuple:808:11: note:   template argument deduction/substitution failed:
/home/galex-713/src/ao/kernel/src/eval/evaluator.cpp:1284:5: note:   ‘__gnu_cxx::__alloc_traits<std::allocator<std::array<float, 256u> > >::value_type {aka std::array<float, 256u>}’ is not derived from ‘std::tuple<_Elements ...>’
     };
     ^
In file included from /usr/include/c++/6/functional:55:0,
                 from /usr/include/c++/6/memory:79,
                 from /home/galex-713/src/ao/kernel/src/eval/evaluator.cpp:20:
/usr/include/c++/6/tuple:796:2: note: candidate: template<class _Alloc, class ... _UElements, typename std::enable_if<(std::_TC<(4u == sizeof... (_UElements)), const float*, const float*, const float*, const float*>::_MoveConstructibleTuple<_UElements ...>() && std::_TC<(4u == sizeof... (_UElements)), const float*, const float*, const float*, const float*>::_ImplicitlyMoveConvertibleTuple<_UElements ...>()), bool>::type <anonymous> > std::tuple< <template-parameter-1-1> >::tuple(std::allocator_arg_t, const _Alloc&, std::tuple<_Args2 ...>&&)
  tuple(allocator_arg_t __tag, const _Alloc& __a,
  ^~~~~
/usr/include/c++/6/tuple:796:2: note:   template argument deduction/substitution failed:
/home/galex-713/src/ao/kernel/src/eval/evaluator.cpp:1284:5: note:   ‘__gnu_cxx::__alloc_traits<std::allocator<std::array<float, 256u> > >::value_type {aka std::array<float, 256u>}’ is not derived from ‘std::tuple<_Elements ...>’
     };
     ^
In file included from /usr/include/c++/6/functional:55:0,
                 from /usr/include/c++/6/memory:79,
                 from /home/galex-713/src/ao/kernel/src/eval/evaluator.cpp:20:
/usr/include/c++/6/tuple:784:11: note: candidate: template<class _Alloc, class ... _UElements, typename std::enable_if<(std::_TC<(4u == sizeof... (_UElements)), const float*, const float*, const float*, const float*>::_ConstructibleTuple<_UElements ...>() && (! std::_TC<(4u == sizeof... (_UElements)), const float*, const float*, const float*, const float*>::_ImplicitlyConvertibleTuple<_UElements ...>())), bool>::type <anonymous> > std::tuple< <template-parameter-1-1> >::tuple(std: [... cut ...]
         result.f(Y, i) = M[0][1] * x + M[1][1] * y + M[2][1] * z + M[3][1];
                      ^
/home/galex-713/src/ao/kernel/src/eval/evaluator.cpp:1375:22: error: no match for call to ‘(std::vector<std::array<float, 256u> >) (Clause::Id&, size_t&)’
         result.f(Z, i) = M[0][2] * x + M[1][2] * y + M[2][2] * z + M[3][2];
                      ^
[14/54] Building CXX object ui/vendor/efsw/C...s/efsw.dir/src/efsw/FileWatcherInotify.cpp.o
ninja: build stopped: subcommand failed.
galex-713@PC713:~/src/ao/build$ 
```

mkeeter, 2016-11-29:
Ah, your machine is using the non-AVX code, which I haven't built in a while.

I've pushed an update that fixes compilation on my machine – can you see if it helps?

galex-713, 2016-11-29:
It works! :D

galex-713, 2016-11-29:
oh wait who is supposed to close the issue if it is fixed?

mkeeter, 2016-11-29:
I'll do it 😄

closed 2016-11-29 by mkeeter