libfive/libfive #24: Segmentation Fault on Ubuntu 16.04

sene-muelmont, 2016-12-01:
Brief Description
-----------------
Using "master" branch with latest commit aa7c5bf, evaluating "ao-show" on a model causes a segmentation fault (including all the models in the "./examples" directory).  Previous builds from the master branch (circa 10/25/2016) on this system worked correctly.

System Information
------------------
OS: ubuntu 16.04 LTS
Processor: AMD A8-6410 APU with AMD Radeon R5 Graphics × 4 
Graphics: Gallium 0.4 on AMD MULLINS (DRM 2.43.0, LLVM 3.8.0)
OS type: 64-bit
uname -a: Linux echo-one 4.4.0-47-generic #68-Ubuntu SMP Wed Oct 26 19:39:52 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
gcc: version 5.4.0

Minimal example to reproduce bug [1]
--------------------------------

```
  ;; start ao-guile and evaluate:
  (define (sphere r)
      (lambda ( x y z )
          (- (sqrt (+ (* x x) (* y y) (* z z))) r)))
  (define s (sphere 2))
  (ao-show s)
```

Abbreviated gdb Output [2]
----------------------

```
Thread 9 "guile" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffdffff700 (LWP 2698)]
0x00007ffff3276791 in eval_clause_derivs (count=32, odz=0x20e70b0, ody=0x20e40a0, odx=0x2124b90, ov=0x2121b80, 
    bdz=0x20e74b0, bdy=0x20e44a0, bdx=0x2124f90, bv=0x2121f80, adz=0x20e74b0, ady=0x20e44a0, adx=0x2124f90, av=0x2121f80, 
    op=<optimized out>) at ../kernel/src/eval/evaluator.cpp:968
968	                                       _mm256_mul_ps(adx[i], bv[i]));
```

Notes:
-----
[1] Different models can cause segmentation fault in different lines of "evaluator.cpp".
 
[2] gdb commands which might be needed to work-around Guile GC signals:
```
    handle SIGPWR nostop noprint
    handle SIGXCPU nostop noprint
```

mkeeter, 2016-12-01:
Okay, I can reproduce this with the `ao-test` suite on a Ubuntu VM so I've got something to go on.  I also suspect this is the same as what @sethalves is reporting in issue #7.  More updates as the situation develops...

mkeeter, 2016-12-02:
Okay, check out the latest commit in `master` (43720740329ed6d0271ed3d27dc3a1c655e12d96).

With the aligned allocator, I can no longer reproduce the bug on my VM; hopefully you'll see similar success.

sene-muelmont, 2016-12-02:
Yes, this fixed the bug for me too.  All the examples work and ao-test passes.  Thanks!

closed 2016-12-02 by mkeeter