libfive/libfive #27: Graphical Window Resize Broken

sene-muelmont, 2016-12-09:
Brief Description
--
When resizing the graphical output window, the area that displays the model stays fixed in the lower left corner.  Expectation is that the model display area would expand to fit the bounds of the window.  I have only tested it on Ubuntu 16.04.

System Information
--
OS: ubuntu 16.04 LTS
Processor: AMD A8-6410 APU with AMD Radeon R5 Graphics × 4
Graphics: Gallium 0.4 on AMD MULLINS (DRM 2.43.0, LLVM 3.8.0)
OS type: 64-bit
uname -a: Linux echo-one 4.4.0-47-generic #68-Ubuntu SMP Wed Oct 26 19:39:52 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
gcc: version 5.4.0

Minimal Example To Reproduce Bug
--

```
  ;; start ao-guile and evaluate:
  (define (sphere r)
      (lambda ( x y z )
          (- (sqrt (+ (* x x) (* y y) (* z z))) r)))
  (define s (sphere 2))
  (ao-show s)

  ;; Resize graphical output window and model stays in lower left corner.
```

mkeeter, 2016-12-10:
Sadly, I can't debug the UI on Linux myself because my VM doesn't support OpenGL 3...

Can you add a debug `printf` to the Window resize callback [here](https://github.com/mkeeter/ao/blob/master/ui/src/window.cpp#L371-L374) and see if it's being called?

sene-muelmont, 2016-12-10:
I added ``printf`` and verified that the Window ``resized`` was being called.

I'm going to check the graphics driver on my system next.  I thought I would also try some sample code for GLFW to see if the resize is working in a simpler example.

mkeeter, 2016-12-10:
Sounds like a good plan – thanks for being so proactive!

sene-muelmont, 2016-12-11:
I was able to correctly resize the sample programs in the GLFW source on my system--so I ruled out a driver problem.

I'm not sure if this is a fix or a work-around but I got resizing to work in Ao by putting:

```
    glViewport(0, 0, width, height);
```

right before the ``draw()`` call in the ``Window::resized`` member function.

The GLFW samples showed this technique in their resize callback.

mkeeter, 2016-12-11:
Yup, that seems like a correct fix to make (not sure why the viewport auto-resizes on Mac but not on Linux).  Want to open up a PR with the change?

sene-muelmont, 2016-12-11:
Good.  I'll let you apply the change to the repo, if you don't mind.

closed 2016-12-11 by sene-muelmont

mkeeter, 2016-12-11:
Okay, should be all set.