libfive/libfive #28: Build failure on Debian Jessie

technomancy, 2017-06-06:
Here's what I get trying to follow the instructions on my machine. (Note that I installed `libpng12-dev` instead of `libpng16-dev`, which is not available on Debian Stable.)

```
~/src/ao $ mkdir build
~/src/ao $ cd build
~/src/ao/build $ cmake -G Ninja ..
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.8") 
-- Found PNG: /usr/lib/x86_64-linux-gnu/libpng.so (found version "1.2.50") 
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28") 
-- checking for one of the modules 'glfw3'
-- Configuring done
-- Generating done
-- Build files have been written to: /home/phil/src/ao/build
~/src/ao/build $ ninja
[1/69] Building CXX object kernel/CMakeFiles/ao-kernel.dir/src/tree/cache.cpp.o
[2/69] Building CXX object kernel/CMakeFiles/ao-kernel.dir/src/tree/tree.cpp.o
[3/69] Building CXX object kernel/CMakeFiles/ao-kernel.dir/src/tree/opcode.cpp.o
[4/69] Building CXX object kernel/CMakeFiles/ao-kernel.dir/src/format/contours.cpp.o
../kernel/src/tree/tree.cpp:77:36: warning: extra ‘;’ [-Wpedantic]
 OP_UNARY(square,    Opcode::SQUARE);
                                    ^
../kernel/src/tree/tree.cpp:78:34: warning: extra ‘;’ [-Wpedantic]
 OP_UNARY(sqrt,      Opcode::SQRT);
                                  ^
../kernel/src/tree/tree.cpp:80:33: warning: extra ‘;’ [-Wpedantic]
 OP_UNARY(abs,       Opcode::ABS);
                                 ^
../kernel/src/tree/tree.cpp:81:33: warning: extra ‘;’ [-Wpedantic]
 OP_UNARY(sin,       Opcode::SIN);
                                 ^
../kernel/src/tree/tree.cpp:82:33: warning: extra ‘;’ [-Wpedantic]
 OP_UNARY(cos,       Opcode::COS);
                                 ^
../kernel/src/tree/tree.cpp:83:33: warning: extra ‘;’ [-Wpedantic]
 OP_UNARY(tan,       Opcode::TAN);
                                 ^
../kernel/src/tree/tree.cpp:84:34: warning: extra ‘;’ [-Wpedantic]
 OP_UNARY(asin,      Opcode::ASIN);
                                  ^
../kernel/src/tree/tree.cpp:85:34: warning: extra ‘;’ [-Wpedantic]
 OP_UNARY(acos,      Opcode::ACOS);
                                  ^
../kernel/src/tree/tree.cpp:86:34: warning: extra ‘;’ [-Wpedantic]
 OP_UNARY(atan,      Opcode::ATAN);
                                  ^
../kernel/src/tree/tree.cpp:87:33: warning: extra ‘;’ [-Wpedantic]
 OP_UNARY(exp,       Opcode::EXP);
                                 ^
../kernel/src/tree/tree.cpp:92:37: warning: extra ‘;’ [-Wpedantic]
 OP_BINARY(operator+,    Opcode::ADD);
                                     ^
../kernel/src/tree/tree.cpp:93:37: warning: extra ‘;’ [-Wpedantic]
 OP_BINARY(operator*,    Opcode::MUL);
                                     ^
../kernel/src/tree/tree.cpp:94:37: warning: extra ‘;’ [-Wpedantic]
 OP_BINARY(min,          Opcode::MIN);
                                     ^
../kernel/src/tree/tree.cpp:95:37: warning: extra ‘;’ [-Wpedantic]
 OP_BINARY(max,          Opcode::MAX);
                                     ^
../kernel/src/tree/tree.cpp:96:37: warning: extra ‘;’ [-Wpedantic]
 OP_BINARY(operator-,    Opcode::SUB);
                                     ^
../kernel/src/tree/tree.cpp:97:37: warning: extra ‘;’ [-Wpedantic]
 OP_BINARY(operator/,    Opcode::DIV);
                                     ^
../kernel/src/tree/tree.cpp:98:39: warning: extra ‘;’ [-Wpedantic]
 OP_BINARY(atan2,        Opcode::ATAN2);
                                       ^
../kernel/src/tree/tree.cpp:99:37: warning: extra ‘;’ [-Wpedantic]
 OP_BINARY(pow,          Opcode::POW);
                                     ^
../kernel/src/tree/tree.cpp:100:42: warning: extra ‘;’ [-Wpedantic]
 OP_BINARY(nth_root,     Opcode::NTH_ROOT);
                                          ^
../kernel/src/tree/tree.cpp:101:37: warning: extra ‘;’ [-Wpedantic]
 OP_BINARY(mod,          Opcode::MOD);
                                     ^
../kernel/src/tree/tree.cpp:102:41: warning: extra ‘;’ [-Wpedantic]
 OP_BINARY(nanfill,      Opcode::NANFILL);
                                         ^
[5/69] Building CXX object kernel/CMakeFiles/ao-kernel.dir/src/format/mesh.cpp.o
[6/69] Building CXX object kernel/CMakeFiles/ao-kernel.dir/src/format/image.cpp.o
[7/69] Building CXX object kernel/CMakeFiles/ao-kernel.dir/src/render/dc.cpp.o
[8/69] Building CXX object kernel/CMakeFiles/ao-kernel.dir/src/render/heightmap.cpp.o
[9/69] Building CXX object kernel/CMakeFiles/ao-kernel.dir/src/render/quadtree.cpp.o
[10/69] Building CXX object kernel/CMakeFiles/ao-kernel.dir/src/render/subregion.cpp.o
[11/69] Building CXX object kernel/CMakeFiles/ao-kernel.dir/src/render/region.cpp.o
[12/69] Building CXX object kernel/CMakeFiles/ao-kernel.dir/src/render/dc2d.cpp.o
[13/69] Building CXX object kernel/CMakeFiles/ao-kernel.dir/src/render/octree.cpp.o
[14/69] Building CXX object kernel/CMakeFiles/ao-kernel.dir/src/eval/evaluator.cpp.o
[15/69] Building CXX object kernel/CMakeFiles/ao-kernel.dir/src/eval/result.cpp.o
[16/69] Building CXX object kernel/CMakeFiles/ao-kernel.dir/src/solve/solver.cpp.o
FAILED: /usr/bin/c++    -Wall -Wextra -Wno-comment -g -fPIC -pedantic -std=c++11 -Werror=switch -march=native -O3 -DRELEASE -isystem ../vendor -I../kernel/include -MMD -MT kernel/CMakeFiles/ao-kernel.dir/src/eval/evaluator.cpp.o -MF kernel/CMakeFiles/ao-kernel.dir/src/eval/evaluator.cpp.o.d -o kernel/CMakeFiles/ao-kernel.dir/src/eval/evaluator.cpp.o -c ../kernel/src/eval/evaluator.cpp
In file included from /usr/include/boost/numeric/interval/hw_rounding.hpp:15:0,
                 from /usr/include/boost/numeric/interval.hpp:18,
                 from ../kernel/include/ao/kernel/eval/interval.hpp:21,
                 from ../kernel/include/ao/kernel/eval/result.hpp:24,
                 from ../kernel/include/ao/kernel/eval/evaluator.hpp:27,
                 from ../kernel/src/eval/evaluator.cpp:27:
/usr/include/boost/numeric/interval/rounded_arith.hpp: In instantiation of ‘T boost::numeric::interval_lib::rounded_arith_std<T, Rounding>::int_down(const T&) [with T = float; Rounding = boost::numeric::interval_lib::rounding_control<float>]’:
/usr/include/boost/numeric/interval/arith2.hpp:40:49:   required from ‘boost::numeric::interval<T, Policies> boost::numeric::fmod(const boost::numeric::interval<T, Policies>&, const boost::numeric::interval<T, Policies>&) [with T = float; Policies = boost::numeric::interval_lib::policies<boost::numeric::interval_lib::detail::save_state_unprotected<boost::numeric::interval_lib::rounded_transc_std<float> >, boost::numeric::interval_lib::checking_base<float> >]’
/usr/include/boost/numeric/interval/transc.hpp:62:32:   required from ‘boost::numeric::interval<T, Policies> boost::numeric::cos(const boost::numeric::interval<T, Policies>&) [with T = float; Policies = boost::numeric::interval_lib::policies<boost::numeric::interval_lib::save_state<boost::numeric::interval_lib::rounded_transc_std<float> >, boost::numeric::interval_lib::checking_base<float> >]’
../kernel/src/eval/evaluator.cpp:1164:41:   required from here
/usr/include/boost/numeric/interval/rounded_arith.hpp:71:61: error: ‘to_int’ was not declared in this scope, and no declarations were found by argument-dependent lookup at the point of instantiation [-fpermissive]
   T int_down(const T& x) { this->downward(); return to_int(x); }
                                                             ^
/usr/include/boost/numeric/interval/rounded_arith.hpp:71:61: note: declarations in dependent base ‘boost::numeric::interval_lib::detail::c99_rounding_control’ are not found by unqualified lookup
/usr/include/boost/numeric/interval/rounded_arith.hpp:71:61: note: use ‘this->to_int’ instead
ninja: build stopped: subcommand failed.
```

mkeeter, 2017-06-06:
From a [little bit of Googling](https://stackoverflow.com/questions/39355906/boost-interval-arithmetic-and-trigonometric-functions), it looks like this is an issue in `boost::interval`.

Can you upgrade your `boost` installation to version 1.58 or later and see if that fixes the issue?

technomancy, 2017-06-06:
Hm... I have no idea how to safely do that upgrade.

However, the next version of Debian is coming out in a couple weeks, so maybe I'll just wait till that happens and try again then. Thanks.

mkeeter, 2017-09-10:
I just pushed a history-changing update to `master` that changes a bunch of the code and includes a more featureful GUI.  Since so much changed, I'm going to close this issue for now.  If you have trouble building the new `master`, let me know and I'll see if I can help.

closed 2017-09-10 by mkeeter