libfive/libfive #29: ao-guile aborts with error message

doug-moen, 2017-07-14:
On macOS 10.11.6, using homebrew to build ao.
guile version 2.2.2

Sorry to bug you, but I've never dealt with Guile errors before.

```
$ bin/ao-guile 
;;; note: auto-compilation is enabled, set GUILE_AUTO_COMPILE=0
;;;       or pass the --no-auto-compile argument to disable.
;;; compiling /Users/doug/git/ao/bin/ao-guile
;;; compiling /Users/doug/git/ao/bin/../bind/guile/ao/sys/libao.scm
;;; compiling /Users/doug/git/ao/bin/../bind/guile/ao/sys/util.scm
;;; compiled /Users/doug/.cache/guile/ccache/2.2-LE-8-3.A/Users/doug/git/ao/bind/guile/ao/sys/util.scm.go
;;; compiled /Users/doug/.cache/guile/ccache/2.2-LE-8-3.A/Users/doug/git/ao/bind/guile/ao/sys/libao.scm.go
;;; WARNING: compilation of /Users/doug/git/ao/bin/ao-guile failed:
;;; ERROR: In procedure scm_to_utf8_stringn: Wrong type argument in position 1 (expecting string): #f
Backtrace:
          11 (primitive-load "/Users/doug/git/ao/bin/ao-guile")
In ice-9/eval.scm:
   721:20 10 (primitive-eval (use-modules ((ice-9 readline) # (…))))
In ice-9/psyntax.scm:
  1234:36  9 (expand-top-sequence ((use-modules ((ice-9 #) # (…)))) _ …)
  1181:24  8 (parse _ (("placeholder" placeholder)) ((top) #(# # …)) …)
   284:10  7 (parse _ (("placeholder" placeholder)) (()) _ c&e (eval) …)
In ice-9/boot-9.scm:
  3369:20  6 (process-use-modules _)
   230:17  5 (map1 (((ice-9 readline) #:select (write-history # #))))
  3370:31  4 (_ ((ice-9 readline) #:select (write-history # #)))
  2805:10  3 (resolve-interface (ice-9 readline) #:select _ #:hide _ …)
   268:13  2 (for-each #<procedure 1079ee040 at ice-9/boot-9.scm:28…> …)
  2810:38  1 (_ _)
In unknown file:
           0 (scm-error misc-error #f "~A" ("no binding `history-f…") …)

ERROR: In procedure scm-error:
ERROR: no binding `history-file' in module (ice-9 readline)
```

mkeeter, 2017-07-14:
Sadly, it looks like Guile 2.2 has changed a lot of bindings and deprecated a bunch of things that I use.  I'll be updating to 2.2 in the future, but right now am working on a big rewrite – can you try using Guile 2.0 and see if that works?

doug-moen, 2017-07-15:
guile 2.0 works for me

closed 2017-07-15 by doug-moen