libfive/libfive #30: macOS: rotating shape in graphics window

doug-moen, 2017-07-15:
In Antimony, on macOS, I can rotate the shape in the graphics window by holding down the CTRL key. However, this doesn't work in AO. Consider this a feature request.

mkeeter, 2017-07-16:
Right now, the graphics window is [configured](https://github.com/mkeeter/ao/blob/master/ui/src/window.cpp#L156-L163) to pan on left-click and rotate on right-click – are you asking for rotation when you CTRL + left-click?

doug-moen, 2017-07-16:
I'm asking for rotation when you CTRL + left-click. I'm using a Mac laptop without an attached USB mouse. The trackpad doesn't have two physical hardware buttons, but the convention is that holding down CTRL and clicking is right click.

This is already implemented in Antimony, but I just checked the source, and it seems that Qt must be doing that for you.

mkeeter, 2017-07-16:
Okay, this should be fixed in b3ac10d37e8cf.

doug-moen, 2017-07-16:
Works for me. Thanks Matt!

closed 2017-07-16 by mkeeter