libfive/libfive #35: 2D shapes are too fun

sfcgeorge, 2017-09-14:
AO looks _awesome_. I've been wanting an excuse to learn LISP for years and this is it. After a lot of frustration with OpenSCAD I'm more than excited about AO.

Anyway, I thought I'd start with 2D shapes right?

```scm
(rectangle #[0 0] #[10 5])

(circle #[0 0] 5)
```

[image]

I'm not sure... what is happening. I guess this must make sense to the function kernel but it's a bit too "fun" for the average user ;)

sfcgeorge, 2017-09-14:
Once `extrude-z` is used to make them 3D the display is fine.

mkeeter, 2017-09-14:
Yup, that's a little too much "fun".

Meshing assumes that shapes are bounded, and 2D shapes are effectively infinite on the Z axis, so you're seeing meshes that are strange around the edges.

As you've found, clamping in Z works fine, but I'll think a little about how to handle 2D shapes.  Special-casing shapes that only use `X` and `Y` is the easiest strategy, but I'd like to support arbitrary 2D "draw planes" in a sensible way.

sfcgeorge, 2017-09-14:
Yeah, you've got some neat possibilities here so no rush, I'm sure you'll come up with something elegant. I'm still wrapping my head around f-rep but it seems so obvious... too obvious... it's very cool.

sfcgeorge, 2017-09-14:
I'm extruding into 3D for creating models for 3D printing so it doesn't affect me too much. Plus there's no 2D export anyway.

doug-moen, 2018-02-27:
Curv marks some shapes as 2D and other shapes as 3D. 2D shapes are confined to the XY plane. They use a special display mode that is automatically selected based on the shape type. The 2D viewer is just flat: no Z axis and no lighting model. It is is quite nice to use. I think this would be nice to have in Libfive.

As for file export, in Curv you can't export a 2D shape to a 3D format like STL, and you can't export a 3D shape to a 2D format like SVG. It is effectively a type error.

In Curv, you can explicitly convert 3D to 2D using `slice`, and you can convert a 2D shape to 3D using `extrude`. If you use `extrude inf 2Dshape`, then the 2D shape is converted to a prism that extends infinitely along the Z axis, like 2D shapes in libfive. If you use `extrude 0 2Dshape`, then you get a 3D object that has zero height and is confined to the XY plane, and this can be rotated arbitrarily in 3D space. So I guess this kind of object is in a "2D draw plane", but there is nothing special in the system that was added to support this concept. It just naturally arises as a consequence of using F-Rep, and sphere tracing has no trouble rendering these kinds of F-Reps. Libfive cannot currently display a 0 height box (the shape is invisible), but maybe that is fixable? I'm suggesting that you consider extending Libfive so that it can render a larger subset of the existing F-Reps, as opposed to an alternate solution where you expose a new kind of object to the user called a "2D draw plane".