libfive/libfive #36: Difference errors

sfcgeorge, 2017-09-14:
A simple difference doesn't seem to work. I've tried different values to see if it was edges overlapping or something but I can't get it to work. Union / intersection work fine which is odd as I see difference is built from those. 

```lisp
(difference
	(box #[0 0 0] #[10 10 10])
	(box #[5 5 5] #[11 11 11])
)
```

Error:
```
goops-error: No applicable method for #<<generic> min (3)> in call (min #<<tree> 124fe7e80>)

Stack trace:
0: (difference (box (vec3 0 0 0) (vec3 10 10 10)) (box (vec3 5 5 5) (vec3 11 11 11)))
```

mkeeter, 2017-09-14:
Okay, this should be fixed now!  The issue was that I forgot to define `(min (a <tree>))`, i.e. the single-argument version of `min`.

closed 2017-09-14 by mkeeter

sfcgeorge, 2017-09-15:
That was quick :) Not quite right though, the error is fixed but the output is wrong. I also exported the STL to see if it was just the GUI but it appears to be the data.

[image]

mkeeter, 2017-09-15:
Ah, that's because your shape is outside of the default mesh bounding box (which is ±10).  Click on the menu icon (in the top right of the 3D window) to adjust the render settings.

sfcgeorge, 2017-09-15:
Aah of course! I expanded those settings on my real model but forgot when I was doing this minimal reproduction. Perfect now, thank you.