libfive/libfive #37: CSG Crashes on Similar Overlapping Shapes

sfcgeorge, 2017-09-15:
CSG operations (tried: union / difference / intersection) crash when shapes are similar and overlapping with curves.

I've had similar crashes with OpenSCAD/CGAL for years, so sorry if this is a tricky one 🙈  

I was working on the same file yesterday and I don't think it was crashing (just the difference error but intersection / union was fine). So maybe the floating point stuff you added has caused a regression?

Code that crashes:
```lisp
(define z 2.5)
(define r 0.4)

(define a-x 16.0)
(define a-y 11.4)

(define b-x (sqrt (- (* a-x a-x) (* z z))))

(define tilt (asin (/ z a-x)))

; non-rounded works fine (uncomment):
;(define a (rectangle #[0 0] #[a-x a-y]))
;(define b (rectangle #[0 0] #[b-x a-y]))
;(intersection
;  (extrude-z b 0 z)
;  (rotate-y (extrude-z a 0 z) tilt)
;)

; rounded crashes
(define a (rounded-rectangle #[0 0] #[a-x a-y] r))
(define b (rounded-rectangle #[0 0] #[b-x a-y] r))

(intersection
  (extrude-z b 0 z)
  (rotate-y (extrude-z a 0 z) tilt)
)
```

<details>
<summary>Crash log</summary>
<pre>
Process:               Studio [8177]
Path:                  /Users/USER/Documents/*/Studio.app/Contents/MacOS/Studio
Identifier:            com.impraxical.studio
Version:               0.0.0
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Studio [8177]
User ID:               501

Date/Time:             2017-09-15 22:26:31.504 +0100
OS Version:            Mac OS X 10.12.6 (16G29)
Report Version:        12
Anonymous UUID:        48194326-790D-81B6-B9DB-2FA65DA67DAD

Sleep/Wake UUID:       FC42AF8C-1DD3-4531-A70D-70557DE3A0BF

Time Awake Since Boot: 4500 seconds
Time Since Wake:       1000 seconds

System Integrity Protection: enabled

Crashed Thread:        16

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
Assertion failed: (itr == choices.end()), function push, file /Users/sfcgeorge/Documents/Projects/3D/ao/ao/src/eval/evaluator.cpp, line 348.
 

[... 721 lines cut ...]
__LINKEDIT                       120.5M       40 
__TEXT                           269.0M      341 
__UNICODE                          556K        2 
mapped file                      102.9M      203 
shared memory                     16.9M       18 
===========                     =======  ======= 
TOTAL                            932.7M     1256 

Model: MacBookPro11,3, BootROM MBP112.0138.B40, 4 processors, Intel Core i7, 2.5 GHz, 16 GB, SMC 2.19f12
Graphics: Intel Iris Pro, Intel Iris Pro, Built-In
Graphics: NVIDIA GeForce GT 750M, NVIDIA GeForce GT 750M, PCIe, 2048 MB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x134), Broadcom BCM43xx 1.0 (7.21.171.130.1a1)
Bluetooth: Version 5.0.5f1, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0512F, 500.28 GB
USB Device: USB 3.0 Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: Atreus
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.1
</pre>
</details>

mkeeter, 2017-09-15:
Thanks for the sample case!  Can you share your render settings as well?  Using the settings
```
#! RENDER -20 -20 -20 / 40 20 20 / 10 / 8 !#
```

I don't see a crash, but this is definitely a dark corner of the code that needs some cleaning.

sfcgeorge, 2017-09-16:
I'm sorry I must've missed that line when simplifying the example code. Updated above, now reliably crashes as desired.

sfcgeorge, 2017-09-16:
Simplified test case below. Not all corner radiuses crash it

```lisp
#! RENDER -100 -100 -100 / 100 100 100 / 10 / 2 !#

(define x 15)
(define y 10)
(define z 3)

; Fine:
;(define r 1)

; Crashes:
(define r 0.4)

(define a (rounded-rectangle #[0 0] #[x y] r))
(define b (rounded-rectangle #[0 0] #[x y] r))

(difference
  (extrude-z b 0 z)
  (rotate-y (extrude-z a 0 z) 0.3)
)
```

mkeeter, 2017-09-16:
With the fix above (2cc30decc87f073bd1ea02cf7df5a05cc0684f60), both test cases are now working for me.

Stability in the face of challenging shapes is a major goal for the project, so you should feel encouraged to report any similar issues!

closed 2017-09-16 by mkeeter

sfcgeorge, 2017-09-16:
Wow you're on fire, that appears to have worked thank you!

That's good to know. Having a fast & reliable scriptable CAD system like this is my dream. Ao's rendering already seems impressively fast.

I have a large-ish CAD project I was working on in OpenSCAD but it got so slow and crashy—I rewrote it twice trying to work around but it always got painful. So I'm tempted to rewrite it in Ao. Trying some smaller things first.