libfive/libfive #40: 'std:atomic_bool' has not been declared in this scope

nickodell, 2017-09-24:
I'm getting this error during compile:

```
$ make clean ; make all
[  1%] Building CXX object ao/src/CMakeFiles/ao-kernel.dir/eval/evaluator.cpp.o
[  3%] Building CXX object ao/src/CMakeFiles/ao-kernel.dir/eval/result.cpp.o
[  5%] Building CXX object ao/src/CMakeFiles/ao-kernel.dir/eval/feature.cpp.o
[  6%] Building CXX object ao/src/CMakeFiles/ao-kernel.dir/render/discrete/heightmap.cpp.o
[  8%] Building CXX object ao/src/CMakeFiles/ao-kernel.dir/render/discrete/voxels.cpp.o
[ 10%] Building CXX object ao/src/CMakeFiles/ao-kernel.dir/render/brep/xtree.cpp.o
[ 12%] Building CXX object ao/src/CMakeFiles/ao-kernel.dir/render/brep/contours.cpp.o
In file included from /home/nick/ao/ao/src/render/brep/contours.cpp:5:0:
/home/nick/ao/ao/src/../include/ao/render/brep/xtree.hpp:34:18: error: ‘std::atomic_bool’ has not been declared
             std::atomic_bool& cancel);
                  ^~~~~~~~~~~
/home/nick/ao/ao/src/../include/ao/render/brep/xtree.hpp:44:18: error: ‘std::atomic_bool’ has not been declared
             std::atomic_bool& cancel);
                  ^~~~~~~~~~~
/home/nick/ao/ao/src/../include/ao/render/brep/xtree.hpp:155:16: error: ‘std::atomic_bool’ has not been declared
           std::atomic_bool& cancel);
                ^~~~~~~~~~~
ao/src/CMakeFiles/ao-kernel.dir/build.make:206: recipe for target 'ao/src/CMakeFiles/ao-kernel.dir/render/brep/contours.cpp.o' failed
make[2]: *** [ao/src/CMakeFiles/ao-kernel.dir/render/brep/contours.cpp.o] Error 1
CMakeFiles/Makefile2:109: recipe for target 'ao/src/CMakeFiles/ao-kernel.dir/all' failed
make[1]: *** [ao/src/CMakeFiles/ao-kernel.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

$ git rev-parse --short HEAD
5377b53

```

mkeeter, 2017-09-24:
Okay, see if 317d976dbd5ab8a78cba74c92f4cf520e3cb7f70 fixes this issue.

nickodell, 2017-09-24:
I made that change, plus a few others, and I fixed most of the compile errors. (See diff at end of post.) I'm getting some more errors:

```
$ make clean ; make
[  1%] Building CXX object ao/src/CMakeFiles/ao-kernel.dir/eval/evaluator.cpp.o
[  3%] Building CXX object ao/src/CMakeFiles/ao-kernel.dir/eval/result.cpp.o
[  5%] Building CXX object ao/src/CMakeFiles/ao-kernel.dir/eval/feature.cpp.o
[  6%] Building CXX object ao/src/CMakeFiles/ao-kernel.dir/render/discrete/heightmap.cpp.o
[  8%] Building CXX object ao/src/CMakeFiles/ao-kernel.dir/render/discrete/voxels.cpp.o
[ 10%] Building CXX object ao/src/CMakeFiles/ao-kernel.dir/render/brep/xtree.cpp.o
[ 12%] Building CXX object ao/src/CMakeFiles/ao-kernel.dir/render/brep/contours.cpp.o
[ 13%] Building CXX object ao/src/CMakeFiles/ao-kernel.dir/render/brep/mesh.cpp.o
[ 15%] Building CXX object ao/src/CMakeFiles/ao-kernel.dir/render/brep/marching.cpp.o
[ 17%] Building CXX object ao/src/CMakeFiles/ao-kernel.dir/solve/solver.cpp.o
[ 18%] Building CXX object ao/src/CMakeFiles/ao-kernel.dir/tree/cache.cpp.o
[ 20%] Building CXX object ao/src/CMakeFiles/ao-kernel.dir/tree/opcode.cpp.o
[ 22%] Building CXX object ao/src/CMakeFiles/ao-kernel.dir/tree/template.cpp.o
[ 24%] Building CXX object ao/src/CMakeFiles/ao-kernel.dir/tree/tree.cpp.o
[ 25%] Building CXX object ao/src/CMakeFiles/ao-kernel.dir/ao.cpp.o
[ 27%] Linking CXX shared library libao-kernel.so
[ 27%] Built target ao-kernel
[ 29%] Generating bundle.cpp
Scanning dependencies of target guile-ao
[ 31%] Building CXX object ao/bind/CMakeFiles/guile-ao.dir/bundle.cpp.o
[ 32%] Linking CXX shared library libguile-ao.so
[ 32%] Built target guile-ao
[ 34%] Building CXX object ao/test/CMakeFiles/ao-test.dir/main.cpp.o
[ 36%] Building CXX object ao/test/CMakeFiles/ao-test.dir/api.cpp.o
[ 37%] Building CXX object ao/test/CMakeFiles/ao-test.dir/cache.cpp.o
[ 39%] Building CXX object ao/test/CMakeFiles/ao-test.dir/contours.cpp.o
[ 41%] Building CXX object ao/test/CMakeFiles/ao-test.dir/dual.cpp.o
[ 43%] Building CXX object ao/test/CMakeFiles/ao-test.dir/eval.cpp.o
[ 44%] Building CXX object ao/test/CMakeFiles/ao-test.dir/heightmap.cpp.o
[ 46%] Building CXX object ao/test/CMakeFiles/ao-test.dir/marching.cpp.o
[ 48%] Building CXX object ao/test/CMakeFiles/ao-test.dir/mesh.cpp.o
[ 50%] Building CXX object ao/test/CMakeFiles/ao-test.dir/feature.cpp.o
[ 51%] Building CXX object ao/test/CMakeFiles/ao-test.dir/solver.cpp.o
[ 53%] Building CXX object ao/test/CMakeFiles/ao-test.dir/region.cpp.o
[ 55%] Building CXX object ao/test/CMakeFiles/ao-test.dir/template.cpp.o
[ 56%] Building CXX object ao/test/CMakeFiles/ao-test.dir/tree.cpp.o
[ 58%] Building CXX object ao/test/CMakeFiles/ao-test.dir/voxels.cpp.o
[ 60%] Building CXX object ao/test/CMakeFiles/ao-test.dir/xtree.cpp.o
[ 62%] Building CXX object ao/test/CMakeFiles/ao-test.dir/util/shapes.cpp.o
[ 63%] Building CXX object ao/test/CMakeFiles/ao-test.dir/guile.cpp.o
[ 65%] Linking CXX executable ao-test
[ 65%] Built target ao-test
[ 67%] Automatic moc, uic and rcc for target Studio
Generating moc compilation Studio_automoc.cpp
Generating qrc source Studio_automoc.dir/qrc_resources_HAS2BXGMQQBHVI.cpp
[ 67%] Built target Studio_automoc
Scanning dependencies of target Studio
[ 68%] Building CXX object gui/CMakeFiles/Studio.dir/src/main.cpp.o
[ 70%] Building CXX object gui/CMakeFiles/Studio.dir/src/app.cpp.o
[ 72%] Building CXX object gui/CMakeFiles/Studio.dir/src/axes.cpp.o
[ 74%] Building CXX object gui/CMakeFiles/Studio.dir/src/background.cpp.o
[ 75%] Building CXX object gui/CMakeFiles/Studio.dir/src/bars.cpp.o
[ 77%] Building CXX object gui/CMakeFiles/Studio.dir/src/busy.cpp.o
[ 79%] Building CXX object gui/CMakeFiles/Studio.dir/src/camera.cpp.o
[ 81%] Building CXX object gui/CMakeFiles/Studio.dir/src/settings.cpp.o
[ 82%] Building CXX object gui/CMakeFiles/Studio.dir/src/shader.cpp.o
[ 84%] Building CXX object gui/CMakeFiles/Studio.dir/src/syntax.cpp.o
[ 86%] Building CXX object gui/CMakeFiles/Studio.dir/src/shape.cpp.o
[ 87%] Building CXX object gui/CMakeFiles/Studio.dir/src/color.cpp.o
[ 89%] Building CXX object gui/CMakeFiles/Studio.dir/src/editor.cpp.o
[ 91%] Building CXX object gui/CMakeFiles/Studio.dir/src/interpreter.cpp.o
[ 93%] Building CXX object gui/CMakeFiles/Studio.dir/src/view.cpp.o
[ 94%] Building CXX object gui/CMakeFiles/Studio.dir/src/window.cpp.o
[ 96%] Building CXX object gui/CMakeFiles/Studio.dir/Studio_automoc.cpp.o
[ 98%] Building CXX object gui/CMakeFiles/Studio.dir/Studio_automoc.dir/qrc_resources_HAS2BXGMQQBHVI.cpp.o
[100%] Linking CXX executable Studio
[... 80 lines cut ...]
-#include "ao/render/region.hpp"
+#include "ao/render/brep/region.hpp"
 #include "ao/eval/evaluator.hpp"
 
 #include "util/shapes.hpp"
@@ -14,7 +14,7 @@ TEST_CASE("Small sphere mesh")
 {
     Tree t = sphere(0.5);
 
-    Region r({-1, 1}, {-1, 1}, {-1, 1}, 1);
+    Region<3> r({-1, 1}, {-1, 1}, {-1, 1}, 1);
 
     auto m = Mesh::render(t, r);
 
@@ -25,7 +25,7 @@ TEST_CASE("Face normals")
 {
     Tree axis[3] = {Tree::X(), Tree::Y(), Tree::Z()};
     Eigen::Vector3f norm[3] = {{1, 0, 0}, {0, 1, 0}, {0, 0, 1}};
-    Region r({-1, 1}, {-1, 1}, {-1, 1}, 2);
+    Region<3> r({-1, 1}, {-1, 1}, {-1, 1}, 2);
 
     SECTION("Positive")
     {

```

mkeeter, 2017-09-24:
I've deleted `dc.cpp` altogether (it was a left-over, and wasn't being compiled) and brought in the other `#include` changes that you've made.

I think the linker error is because of a mistake in the Linux build command, which may be fixed in  eb4beb5fccbed19572b25550e89e98a8f32545f4.  Try this out; if it fixes your build, we can figure out which of the shotgun changes are actually necessary.

nickodell, 2017-09-24:
[It works!](https://i.imgur.com/agUjHVy.png)

I tried the current master, and it works after I make one change:

```$ git diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index cb50515..46c0116 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -9,6 +9,9 @@ set(CMAKE_MODULE_PATH ${PROJECT_SOURCE_DIR}/cmake)
 
 ################################################################################
 
+# link pthread 
+set(CMAKE_EXE_LINKER_FLAGS  "${CMAKE_EXE_LINKE_FLAGS} -lpthread")
+
 set(CMAKE_CXX_FLAGS "-Wall -Wextra -g -fPIC -pedantic -Werror=switch -march=native")
 set(CMAKE_CXX_FLAGS_RELEASE  "-O3 -DRELEASE -DEIGEN_NO_DEBUG")
 set(CMAKE_CXX_FLAGS_DEBUG    "-O0")
```

This change is needed for me to build it - taking it out causes the error "undefined reference to symbol 'pthread_create@@GLIBC_2.2.5'"

closed 2017-09-24 by nickodell