libfive/libfive #41: Crash on weird shape

nickodell, 2017-09-24:
This works, and renders a cylinder:

```
(extrude-z
(lambda-shape (x y z)
 (define r (sqrt (+ (expt x 2) (expt y 2))))
 (define theta (atan (/ y x)))
 (- r (+ 2)))
0 1)
```

This crashes:

```
(extrude-z
(lambda-shape (x y z)
 (define r (sqrt (+ (expt x 2) (expt y 2))))
 (define theta (atan (/ y x)))
 (- r (+ 2 theta)))
0 1)
```

It gives this error:

```
Studio: /home/nick/ao/ao/src/render/brep/mesh.cpp:39: void Kernel::Mesh::load(const std::array<const Kernel::XTree<3u>*, 4ul>&) [with Kernel::Axis::Axis A = (Kernel::Axis::Axis)4u; bool D = false]: Assertion `vi != -1' failed.
Aborted
```

I can sometimes get this to run and produce a shape, though it isn't solid - which might be because theta isn't defined for x=0. Is there a better way to define an object in polar coordinates?

(Lemme know if the problem is that I'm using ao in an unsupported way.)

mkeeter, 2017-09-24:
I think you're right and that it's failing when `x=0`.  It's a bit rude to put divisions by zero into the meshing system, but the kernel should be robust enough to not crash, so I'll look into this.

As a workaround, when I switch to the two-argument form `(atan y x)` it renders... something.

mkeeter, 2017-09-25:
Okay, I think I've tracked this down and fixed it.  The shape is still odd-looking, but I think that's because it's a weird shape with discontinuities where `atan` wraps.

Please re-open if you see more crashing.

closed 2017-09-25 by mkeeter

nickodell, 2017-09-25:
Yep, this is fixed.