libfive/libfive #42: Crash on startup

nickodell, 2017-09-25:
About half the time, ao crashes on startup. There are many different kinds of errors that it produces to the terminal when this happens. I ran it 2000 times, and it successfully started 994 times.

Here is a table of outcomes after being run 2000 times:

```
success: 994
fail: 1006
  bad_alloc: 830
  segfault: 150
  sysmalloc: 14
  fp: 1
  unknown failure reason: 11
```

Here is a representative sample of each kind of error: (I have silenced the setWindowModified error in each of them.)

bad_alloc:

```
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
```

segfault:

```
./test.sh: line 2: 18122 Terminated              ( ./gui/Studio 2>&1 )
./test.sh: line 2: 18187 Segmentation fault      ( ./gui/Studio 2>&1 )
```

sysmalloc:

```
Studio: malloc.c:2406: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.
```

fp:

```
./test.sh: line 2: 20081 Floating point exception( ./gui/Studio 2>&1 )
```

(`test.sh` is the script I'm using to run it 2000 times and collect stdout/stderr. The failure rate is pretty similar when Studio is run directly from terminal.)

mkeeter, 2017-09-26:
That's pretty strange.  This is running Studio, not the `ao-test` suite, right?
Can you run it in an debugger to get a more precise picture of what's going wrong?

nickodell, 2017-09-26:
This is running in Studio.

I'll use a debugger and get back to you.

nickodell, 2017-09-26:
The bad_alloc error is failing on the line `a.exec();` in `gui/src/main.cpp`.

I captured the state of the variable a on a failed run and a successful run.

successful run:

```
Thread 1 "Studio" hit Breakpoint 2, main (argc=1, argv=0x7fffffffdfb8)
    at /home/nick/ao/gui/src/main.cpp:51
51	    a.exec();
(gdb) print a
$3 = {<QApplication> = {<No data fields>}, static staticMetaObject = {d = {
      superdata = 0x7ffff62c41a0 <QApplication::staticMetaObject>, 
      stringdata = 0x555555612520 <qt_meta_stringdata_App>, 
      data = 0x555555612540 <qt_meta_data_App>, 
      static_metacall = 0x55555560b304 <App::qt_static_metacall(QObject*, QMetaObject::Call, int, void**)>, relatedMetaObjects = 0x0, extradata = 0x0}}, 
  window = {<QMainWindow> = {<No data fields>}, static staticMetaObject = {
      d = {superdata = 0x7ffff62cea60 <QMainWindow::staticMetaObject>, 
        stringdata = 0x5555556141a0 <qt_meta_stringdata_Window>, 
        data = 0x5555556142c0 <qt_meta_data_Window>, 
        static_metacall = 0x55555560e672 <Window::qt_static_metacall(QObject*, QMetaObject::Call, int, void**)>, relatedMetaObjects = 0x0, extradata = 0x0}}, 
    filename = {static null = {<No data fields>}, d = 0x5555559a8840}, 
    export_filename = {static null = {<No data fields>}, 
      d = 0x7ffff55d0ee0 <QArrayData::shared_null>}, editor = 0x55555593cb60, 
    view = 0x5555559520c0, closing = false}}
```

failed run:

```
Thread 1 "Studio" hit Breakpoint 2, main (argc=1, argv=0x7fffffffdfb8)
    at /home/nick/ao/gui/src/main.cpp:51
51	    a.exec();
(gdb) print a
$4 = {<QApplication> = {<No data fields>}, static staticMetaObject = {d = {
      superdata = 0x7ffff62c41a0 <QApplication::staticMetaObject>, 
      stringdata = 0x555555612520 <qt_meta_stringdata_App>, 
      data = 0x555555612540 <qt_meta_data_App>, 
      static_metacall = 0x55555560b304 <App::qt_static_metacall(QObject*, QMetaObject::Call, int, void**)>, relatedMetaObjects = 0x0, extradata = 0x0}}, 
  window = {<QMainWindow> = {<No data fields>}, static staticMetaObject = {
      d = {superdata = 0x7ffff62cea60 <QMainWindow::staticMetaObject>, 
        stringdata = 0x5555556141a0 <qt_meta_stringdata_Window>, 
        data = 0x5555556142c0 <qt_meta_data_Window>, 
        static_metacall = 0x55555560e672 <Window::qt_static_metacall(QObject*, QMetaObject::Call, int, void**)>, relatedMetaObjects = 0x0, extradata = 0x0}}, 
    filename = {static null = {<No data fields>}, d = 0x555555a85fb0}, 
    export_filename = {static null = {<No data fields>}, 
      d = 0x7ffff55d0ee0 <QArrayData::shared_null>}, editor = 0x55555598cbd0, 
    view = 0x555555a30a00, closing = false}}
(gdb) s
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
```

mkeeter, 2017-09-26:
Okay, you can ignore the first breakpoint – I think `gdb` drops a breakpoint on main so you can watch the program start up.  Can you get a backtrace (`bt`) on the `bad_alloc`, or put a breakpoint on `__cxa_throw` to see exactly where it happens?

nickodell, 2017-09-26:
__cxa_throw backtrace:

```
#0  0x00007ffff5c42260 in __cxa_throw ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007ffff5fc59b2 in qBadAlloc() ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#2  0x00007ffff6043bba in QListData::detach(int) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#3  0x00007ffff5fef4de in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#4  0x00007ffff61b6d2f in QCoreApplication::arguments() ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#5  0x00007ffff7ea2eb7 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#6  0x00007ffff7ea3a04 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#7  0x00007fffed811d90 in _SmcProcessMessage ()
   from /usr/lib/x86_64-linux-gnu/libSM.so.6
#8  0x00007fffed6012a8 in IceProcessMessages ()
   from /usr/lib/x86_64-linux-gnu/libICE.so.6
#9  0x00007ffff61df5e9 in QMetaObject::activate(QObject*, int, int, void**) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#10 0x00007ffff62595ce in QSocketNotifier::activated(int, QSocketNotifier::QPrivateSignal) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#11 0x00007ffff61ebba2 in QSocketNotifier::event(QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#12 0x00007ffff6a96b8c in QApplicationPrivate::notify_helper(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#13 0x00007ffff6a9e341 in QApplication::notify(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#14 0x00007ffff61b39e0 in QCoreApplication::notifyInternal2(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#15 0x00007ffff62084fd in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#16 0x00007ffff2c2b7f7 in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#17 0x00007ffff2c2ba60 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#18 0x00007ffff2c2bb0c in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#19 0x00007ffff620804f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#20 0x00007ffff61b19ca in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#21 0x00007ffff61ba13c in QCoreApplication::exec() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#22 0x000055555556e7ba in main (argc=1, argv=0x7fffffffdfb8) at /home/nick/ao/gui/src/main.cpp:28

```

mkeeter, 2017-09-26:
This reminds me of [this bug](https://github.com/mkeeter/antimony/commit/e0b7d290d0e97adfe35202b5e5931e74212155e9); see if 917223eb7206ab018643e39308f34a93b13e3f00 fixes the issue.

nickodell, 2017-09-27:
That's looking really good. Looks like you squashed the bad_alloc bug entirely. :D After that revision, I have a 94% success rate in starting the program.

rev effd24e:

```
success: 448
fail:
  bad_alloc: 454
  segfault: 85
  sysmalloc: 10
  unknown: 3
```

rev 917223e:

```
success: 943
fail:
  segfault:43
  sysmalloc: 10
  fp: 3
  unknown: 1
```

nickodell, 2017-09-27:
On the subject of ao-test, I have 2 failures. Maybe they're related to the remaining problems:

```

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
ao-test is a Catch v1.1 b3 host application.
Run with -? for options

-------------------------------------------------------------------------------
Heightmap::render: Performance
-------------------------------------------------------------------------------
/home/nick/ao/ao/test/heightmap.cpp:251
...............................................................................

/home/nick/ao/ao/test/heightmap.cpp:300: 
warning:
  Rendered sphere in 0.062705 sec
  Rendered sponge in 0.177429 sec

-------------------------------------------------------------------------------
Mesh::render (performance)
-------------------------------------------------------------------------------
/home/nick/ao/ao/test/mesh.cpp:54
...............................................................................

/home/nick/ao/ao/test/mesh.cpp:75: 
warning:
  
  Made sponge mesh in 2.001005 sec

-------------------------------------------------------------------------------
Feature::push
  Separability testing
-------------------------------------------------------------------------------
/home/nick/ao/ao/test/feature.cpp:7
...............................................................................

/home/nick/ao/ao/test/feature.cpp:27: FAILED:
  REQUIRE( a.push({-1, -1, -1}) == false )
with expansion:
  true == false

-------------------------------------------------------------------------------
Solver::findRoot
  Sum-of-squares performance
-------------------------------------------------------------------------------
/home/nick/ao/ao/test/solver.cpp:8
...............................................................................

/home/nick/ao/ao/test/solver.cpp:111: 
warning:
  
  Solved SOS root in 0.000129 sec

/home/nick/ao/ao/test/solver.cpp:115: FAILED:
  REQUIRE( res < 1e-3 )
with expansion:
  0.00206f < 0.001

===============================================================================
test cases:   87 |   85 passed | 2 failed
assertions: 4824 | 4822 passed | 2 failed
```

mkeeter, 2017-09-27:
Those test failures *may* be relevant.  One way to check is opening with an empty file (as the first command line argument, e.g. `./gui/studio empty.ao`) so that it doesn't start meshing on startup, and see if the crashes continue.

The last backtrace was helpful – can you get one for the new, rarer crashes on startup?

nickodell, 2017-09-27:
I'll try to get a backtrace of the other errors. I'll need to figure out how to script gdb first - I don't want to manually run ao 1000 times.

Here's the  result of running ao on an empty file 1000 times:

    $ ./parse_test.py ../mine/aoerrors_nomesh.txt 
    defaultdict(<class 'int'>, {'segfault': 39, 'unknown': 4, 'success': 944, 'sysmalloc': 13})

Looks pretty similar.

nickodell, 2017-09-27:
Here's a sysmalloc trace:

```
Studio: malloc.c:2406: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.
    assertion=assertion@entry=0x7ffff545f088 "(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)", 
    file=file@entry=0x7ffff545b867 "malloc.c", line=line@entry=2406, 
    function=function@entry=0x7ffff545f8d0 <__func__.11275> "sysmalloc")
    at malloc.c:301

Thread 1 "Studio" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff532d3fa in __GI_abort () at abort.c:89
#2  0x00007ffff536f9c8 in __malloc_assert (
#3  0x00007ffff5371546 in sysmalloc (nb=nb@entry=112, 
    av=0x7ffff5692b00 <main_arena>) at malloc.c:2403
#4  0x00007ffff537242d in _int_malloc (
    av=av@entry=0x7ffff5692b00 <main_arena>, bytes=bytes@entry=96)
    at malloc.c:3865
#5  0x00007ffff5374b4b in __libc_calloc (n=<optimized out>, 
    elem_size=<optimized out>) at malloc.c:3274
#6  0x00007fffcb9cb022 in ?? () from /usr/lib/x86_64-linux-gnu/dri/i965_dri.so
#7  0x00007fffcb95586a in ?? () from /usr/lib/x86_64-linux-gnu/dri/i965_dri.so
#8  0x00007fffcb94d2e1 in ?? () from /usr/lib/x86_64-linux-gnu/dri/i965_dri.so
#9  0x00007fffcb94f998 in ?? () from /usr/lib/x86_64-linux-gnu/dri/i965_dri.so
#10 0x00007fffcb957da5 in ?? () from /usr/lib/x86_64-linux-gnu/dri/i965_dri.so
#11 0x00007fffcb95b2e9 in ?? () from /usr/lib/x86_64-linux-gnu/dri/i965_dri.so
#12 0x00007fffcb957bdf in ?? () from /usr/lib/x86_64-linux-gnu/dri/i965_dri.so
#13 0x00007fffcb95dd2a in ?? () from /usr/lib/x86_64-linux-gnu/dri/i965_dri.so
#14 0x00007fffcb954b68 in ?? () from /usr/lib/x86_64-linux-gnu/dri/i965_dri.so
#15 0x00007fffcb9aaf51 in ?? () from /usr/lib/x86_64-linux-gnu/dri/i965_dri.so
#16 0x00007fffcb844b4e in ?? () from /usr/lib/x86_64-linux-gnu/dri/i965_dri.so
#17 0x00007ffff67f1aa1 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#18 0x00007ffff67f2819 in QOpenGLShader::compileSourceCode(char const*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#19 0x00007ffff67f825a in QOpenGLShader::compileSourceFile(QString const&) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#20 0x00007ffff67f8443 in QOpenGLShaderProgram::addShaderFromSourceFile(QFlags<QOpenGLShader::ShaderTypeBit>, QString const&) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#21 0x000055555557b0a7 in Shader::initializeGL ()
    at /home/nick/ao/gui/src/shader.cpp:33
#22 0x0000555555593ee2 in View::initializeGL (this=0x555555ab4800)
    at /home/nick/ao/gui/src/view.cpp:150
#23 0x00007ffff6afd71d in ?? ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#24 0x00007ffff6afd84a in QOpenGLWidget::resizeEvent(QResizeEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#25 0x00007ffff6ade6ce in QWidget::event(QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#26 0x00007ffff6a96b8c in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#27 0x00007ffff6a9e341 in QApplication::notify(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#28 0x00007ffff61b39e0 in QCoreApplication::notifyInternal2(QObject*, QEvent*)
    () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#29 0x00007ffff6ad6f72 in QWidgetPrivate::sendPendingMoveAndResizeEvents(bool, bool) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#30 0x00007ffff6adae23 in QWidgetPrivate::show_helper() ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#31 0x00007ffff6addc75 in QWidget::setVisible(bool) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#32 0x00007ffff6adadb8 in QWidgetPrivate::showChildren(bool) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#33 0x00007ffff6adae3f in QWidgetPrivate::show_helper() ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#34 0x00007ffff6addc75 in QWidget::setVisible(bool) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#35 0x00007ffff6adadb8 in QWidgetPrivate::showChildren(bool) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#36 0x00007ffff6adae3f in QWidgetPrivate::show_helper() ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#37 0x00007ffff6addc75 in QWidget::setVisible(bool) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#38 0x000055555559c49d in Window::Window (this=0x7fffffffde30, target=...)
    at /home/nick/ao/gui/src/window.cpp:148
#39 0x0000555555571306 in App::App (this=0x7fffffffde20, argc=<optimized out>, 
    argv=0x7fffffffdfa8) at /home/nick/ao/gui/src/app.cpp:8
#40 0x000055555556e7f8 in main (argc=<optimized out>, argv=0x7fffffffdfa8)
    at /home/nick/ao/gui/src/main.cpp:27
```