libfive/libfive #43: Consistent crash on startup

omgitsraven, 2017-10-02:
I'm on Mac OS 10.12.6, and I'm trying to run the ao-guile-repl branch. (I know that branch is marked as deprecated, but the master branch is marked as not ready yet, so I figured it was the best bet... but if I just shouldn't be trying to use that older branch, then please let me know!)

I got everything installed & compiled through brew & git, but when I actually double-click ao-guile, I just get the following message in a Terminal window every time:

```
Andrews-MacBook-Air:~ andrew$ /Users/andrew/ao/bin/ao-guile ; exit;
;;; note: auto-compilation is enabled, set GUILE_AUTO_COMPILE=0
;;;       or pass the --no-auto-compile argument to disable.
;;; compiling /Users/andrew/ao/bin/ao-guile
;;; WARNING: compilation of /Users/andrew/ao/bin/ao-guile failed:
;;; ERROR: In procedure scm_to_utf8_stringn: Wrong type argument in position 1 (expecting string): #f
Backtrace:
          11 (primitive-load "/Users/andrew/ao/bin/ao-guile")
In ice-9/eval.scm:
   721:20 10 (primitive-eval (use-modules ((ice-9 readline) # (…))))
In ice-9/psyntax.scm:
  1234:36  9 (expand-top-sequence ((use-modules ((ice-9 #) # (…)))) _ …)
  1181:24  8 (parse _ (("placeholder" placeholder)) ((top) #(# # …)) …)
   284:10  7 (parse _ (("placeholder" placeholder)) (()) _ c&e (eval) …)
In ice-9/boot-9.scm:
  3369:20  6 (process-use-modules _)
   230:17  5 (map1 (((ice-9 readline) #:select (write-history # #))))
  3370:31  4 (_ ((ice-9 readline) #:select (write-history # #)))
  2805:10  3 (resolve-interface (ice-9 readline) #:select _ #:hide _ …)
   268:13  2 (for-each #<procedure 1103f7980 at ice-9/boot-9.scm:28…> …)
  2810:38  1 (_ _)
In unknown file:
           0 (scm-error misc-error #f "~A" ("no binding `history-f…") …)

ERROR: In procedure scm-error:
ERROR: no binding `history-file' in module (ice-9 readline)
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[Process completed]
```

mkeeter, 2017-10-03:
I'd definitely recommend switching to `master`:
- That's where new development is happening
- It's got a user-friendlier GUI (rather than the bare REPL)
- The kernel has a lot more robustification

The warning in the README is intended to discourage folks from posting to Reddit / HN / other places that will bring a huge influx of traffic, so I'll clarify the wording.

mkeeter, 2017-10-20:
As evidenced by all the issues you've opened, you're clearly on `master` now, so I'll close this.
Thanks for all the feedback so far!

closed 2017-10-20 by mkeeter