libfive/libfive #44: Can't use lambda-shape arguments in comparison function

omgitsraven, 2017-10-05:
I'm extremely new to Scheme, so it's possible this is a silly mistake on my end, but I'm having an issue that works in the general case but not inside of a lambda-shape:

`(lambda-shape (x y z) (if (> x 0) x 0) )`
gives an error that says
```
wrong-type-arg: Wrong type: #<<tree> 139e64140>

Stack trace:
0: (lambda-shape (x y z) (if (> x 0) x 0))
```

even though
`(let ((x 1)) (if (> x 0) x 0) )`
works fine, so it isn't some kind of syntax thing; it seems like for some operations, the `x y z` arguments aren't treated like numbers? (even though for functions like `+` it works fine...)

mkeeter, 2017-10-05:
`lambda-shape` isn't powerful enough to evaluate arbitrary Scheme code; it's intended to JIT a  (basically) arithmetic expression.  In this case, the issue is that `lambda-shape` doesn't accept conditionals, since the branches are evaluated lazily rather than eagerly.

There's a little more theory about the JITting [here](http://www.mattkeeter.com/blog/2016-02-21-jit/).

In this case, you can use `(lambda-shape (x y z) (max x 0))` as an alternative.

In general, the functions that `lambda-shape` will accept are defined in `ao-guile.cpp` [here](https://github.com/mkeeter/ao/blob/master/ao/bind/ao-guile.cpp#L185).

mkeeter, 2018-02-24:
This came up again [recently](https://github.com/libfive/libfive/issues/138), and now both gives a better error message and is documented [here](https://github.com/libfive/libfive/blob/master/doc/guide.md#lambda-shape-and-define-shape), so I'm going to close this issue.

closed 2018-02-24 by mkeeter