libfive/libfive #45: Possibility of assigning normals based on distance function?

omgitsraven, 2017-10-05:
Would it be possible for Ao to sample the distance field to generate smoothed normals for the mesh?

When I was messing around with marching tetrahedra, I was able to (in some cases) greatly improve the visual smoothness of the meshes by assigning normals to the vertices based on three axis-aligned samples from the distance function, the way it's usually done in raymarching: http://jamie-wong.com/2016/07/15/ray-marching-signed-distance-functions/#surface-normals-and-lighting

No normal smoothing:
[image]

Typical mesh normal-averaging:
[image]

Normals generated by sampling the distance function:
[image]

Obviously this approach has a lot of compromises and tradeoffs—I'm not sure whether Dual Contouring provides any hint as to whether an edge is meant to be sharp or smooth, for one thing—but it's a drastic enough improvement when it works that I think it's worth considering offering as an option, and it's specific enough to the same distance field that was used to generate the mesh that it's really something that would have to be done by the same program to be done at all, unlike traditional normal averaging. (Users could still manually re-sharpen the sharp edges in another program afterwards though, for example.)

mkeeter, 2017-10-06:
That's an interesting suggestion.  If I understand this correctly, you're not actually changing the mesh itself, just the normals for rendering (to give a smoothing effect on surfaces).

I think the big challenge in blending this with dual contouring is that with DC, points end up snapped to sharp features in the object, so they have the equivalent of multiple normals (i.e. a cube corner would have three possible normals, because the gradient is discontinuous).

An easier solution could be to use the existing flat shading for triangles with multi-normal points, then smoothed shading of some kind for other vertices / triangles.  This seems like a bit of a project to implement, but I'll keep it in mind next time I'm digging into the rendering code.

omgitsraven, 2017-10-06:
That's right, the mesh is the same, this is just changing the normals to disguise the faceting.

If you can actually distinguish between multi-normal points and ones that are meant to be smooth, then that sounds perfect, yeah -- I guess the only issue is how to handle vertices that are sharp on one edge, but curved on another... maybe you could just copy the normals of the next vertex over, in the direction away from the edge... (at a high enough resolution, the difference should be minor?)

But honestly, even if the first version of this feature just always sampled the field, even if that meant ruining sharp edges, it could still be useful -- anyone making a shape that DOESN'T include sharp edges would still be able to use it (and like I mentioned, people could probably at least clean up any sharp edges in another program after export).

Anyway, yeah, I can understand if this isn't an immediate priority, but I thought I'd bring it up as a possibility :) Thanks!

mkeeter, 2017-12-28:
I started working on this in [this branch](https://github.com/mkeeter/ao/tree/smooth-rendering-again), figured some [really clever math](http://www.mattkeeter.com/blog/2017-11-22-eigen/) to automatically pull normals from details of the QEF, and then ran into the issue that the meshes aren't quite good enough – there are occasional overlapping triangles, which look particularly bad with smooth rendering.

That's good information to have, but it means that I've got more kernel work to do before smooth rendering will be feasible!

omgitsraven, 2017-12-28:
Fair enough! Happy to hear it's lead to some fruitful research at least :)

kayarre, 2020-06-05:
Is there any progress on this front in terms of sampling the surface normal from the distance field?

mkeeter, 2020-06-06:
Sampling the surface normals from the distance field is already implemented – [`DerivArrayEvaluator::deriv`](https://github.com/libfive/libfive/blob/master/libfive/include/libfive/eval/eval_deriv_array.hpp#L51) will tell you the partial derivatives w.r.t x/y/z at any point, which is the same as the normal near the surface of the model.

This issue was asking about an alternate shading mode, where we do smooth rendering rather than per-triangle flat rendering.

lf94, 2023-10-04:
@omgitsraven

> Normals generated by sampling the distance function:

How is this done? All I can think of is you get to the boundary (0), and then step back somehow.