libfive/libfive #46: Y-up

omgitsraven, 2017-10-05:
Is there any hope of adding a Y-up option for the display in Studio :) It would mainly just impact the way the mouse orbit controls work; when I rotate the shape so that Y is up, I'm not able to pivot around Y to view the sides without pivoting around X first...

mkeeter, 2017-10-06:
Yup, I'm eventually planning to add arcball rotation as an alternate mode.  Not sure when I'll get to it, so I'll leave this open as a tracking issue.

Paul-Wray, 2018-06-19:
Some controls for the various orthogonal views would be nice too. 
Or perhaps some scheme fns to control the view, set up a turntable, resolution etc.

mkeeter, 2018-06-19:
Agreed, I think I'd prefer having Scheme functions to replace the existing magic comments.

mkeeter, 2018-06-30:
Y-up mode is now in the latest commit to `master`, in the `View -> Rotation mode` menu.

I'm just coming off a deep dive in the kernel (to make meshing faster + more robust!), and plan to devote a bit more time to the UI in coming months.

closed 2018-06-30 by mkeeter

omgitsraven, 2018-06-30:
Ahh, thank you :D