libfive/libfive #49: Crash when passing zero arguments to *

nickodell, 2017-10-10:
Build: a92168e

Steps to reproduce:

1. Open tutorial.ao
2. Find the expression `(* x x)`
3. Delete both x's.
4. Studio crashes.

Here's a trace in gdb:

```
GNU gdb (Debian 7.12-6) 7.12.0.20161007-git
[snipped]
QWidget::setWindowModified: The window title does not contain a '[*]' placeholder
[snipped]

Thread 6 "QThread" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffda7a9700 (LWP 2050)]
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff531d3fa in __GI_abort () at abort.c:89
#2  0x00007ffff5359bd0 in __libc_message (do_abort=do_abort@entry=2, 
    fmt=fmt@entry=0x7ffff544edd0 "*** Error in `%s': %s: 0x%s ***\n")
    at ../sysdeps/posix/libc_fatal.c:175
#3  0x00007ffff535ff96 in malloc_printerr (action=3, 
    str=0x7ffff544b9c8 "malloc(): memory corruption", ptr=<optimized out>, 
    ar_ptr=<optimized out>) at malloc.c:5049
#4  0x00007ffff5362059 in _int_malloc (av=av@entry=0x7fffd0000020, 
    bytes=bytes@entry=164) at malloc.c:3512
#5  0x00007ffff5363f34 in __GI___libc_malloc (bytes=164) at malloc.c:2928
#6  0x00007ffff795ff50 in Eigen::internal::handmade_aligned_malloc (size=132)
    at /usr/include/eigen3/Eigen/src/Core/util/Memory.h:88
#7  Eigen::internal::aligned_malloc (size=132)
    at /usr/include/eigen3/Eigen/src/Core/util/Memory.h:164
#8  Eigen::internal::conditional_aligned_malloc<true> (size=132)
    at /usr/include/eigen3/Eigen/src/Core/util/Memory.h:214
#9  Eigen::internal::conditional_aligned_new_auto<float, true> (size=33)
    at /usr/include/eigen3/Eigen/src/Core/util/Memory.h:374
#10 Eigen::DenseStorage<float, -1, 3, -1, 0>::resize (cols=11, size=33, 
    this=0x7fffda7a86c8)
    at /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:477
#11 Eigen::PlainObjectBase<Eigen::Array<float, 3, -1, 0, 3, -1> >::resize (
    cols=11, rows=3, this=0x7fffda7a86c8)
    at /usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:293
#12 Eigen::PlainObjectBase<Eigen::Array<float, 3, -1, 0, 3, -1> >::_init2<int, unsigned long> (cols=11, rows=3, this=0x7fffda7a86c8)
    at /usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:739
#13 Eigen::Array<float, 3, -1, 0, 3, -1>::Array<int, unsigned long> (
    val1=<optimized out>, val0=<optimized out>, this=0x7fffda7a86c8)
    at /usr/include/eigen3/Eigen/src/Core/Array.h:181
#14 Kernel::DerivEvaluator::DerivEvaluator (this=0x7fffda7a86a8, t=..., 
    vars=...) at /home/nick/ao/ao/src/eval/eval_deriv.cpp:31
#15 0x00007ffff7960672 in Kernel::FeatureEvaluator::FeatureEvaluator (
    this=<optimized out>, t=..., vars=...)
    at /home/nick/ao/ao/src/eval/eval_feature.cpp:31
#16 0x0000555555583a11 in Kernel::XTreeEvaluator::XTreeEvaluator (
    vars=std::map with 0 elements, t=..., this=<optimized out>)
    at /home/nick/ao/ao/src/../include/ao/render/brep/eval_xtree.hpp:36
#17 Shape::Shape (this=0x7fffd0149b60, t=..., 
    vars=std::shared_ptr (empty) 0x7fffda7a88c0)
    at /home/nick/ao/gui/src/shape.cpp:36
#18 0x0000555555591080 in _Interpreter::eval (this=0x555555ad5730)
    at /home/nick/ao/gui/src/interpreter.cpp:187
#19 0x00007ffff61d0499 in QObject::event(QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#20 0x00007ffff6a86b8c in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#21 0x00007ffff6a8e341 in QApplication::notify(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#22 0x00007ffff61a39e0 in QCoreApplication::notifyInternal2(QObject*, QEvent*)
    () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#23 0x00007ffff61a616d in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#24 0x00007ffff61f7c43 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#25 0x00007ffff2c1b7f7 in g_main_context_dispatch ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#26 0x00007ffff2c1ba60 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#27 0x00007ffff2c1bb0c in g_main_context_iteration ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#28 0x00007ffff61f804f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#29 0x00007ffff61a19ca in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#30 0x00007ffff5fcf0f3 in QThread::exec() ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#31 0x00007ffff5fd3da8 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#32 0x00007ffff72c1494 in start_thread (arg=0x7fffda7a9700)
    at pthread_create.c:333
#33 0x00007ffff53d1aff in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:97
(gdb) quit
A debugging session is active.

	Inferior 1 [process 2042] will be killed.

Quit anyway? (y or n) y
```

mkeeter, 2017-10-10:
Interesting, I can't reproduce this on my system:

[image]

Do any of the unit tests fail (running `./ao/test/ao-test`) in the build directory?

nickodell, 2017-10-10:
Update to the reproduction steps: It seems to happen whenever I remove one of the variables. Entirely deleting the x clause also triggers it.

The tests never finish or produce any output. Running under gdb, I see a segmentation fault.

```
Program received signal SIGSEGV, Segmentation fault.
_int_malloc (av=av@entry=0x7ffff66edb00 <main_arena>, bytes=bytes@entry=27680)
    at malloc.c:3840
3840	malloc.c: No such file or directory.
(gdb) bt
#0  _int_malloc (av=av@entry=0x7ffff66edb00 <main_arena>, 
    bytes=bytes@entry=27680) at malloc.c:3840
#1  0x00007ffff63cef34 in __GI___libc_malloc (bytes=27680) at malloc.c:2928
#2  0x00007ffff7957694 in Eigen::internal::handmade_aligned_malloc (size=27648)
    at /usr/include/eigen3/Eigen/src/Core/util/Memory.h:88
#3  Eigen::internal::aligned_malloc (size=27648)
    at /usr/include/eigen3/Eigen/src/Core/util/Memory.h:164
#4  Eigen::internal::conditional_aligned_malloc<true> (size=27648)
    at /usr/include/eigen3/Eigen/src/Core/util/Memory.h:214
#5  Eigen::internal::conditional_aligned_new_auto<Eigen::Array<float, 3, 256, 0, 3, 256>, true> (size=<optimized out>)
    at /usr/include/eigen3/Eigen/src/Core/util/Memory.h:374
#6  Eigen::DenseStorage<Eigen::Array<float, 3, 256, 0, 3, 256>, -1, -1, 1, 0>::resize (rows=9, size=9, this=0x7fffffffb340)
    at /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:550
#7  Eigen::PlainObjectBase<Eigen::Array<Eigen::Array<float, 3, 256, 0, 3, 256>, -1, 1, 0, -1, 1> >::resize (cols=1, rows=9, this=0x7fffffffb340)
    at /usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:293
#8  Eigen::PlainObjectBase<Eigen::Array<Eigen::Array<float, 3, 256, 0, 3, 256>, -1, 1, 0, -1, 1> >::_init2<unsigned long, int> (cols=1, rows=9, 
    this=0x7fffffffb340)
    at /usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:739
#9  Eigen::Array<Eigen::Array<float, 3, 256, 0, 3, 256>, -1, 1, 0, -1, 1>::Array---Type <return> to continue, or q <return> to quit---
<unsigned long, int> (val1=<optimized out>, val0=<optimized out>, 
    this=0x7fffffffb340) at /usr/include/eigen3/Eigen/src/Core/Array.h:181
#10 Kernel::DerivArrayEvaluator::DerivArrayEvaluator (this=0x7fffffffb200, 
    t=..., vars=...) at /home/nick/ao/ao/src/eval/eval_deriv_array.cpp:31
#11 0x00007ffff797a69b in Kernel::XTreeEvaluator::XTreeEvaluator (
    this=0x7fffffffb1e0, t=..., vars=std::map with 0 elements)
    at /home/nick/ao/ao/src/../include/ao/render/brep/eval_xtree.hpp:36
#12 0x00007ffff7989c93 in Kernel::XTree<2u>::build (t=..., 
    vars=std::map with 0 elements, region=..., 
    min_feature=0.10000000149011612, max_err=1e-08, 
    multithread=<optimized out>, cancel=...)
    at /home/nick/ao/ao/src/render/brep/xtree.cpp:63
#13 0x00007ffff798a94f in Kernel::XTree<2u>::build (t=..., region=..., 
    min_feature=<optimized out>, max_err=max_err@entry=1e-08, 
    multithread=multithread@entry=true)
    at /home/nick/ao/ao/src/render/brep/xtree.cpp:47
#14 0x00007ffff799358a in Kernel::Contours::render (t=..., r=..., 
    min_feature=<optimized out>)
    at /home/nick/ao/ao/src/render/brep/contours.cpp:80
#15 0x00007ffff79ba0e2 in ao_tree_render_slice (
    tree=tree@entry=0x5555558cc360, R=..., z=z@entry=0, res=res@entry=10)
    at /home/nick/ao/ao/src/ao.cpp:176
#16 0x00005555555b16b1 in ____C_A_T_C_H____T_E_S_T____66 ()
---Type <return> to continue, or q <return> to quit---
    at /home/nick/ao/ao/test/api.cpp:76
#17 0x00005555555adcf9 in Catch::RunContext::invokeActiveTestCase (
    this=0x7fffffffd2d0) at /home/nick/ao/ao/test/catch.hpp:5473
#18 Catch::RunContext::runCurrentTest (redirectedCerr="", redirectedCout="", 
    this=0x7fffffffd2d0) at /home/nick/ao/ao/test/catch.hpp:5445
#19 Catch::RunContext::runTest (testCase=..., this=0x7fffffffd2d0)
    at /home/nick/ao/ao/test/catch.hpp:5284
#20 Catch::Runner::runTests (this=0x7fffffffd9f0)
    at /home/nick/ao/ao/test/catch.hpp:5603
#21 0x00005555555ae664 in Catch::Session::run (this=<optimized out>)
    at /home/nick/ao/ao/test/catch.hpp:5734
#22 0x000055555557eaa7 in Catch::Session::run (argv=0x7fffffffdf98, argc=1, 
    this=0x7fffffffdd40) at /home/nick/ao/ao/test/catch.hpp:5714
#23 main (argc=1, argv=0x7fffffffdf98) at /home/nick/ao/ao/test/main.cpp:24
```

mkeeter, 2017-10-10:
I think this is ye olde aligned `new` operator issue on Linux.  Can you see whether `488bf4c2177f5435be4afab919568e003d46e195` fixes either the unit tests or the crashes?

nickodell, 2017-10-10:
No, it doesn't fix either. On `ao-test`, there's a nearly identical stack trace and segfault.

mkeeter, 2017-10-11:
One more thing to try: switch the compile mode from `RELEASE` to `DEBUG` in the top-level `CMakeLists.txt`, rebuild, and re-run that unit test (looks like it's `./ao/test/ao-test ao_tree_render_slice`).  In debug builds, more Eigen assertions are enabled, so it will tell us if we're doing something wrong with the matrices.

nickodell, 2017-10-11:
Got it. Here's the output of `ao-test`:

```
ao-test: /usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:425: Eigen::DenseCoeffsBase<Derived, 1>::Scalar& Eigen::DenseCoeffsBase<Derived, 1>::operator()(Eigen::Index) [with Derived = Eigen::Array<float, -1, 1>; Eigen::DenseCoeffsBase<Derived, 1>::Scalar = float; Eigen::Index = long int]: Assertion `index >= 0 && index < size()' failed.

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
ao-test is a Catch v1.1 b3 host application.
Run with -? for options

-------------------------------------------------------------------------------
ao_tree_eval_f
-------------------------------------------------------------------------------
/home/nick/ao/ao/test/api.cpp:51
...............................................................................

/home/nick/ao/ao/test/api.cpp:51: FAILED:
due to a fatal error condition:
  SIGABRT - Abort (abnormal termination) signal

===============================================================================
test cases: 3 | 2 passed | 1 failed
assertions: 8 | 7 passed | 1 failed
```

mkeeter, 2017-10-11:
Awesome!  That's still a little vague – can you run it in a debugger to get the abort's backtrace, so we can see what part of the kernel is misbehaving?

nickodell, 2017-10-11:
Sure:

```
$ gdb -x cmd ./ao/test/ao-test
GNU gdb (Debian 7.12-6) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./ao/test/ao-test...done.
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
ao-test: /usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:425: Eigen::DenseCoeffsBase<Derived, 1>::Scalar& Eigen::DenseCoeffsBase<Derived, 1>::operator()(Eigen::Index) [with Derived = Eigen::Array<float, -1, 1>; Eigen::DenseCoeffsBase<Derived, 1>::Scalar = float; Eigen::Index = long int]: Assertion `index >= 0 && index < size()' failed.

Program received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff563d3fa in __GI_abort () at abort.c:89
#2  0x00007ffff5634e37 in __assert_fail_base (fmt=<optimized out>, 
    assertion=assertion@entry=0x7ffff77abdf5 "index >= 0 && index < size()", 
    file=file@entry=0x7ffff77abdc0 "/usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h", line=line@entry=425, 
    function=function@entry=0x7ffff77ac7c0 <Eigen::DenseCoeffsBase<Eigen::Array<float, -1, 1, 0, -1, 1>, 1>::operator()(long)::__PRETTY_FUNCTION__> "Eigen::DenseCoeffsBase<Derived, 1>::Scalar& Eigen::DenseCoeffsBase<Derived, 1>::operator()(Eigen::Index) [with Derived = Eigen::Array<float, -1, 1>; Eigen::DenseCoeffsBase<Derived, 1>::Scalar = float;"...) at assert.c:92
#3  0x00007ffff5634ee2 in __GI___assert_fail (
    assertion=0x7ffff77abdf5 "index >= 0 && index < size()", 
    file=0x7ffff77abdc0 "/usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h", line=425, 
    function=0x7ffff77ac7c0 <Eigen::DenseCoeffsBase<Eigen::Array<float, -1, 1, 0, -1, 1>, 1>::operator()(long)::__PRETTY_FUNCTION__> "Eigen::DenseCoeffsBase<Derived, 1>::Scalar& Eigen::DenseCoeffsBase<Derived, 1>::operator()(Eigen::Index) [with Derived = Eigen::Array<float, -1, 1>; Eigen::DenseCoeffsBase<Derived, 1>::Scalar = float;"...) at assert.c:101
#4  0x00007ffff75ad99c in Eigen::DenseCoeffsBase<Eigen::Array<float, -1, 1, 0, -1, 1>, 1>::operator() (this=0x7fffffffc4e0, index=5)
    at /usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:425
#5  0x00007ffff761be14 in Kernel::PointEvaluator::eval (this=0x7fffffffc4d0, 
    pt=...) at /home/nick/ao/ao/src/eval/eval_point.cpp:51
#6  0x00007ffff77a9913 in ao_tree_eval_f (t=0x555555adce20, p=...)
    at /home/nick/ao/ao/src/ao.cpp:161
#7  0x0000555555760faf in ____C_A_T_C_H____T_E_S_T____51 ()
    at /home/nick/ao/ao/test/api.cpp:57
#8  0x000055555572ff58 in Catch::FreeFunctionTestCase::invoke (
    this=0x555555acea40) at /home/nick/ao/ao/test/catch.hpp:5874
#9  0x000055555571cae3 in Catch::TestCase::invoke (this=0x555555afaf10)
    at /home/nick/ao/ao/test/catch.hpp:6779
#10 0x000055555572e007 in Catch::RunContext::invokeActiveTestCase (
    this=0x7fffffffd630) at /home/nick/ao/ao/test/catch.hpp:5473
#11 0x000055555572dce5 in Catch::RunContext::runCurrentTest (
    this=0x7fffffffd630, redirectedCout="", redirectedCerr="")
    at /home/nick/ao/ao/test/catch.hpp:5445
#12 0x000055555572c6e9 in Catch::RunContext::runTest (this=0x7fffffffd630, 
    testCase=...) at /home/nick/ao/ao/test/catch.hpp:5284
#13 0x000055555572e6e8 in Catch::Runner::runTests (this=0x7fffffffda50)
    at /home/nick/ao/ao/test/catch.hpp:5603
#14 0x000055555572f6bd in Catch::Session::run (this=0x7fffffffdd50)
    at /home/nick/ao/ao/test/catch.hpp:5734
#15 0x000055555572f572 in Catch::Session::run (this=0x7fffffffdd50, argc=1, 
    argv=0x7fffffffdf98) at /home/nick/ao/ao/test/catch.hpp:5714
#16 0x0000555555721745 in main (argc=1, argv=0x7fffffffdf98)
    at /home/nick/ao/ao/test/main.cpp:24
Kill the program being debugged? (y or n) [answered Y; input not from terminal]
```

nickodell, 2017-10-12:
PS: I wrote a guide to building this on debian. https://gist.github.com/nickodell/46180d736a7c2cb6836291a8f168ef20

mkeeter, 2017-10-12:
Thanks for the Debian guide!

I'm pretty mystified by the bug you're seeing.  Let me explain why, to help you debug on your end:

The failing test for [`ao_tree_eval_f`](https://github.com/mkeeter/ao/blob/master/ao/test/api.cpp#L51-L64) builds a tree representing `f(x, y) = x / y`.  It then converts this into a tape, which is a flattened 1D structure that can be walked to evaluate the tree.  The tape will include clauses for `x`, `y`, `x / y`, and `z` (because we always create slots for the base variables, even if they're unused).  It also includes the null clause, with index 0.

The tape also has an associated array of results, which should have room for 5 values (null, `x`, `y`, `x/y`, and `z`).  When I instrument the tape constructor with debug print statements, it tells me that X is at slot 3, Y is at slot 2, and Z is at slot 4.  In your system, the index of the `z` clause into the result array is being recorded as 5 instead of 4; you can see this in items `#5` and `#4` of your traceback.

Can you drop a debug print statement at the end of the `Tape` constructor, e.g.
```
    std::cout << "Got " << num_clauses << " clauses with\n"
              << "X = " << X << "\n"
              << "Y = " << Y << "\n"
              << "Z = " << Z << "\n";
```
then rerun the failing unit test (`./ao/test/ao-test ao_tree_eval_f`)?

If my explanation helped clarify the underlying logic, feel free to debug more independently on your end as well 😃

nickodell, 2017-10-12:
Reading the logic in Tape's constructor, it looks like it's functioning as intended.

```
    // Make sure that X, Y, Z have been allocated space
    std::vector<Tree> axes = {Tree::X(), Tree::Y(), Tree::Z()};
    for (auto a : axes)
    {
        if (clauses.find(a.id()) == clauses.end())
        {
            std::cout << "can't find " << a;
            clauses[a.id()] = clauses.size();
        }
    }
```

The cout here says "can't find z" That's not surprising, because z isn't in the expression. This propagates through and sets Z to 5, the size of the vector. If we change the testcase to involve z, then z is in `clauses`, and there's no crash.

This testcase works:

```
TEST_CASE("ao_tree_eval_f")
{
    auto a = ao_tree_x();
    auto b = ao_tree_y();
    auto d = ao_tree_z();
    auto c = ao_tree_binary(Opcode::DIV, a, b);
    auto e = ao_tree_binary(Opcode::ADD, c, d);

    REQUIRE(ao_tree_eval_f(e, {1,2,3}) == 3.5);
    REQUIRE(ao_tree_eval_f(e, {1,4,3}) == 3.25);
    REQUIRE(ao_tree_eval_f(e, {1,-1,3}) == 2);

    ao_tree_delete(a);
    ao_tree_delete(b);
    ao_tree_delete(c);
}
```

mkeeter, 2017-10-12:
>The cout here says "can't find z" That's not surprising, because z isn't in the expression. This propagates through and sets Z to 5, the size of the vector.

This is where things are misbehaving: `Z` should be set to 4, because there's only three clauses at that point (`x`, `y`, and `x / y`).  Can you sprinkle a few more print statements around to try and figure out why `Z` is set to 5 rather than 4?

nickodell, 2017-10-13:
Can you sanity check something for me?

Run this on your system. It prints 1 for me.

```
#include <iostream>
#include <unordered_map>
using namespace std;

int main() {
	std::unordered_map<int, int> u;
	u[0] = u.size();
	cout << u[0] << endl;
	return 0;
}
```

mkeeter, 2017-10-13:
Good catch, that prints `0` for me.  It's probably related to how `operator[]` works on maps, where it adds the slot before checking the size on your system but not mine.

Can you see if 2b71b9aaf649f338384a749b4567da13f79f6187 fixes the problem?

nickodell, 2017-10-13:
Fix confirmed.

```
$ ./ao/test/ao-test ao_tree_eval_f 
===============================================================================
All tests passed (3 assertions in 1 test case)
```

closed 2017-10-13 by nickodell

mkeeter, 2017-10-13:
Cool, thanks for your help in debugging!  I assume this also fixes the original meshing crash?

nickodell, 2017-10-20:
Yes, it does.