libfive/libfive #50: boost not found

fponticelli, 2017-10-11:
I can't seem to build Ao on Mac OS High Sierra. I installed `boost`, `eigen` and `qt` using brew.

I run

```bash
> cmake -D CMAKE_PREFIX_PATH=/usr/local/Cellar/qt5/5.7.0  ..
-- The C compiler identification is AppleClang 9.0.0.9000037
-- The CXX compiler identification is AppleClang 9.0.0.9000037
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ZLIB: /usr/lib/libz.dylib (found version "1.2.11")
-- Found PNG: /usr/local/lib/libpng.dylib (found version "1.5.18")
-- Found Eigen3: /usr/local/include/eigen3 (Required is at least version "2.91.0")
Checking dependencies:
  libao-kernel: ✓
  guile-ao:     ✓
  Studio:       ✓
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/francoponticelli/projects/ao/build
```

Which seems ok. When I run `make`, I get:

```bash
Scanning dependencies of target ao-kernel
[  1%] Building CXX object ao/src/CMakeFiles/ao-kernel.dir/eval/base.cpp.o
In file included from /Users/francoponticelli/projects/ao/ao/src/eval/base.cpp:21:
In file included from /Users/francoponticelli/projects/ao/ao/src/../include/ao/eval/base.hpp:20:
/Users/francoponticelli/projects/ao/ao/src/../include/ao/eval/tape.hpp:23:10: fatal error: 'boost/bimap.hpp' file not
      found
#include <boost/bimap.hpp>
         ^~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [ao/src/CMakeFiles/ao-kernel.dir/eval/base.cpp.o] Error 1
make[1]: *** [ao/src/CMakeFiles/ao-kernel.dir/all] Error 2
make: *** [all] Error 2
```

mkeeter, 2017-10-11:
Are you on the very latest commit (488bf4c2177f5435be4afab919568e003d46e195)?  I recently upgraded to High Sierra myself, and made a few changes in 9bb63d3750c32950ca402aed522b2a56fb0e28a6 to get things working.

fponticelli, 2017-10-11:
I just pulled `f73c824` and I am having the same issue.

mkeeter, 2017-10-11:
Can you delete your build folder and start again, then paste whatever `cmake` finds?  It should be more verbose about whether it has found useful libraries.

fponticelli, 2017-10-11:
done, but still same. See attached images:

[image]
[image]

mkeeter, 2017-10-12:
It looks like I was using the wrong name for the `boost` include directories.  Can you re-pull and see if it works now (878ea7943bf49d781bc21b15c95abdd2d500be65)?

fponticelli, 2017-10-12:
Now `boost` is fixed and I get to 33%. Than it fails with this message:

```
Scanning dependencies of target guile-ao
[ 34%] Building CXX object ao/bind/CMakeFiles/guile-ao.dir/bundle.cpp.o
In file included from /Users/francoponticelli/projects/ao/build/ao/bind/bundle.cpp:22:
In file included from /Users/francoponticelli/projects/ao/ao/bind/./ao-guile.h:21:
/usr/local/include/guile/2.2/libguile.h:25:10: fatal error: 'gmp.h' file not found
#include <gmp.h>
         ^~~~~~~
1 error generated.
make[2]: *** [ao/bind/CMakeFiles/guile-ao.dir/bundle.cpp.o] Error 1
make[1]: *** [ao/bind/CMakeFiles/guile-ao.dir/all] Error 2
make: *** [all] Error 2
```

I do have `gmp` v.6.1.2 installed using `brew`.

mkeeter, 2017-10-12:
Is there something strange going on with `brew`?

On my system, both `boost/bimap.hpp` and `gmp.h` are found in `/usr/local/include`, so any search path that finds one will find the other.  Where did `brew` put `gmp.h` in your system?

fponticelli, 2017-10-12:
It is really strange. I reinstalled `gmp`/`guile` just to make sure and `brew doctor` does not report any anomalies. I do have `gmp.h` in `/usr/local/include` ... this is the end of `ls` on that folder and the command that fails:

[image]

mkeeter, 2017-10-12:
Can you run `VERBOSE=1 make` to see the actual build commands?

fponticelli, 2017-10-12:
here is the output

```
VERBOSE=1 make                                                                                                                                         
/usr/local/Cellar/cmake/3.9.4/bin/cmake -H/Users/francoponticelli/projects/ao -B/Users/francoponticelli/projects/ao/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/local/Cellar/cmake/3.9.4/bin/cmake -E cmake_progress_start /Users/francoponticelli/projects/ao/build/CMakeFiles /Users/francoponticelli/projects/ao/build/CMakeFiles/progress.marks
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f CMakeFiles/Makefile2 all
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f ao/src/CMakeFiles/ao-kernel.dir/build.make ao/src/CMakeFiles/ao-kernel.dir/depend
cd /Users/francoponticelli/projects/ao/build && /usr/local/Cellar/cmake/3.9.4/bin/cmake -E cmake_depends "Unix Makefiles" /Users/francoponticelli/projects/ao /Users/francoponticelli/projects/ao/ao/src /Users/francoponticelli/projects/ao/build /Users/francoponticelli/projects/ao/build/ao/src /Users/francoponticelli/projects/ao/build/ao/src/CMakeFiles/ao-kernel.dir/DependInfo.cmake --color=
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f ao/src/CMakeFiles/ao-kernel.dir/build.make ao/src/CMakeFiles/ao-kernel.dir/build
make[2]: Nothing to be done for `ao/src/CMakeFiles/ao-kernel.dir/build'.
[ 32%] Built target ao-kernel
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f ao/bind/CMakeFiles/guile-ao.dir/build.make ao/bind/CMakeFiles/guile-ao.dir/depend
cd /Users/francoponticelli/projects/ao/build && /usr/local/Cellar/cmake/3.9.4/bin/cmake -E cmake_depends "Unix Makefiles" /Users/francoponticelli/projects/ao /Users/francoponticelli/projects/ao/ao/bind /Users/francoponticelli/projects/ao/build /Users/francoponticelli/projects/ao/build/ao/bind /Users/francoponticelli/projects/ao/build/ao/bind/CMakeFiles/guile-ao.dir/DependInfo.cmake --color=
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f ao/bind/CMakeFiles/guile-ao.dir/build.make ao/bind/CMakeFiles/guile-ao.dir/build
[ 33%] Building CXX object ao/bind/CMakeFiles/guile-ao.dir/bundle.cpp.o
cd /Users/francoponticelli/projects/ao/build/ao/bind && /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++  -Dguile_ao_EXPORTS -I/Users/francoponticelli/projects/ao/ao/bind/. -I/usr/local/include/guile/2.2 -isystem /usr/local/Cellar/boost/1.65.1/include -isystem /Library/Frameworks/UnixImageIO.framework/Headers -isystem /usr/local/include/eigen3 -I/Users/francoponticelli/projects/ao/ao/src/../include  -Wall -Wextra -g -fPIC -pedantic -Werror=switch -march=native -O3 -DRELEASE -DEIGEN_NO_DEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk -fPIC   -D'GITREV="'878ea79'"' -D'GITTAG=""' -D'GITBRANCH="master "' -std=gnu++11 -o CMakeFiles/guile-ao.dir/bundle.cpp.o -c /Users/francoponticelli/projects/ao/build/ao/bind/bundle.cpp
In file included from /Users/francoponticelli/projects/ao/build/ao/bind/bundle.cpp:22:
In file included from /Users/francoponticelli/projects/ao/ao/bind/./ao-guile.h:21:
/usr/local/include/guile/2.2/libguile.h:25:10: fatal error: 'gmp.h' file not found
#include <gmp.h>
         ^~~~~~~
1 error generated.
make[2]: *** [ao/bind/CMakeFiles/guile-ao.dir/bundle.cpp.o] Error 1
make[1]: *** [ao/bind/CMakeFiles/guile-ao.dir/all] Error 2
make: *** [all] Error 2
```

mkeeter, 2017-10-12:
I've changed the build system to use `pkg-config` rather than `FindXYZ.cmake` scripts from random places on the internet.  Maybe this will fix things!

Can you pull to the latest `master` and see if things are looking any more robust?  If not, the `VERBOSE=1 make` output is very helpful.

fponticelli, 2017-10-13:
That's fantastic, it worked! Now it is time to start playing ;)

Thank you a lot!

closed 2017-10-13 by fponticelli