libfive/libfive #54: Bounding box geometry goes haywire if the camera is inside of it

omgitsraven, 2017-10-19:
Try either turning up the bounding box values to something large, or using the zoom/scroll gesture to bring the camera inside of it; you'll wind up seeing an 'X' drawn across the background for some reason.

omgitsraven, 2017-10-20:
This commit does fix the 'x' when zooming too close, but now it seems like the bounding box is permanently orthographic, even when the view is set to 'perspective' mode?

mkeeter, 2017-10-20:
Well, I fixed the always-orthographic issue, but now we're back to having the cross show up.  I think that the perspective projection matrix isn't *quite* right.

omgitsraven, 2017-10-20:
I haven't looked closely into this, but based on stuff I've dealt with in the past, I think the projection matrix is fine, it's just lacking a nearplane?

closed 2017-10-21 by mkeeter

omgitsraven, 2017-10-21:
There we go, looks fine now :) Thanks!

commit 604c80b5 (2017-10-20): Correctly apply transform to bounding box corners (fixes #54)
commit 20550651 (2017-10-21): Clamp w values to 0 (fixes #54 for real)