libfive/libfive #55: Backspace deletes two characters almost always

omgitsraven, 2017-10-19:
I see there's a commit saying that backspace will delete both characters of a pseudo-tab, but I think it's having a lot of false positives: try opening a new document, and typing 123456789, then hitting backspace a few times. At least for me (on macOS), it always deletes to the closest even-numbered character (the first backspace deletes only '9', but the next deletes '78', and so on...)

mkeeter, 2017-10-19:
Oops!  That commit should fix things.

omgitsraven, 2017-10-20:
It's still not quite right -- if you delete a character with a space before it, it'll delete the space too.

i.e., if you've got a line like `(+ 1 2)` and your cursor is DIRECTLY after the 1, and you hit backspace, then your cursor will be directly after the +.

This means that, if you're trying to change a value to something else, you'll wind up having to hit spacebar after hitting backspace, or else `(+ 1 2)` becomes `(+0 2)` (note the lack of space between the + and the 0).

mkeeter, 2017-10-20:
Yep, you're right!  I'll reopen this and fix soon.

omgitsraven, 2017-10-20:
Seems like the new commit fixes it :) Thanks!

commit 3a36ff63 (2017-10-19): Only double-delete whitespace (fixes #55)
closed 2017-10-20 by omgitsraven