libfive/libfive #56: Text sometimes is invisible when first loaded

omgitsraven, 2017-10-19:
I'm having a strange issue where there's a small chance that a saved .ao document will have some of its lines invisibly hidden until something triggers a reflow. Some entire lines (seemingly only ones containing only close-parentheses?) will be invisible, but I'll still be able to move the cursor through them with the arrow keys -- I can tell because when the cursor reaches the (invisible) close-parentheses, the corresponding open-parenthesis will highlight like normal.

Typing out a `#!` comment block will fix any affected text that falls inside the new comment (but if there was already another block further on for example, that block will stop the new block since comment blocks don't nest, and nothing further on will be fixed.)

I've made a video demonstrating the bug (watch the areas that I point to with the cursor, the parenthesis-matching shows up kind of faintly unfortunately), and I've attached the file that the issue occurs with -- although note that it seems to only happen on maybe 25% of loads. (I've had it happen a few times in a row, though; I'm not sure if there's something persisting between launches that causes it to happen or not happen even though I'm not saving the file between attempts...)

Also, even when this happens, it doesn't seem to apply to the entire file; it seems as though it's hitting something that breaks everything after that point. Try skipping down to the `;eyes` section featured in the video; that part is usually affected if anything is.

Video: 
[linebreaks.zip](https://github.com/mkeeter/ao/files/1399869/linebreaks.zip)

.ao file: 
[idol.zip](https://github.com/mkeeter/ao/files/1399870/idol.zip)

2TallTim, 2017-10-20:
I'm also seeing this issue, except that it seems the entire text editor side is not refreshing until the preview area is refreshed. If the code is valid, the editor seems to refresh fine, but if there is any error, the editor doesn't refresh.

mkeeter, 2017-10-22:
Looks like [QTBUG-20354](https://bugreports.qt.io/browse/QTBUG-20354).  I'll see if I can find a workaround.

2TallTim, 2017-10-22:
I'm using Qt 5.9.2, not sure if it shows up in any other versions.

mkeeter, 2017-10-22:
Can you check whether 58b3b415de9f47132a1033ffc0a23298dfbd1e90 helps?  It's a terrible hack, but seems to fix @omgitsraven 's script.

omgitsraven, 2017-10-22:
FWIW, I was on Qt 5.9.1 when I reported this (it didn't occur to me it could be a Qt bug), and upgrading to Qt 5.9.2 and rebuilding (just now, but with sources from before 58b3b41) didn't fix my bug. (Will try now with the new changes...)

2TallTim, 2017-10-22:
Still seems to be the same issue, even after updating to 58b3b41.

omgitsraven, 2017-10-22:
So, it's always hard to conclusively confirm that an only-occasional bug is gone for good, but I can say that out of a dozen or so tries, I haven't seen it yet since 58b3b41 at least :) Will come back to this thread if I ever see it again. Thanks!

2TallTim, 2017-10-22:
Yeah, I'm still experiencing this 100% of the time. Should I open an new issue, or continue in this thread? What I'm getting might be different, though. I don't see any updating of the text area, while @omgitsraven notes that it's just certain lines that are missing. I'm not experiencing missing lines, just a lack of updating in general of the text area. The cursor doesn't even blink unless the viewport is being moved.

mkeeter, 2017-10-22:
@2TallTim can you open up a new issue?  It sounds like a different bug, especially because this commit fixed @omgitsraven 's issue.

omgitsraven, 2017-12-06:
Uh oh -- I'm now experiencing this bug again, with the latest sources.
Is this something that's been fixed in Qt and so you took out the workaround? I was still using Qt 5.9.2 when it happened just now; I've now upgraded to Qt 5.9.3 and will rebuild and see if it's still happening, but since it's an intermittent sort of bug I might not know for a while.

omgitsraven, 2017-12-06:
Nope, it's definitely still happening even when built with Qt 5.9.3 :(

mkeeter, 2017-12-28:
Can you upload the file that you see it on?  I tested it out on `idol.ao` from earlier in the thread, and don't see anything missing, but wouldn't be surprised if there's still something wrong...

omgitsraven, 2017-12-28:
hmm—this is getting a bit off topic, but have the build instructions changed lately? I just downloaded the latest sources to test this and ran the usual commands, but as soon as it gets to 1% it says "fatal: Not a git repository..." (I am indeed just downloading a zip rather than bothering with git, but that's always worked in the past.)

(If this isn't something obvious, I'll try actually checking it out with git, and if that still doesn't work, I'll make a standalone thread for this...)

But, fwiw, I was actually testing on idol.ao, although possibly a later version of it, I don't remember exactly any more...

mkeeter, 2017-12-28:
I've recently changed how I embed version / revision into the build, so that's probably the cause; let me make a standalone issue for it.

omgitsraven, 2017-12-28:
OK, I've got it building again, and I've also updated to qt 5.10.0, and it's definitely still happening -- however, it's indeed only happening in a later version of idol.ao not the version that I'd posted here, sorry about that. Here's the version it happens in, right near the bottom of the file, every time I've tested:
[idol 2.ao.zip](https://github.com/mkeeter/ao/files/1591907/idol.2.ao.zip)

mkeeter, 2017-12-29:
Okay, I've switched from `QTextEdit` to `QPlainTextEdit`, and everything seems to be working again on `idol 2.ao`, so... mission accomplished?

Fee free to re-open if this shows up again.

closed 2017-12-29 by mkeeter

omgitsraven, 2017-12-29:
Success! Thanks.