libfive/libfive #58: Crash when zooming, panning, and changing detail

nickodell, 2017-10-20:
Steps to reproduce: (tentative)

1. Open tutorial.ao
2. Zoom around some
3. Change the detail setting (the last argument on the render line)
4. Pan the screen

This does not happen every time, and I'm trying to nail down the reproduction steps more definitely.

Crash log:

```
$ ./gui/Studio
QWidget::setWindowModified: The window title does not contain a '[*]' placeholder
ASSERT failure in QVector<T>::at: "index out of range", file /usr/include/x86_64-linux-gnu/qt5/QtCore/qvector.h, line 429
Aborted
```

I have not been able to reproduce this under a debugger.

mkeeter, 2017-10-21:
That's... odd.

The only `QVector` in the whole program is `View::pick_depth`, and the only place we call `.at()` on it is in [`toModelPos`](https://github.com/mkeeter/ao/blob/master/gui/src/view.cpp#L360), but that should always be in tight sync with the viewport size (enforced by calls to `syncPicker` in the mouse handlers).

I would bet that this is a crash somewhere deeper in Qt, but it's hard to tell without a backtrace.

nickodell, 2017-10-23:
Who's got two thumbs and a stack trace of this happening? This guy!

```
ASSERT failure in QVector<T>::at: "index out of range", file /usr/include/x86_64-linux-gnu/qt5/QtCore/qvector.h, line 429

Thread 1 "Studio" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff43643fa in __GI_abort () at abort.c:89
#2  0x00007ffff5001561 in QMessageLogger::fatal(char const*, ...) const ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#3  0x00007ffff4ffca91 in qt_assert_x(char const*, char const*, char const*, int) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#4  0x000055555561b583 in QVector<float>::at (this=0x55555598afc8, i=578913)
    at /usr/include/x86_64-linux-gnu/qt5/QtCore/qvector.h:429
#5  0x0000555555618c5d in View::toModelPos (this=0x55555598ad60, pt=...)
    at /home/nick/ao/gui/src/view.cpp:362
#6  0x00005555556195ac in View::checkHoverTarget (this=0x55555598ad60, pos=...)
    at /home/nick/ao/gui/src/view.cpp:475
#7  0x0000555555618b29 in View::mouseMoveEvent (this=0x55555598ad60, 
    event=0x7fffffffd600) at /home/nick/ao/gui/src/view.cpp:353
#8  0x00007ffff5b15278 in QWidget::event(QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#9  0x00007ffff5acdb8c in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#10 0x00007ffff5ad61fd in QApplication::notify(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#11 0x00007ffff51ea9e0 in QCoreApplication::notifyInternal2(QObject*, QEvent*)
    () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#12 0x00007ffff5ad48ad in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEv---Type <return> to continue, or q <return> to quit---
ent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#13 0x00007ffff5b2f906 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#14 0x00007ffff5b32313 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#15 0x00007ffff5acdb8c in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16 0x00007ffff5ad5341 in QApplication::notify(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#17 0x00007ffff51ea9e0 in QCoreApplication::notifyInternal2(QObject*, QEvent*)
    () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#18 0x00007ffff5532b03 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#19 0x00007ffff5534685 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#20 0x00007ffff551261b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#21 0x00007ffff7ea74c0 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
---Type <return> to continue, or q <return> to quit---
#22 0x00007ffff1ed27f7 in g_main_context_dispatch ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#23 0x00007ffff1ed2a60 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#24 0x00007ffff1ed2b0c in g_main_context_iteration ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#25 0x00007ffff523f04f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#26 0x00007ffff51e89ca in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#27 0x00007ffff51f113c in QCoreApplication::exec() ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#28 0x00005555555de331 in main (argc=1, argv=0x7fffffffdfa8)
    at /home/nick/ao/gui/src/main.cpp:46
```

I think this is triggered by mousing off of the window, but I'm still not 100% sure. It's really finicky.

Here's the value of pt.x and pt.y inside toModelPos:

```
(gdb) frame 5
#5  0x0000555555618c5d in View::toModelPos (this=0x55555598ad60, pt=...)
    at /home/nick/ao/gui/src/view.cpp:362
362	                            (pick_img.height() - pt.y())) - 1);
(gdb) print pt.y()
$1 = 0
(gdb) print pt.x()
[Thread 0x7fffc5806700 (LWP 25808) exited]
[Thread 0x7fffc6007700 (LWP 25807) exited]
$2 = 573
```

mkeeter, 2017-10-23:
Cool, that stacktrace harkens back to my original theory (that I thought couldn't be the problem).  I still can't trigger it, but I think it's caused by mouse events that are outside of the actual window (so looking up their position in the depth image fails).

Can you see if 407e0181a62ce2928166ff2e98d16e07bd7a0eb8 fixes the problem?

nickodell, 2017-10-23:
Not fixed.

```
$ git checkout 407e018
[...]
$ make
[...]
$ ./gui/Studio
QWidget::setWindowModified: The window title does not contain a '[*]' placeholder
ASSERT failure in QVector<T>::at: "index out of range", file /usr/include/x86_64-linux-gnu/qt5/QtCore/qvector.h, line 429
Aborted
```

I think you have an off-by-one error here. Think about this algebraically. When can the argument to `at()` be more than width*height?

    pick_depth.at(pt.x() + pick_img.width() * (pick_img.height() - pt.y()))
    index = x + width * (height - y) 

Plug in (x=50, y=0).

    index = 50 + width * (height - 0) 
    index = 50 + width * height
    50 + width * height > width * height

mkeeter, 2017-10-23:
You're right: there's an off-by-one error in the Y subtraction.  See if 9d57a23d8951aea5588cc791e1518bc0a064985b fixes the issue.

nickodell, 2017-10-23:
Fix confirmed.

On a related note, could I convince you to split that ternary statement into an if statement? It's pretty hard to follow.

closed 2017-10-23 by nickodell