libfive/libfive #59: Text editor doesn't update

2TallTim, 2017-10-22:
If the code in the editor window contains any errors, the text will not update until the viewport is moved. In addition, the entire window will not update if resized until the viewport is moved. If the code _is_ valid, it will update with significant delay, and the blinking cursor is not displayed in any situations. Tumbling the viewport continuously reveals the location of the cursor. I'm on Ubuntu 17.10 using Qt 5.9.2, eigen 3.3.4-3, and Guile 2.2.2. This issue is present with all files, including the tutorial files, empty files, and any files I've written myself.

mkeeter, 2017-10-22:
What do you mean by "text will not update"?  Does this mean that new characters aren't drawn as you type them?

2TallTim, 2017-10-22:
That's correct. Nothing changes in the text area until the code becomes valid, or the viewport is moved.

mkeeter, 2017-10-22:
Yikes!  Obviously, I don't see this issue on my home machine, so you'll have to do most of the debugging legwork.  Is there a sampling profiler (like [Instruments](https://en.wikipedia.org/wiki/Instruments_(software)) on Mac) that you could run to see where the code is spending its time?

2TallTim, 2017-10-22:
I just tried a couple, and I got no useful information from gProf, and valgrind runs out of memory or gives a segfault. Not really sure what to do here.

mkeeter, 2017-10-22:
Can you try out some of the Qt examples, e.g. the [notepad](http://doc.qt.io/qt-5/gettingstartedqt.html)?

2TallTim, 2017-10-22:
Sure, I'll give that a go. The output of valgrind might actually be useful, so here it is.
`tim@7559:~/Code/ao/build/gui$ valgrind --tool=callgrind ./Studio
==13969== Callgrind, a call-graph generating cache profiler
==13969== Copyright (C) 2002-2017, and GNU GPL'd, by Josef Weidendorfer et al.
==13969== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
==13969== Command: ./Studio
==13969== 
==13969== For interactive control, run 'callgrind_control -h'.
==13969== brk segment overflow in thread #7: can't grow to 0x4a51000
==13969== (see section Limitations in user manual)
==13969== NOTE: further instances of this message will not be shown
QWidget::setWindowModified: The window title does not contain a '[*]' placeholder
==13969== Warning: noted but unhandled ioctl 0x6458 with no size/direction hints.
==13969==    This could cause spurious value errors to appear.
==13969==    See README_MISSING_SYSCALL_OR_IOCTL for guidance on writing a proper wrapper.
==13969== 
==13969== Process terminating with default action of signal 11 (SIGSEGV)
==13969==  Access not within mapped region at address 0x8
==13969==    at 0x5FBB730: QOpenGLContext::shareGroup() const (in /home/tim/Qt/5.9.2/gcc_64/lib/libQt5Gui.so.5.9.2)
==13969==    by 0x6247CE3: ??? (in /home/tim/Qt/5.9.2/gcc_64/lib/libQt5Gui.so.5.9.2)
==13969==    by 0x6247E86: ??? (in /home/tim/Qt/5.9.2/gcc_64/lib/libQt5Gui.so.5.9.2)
==13969==    by 0x153CE5: View::redrawPicker() (qopenglfunctions.h:1244)
==13969==    by 0x68A8B95: QMetaObject::activate(QObject*, int, int, void**) (in /home/tim/Qt/5.9.2/gcc_64/lib/libQt5Core.so.5.9.2)
==13969==    by 0x68B5056: QTimer::timeout(QTimer::QPrivateSignal) (in /home/tim/Qt/5.9.2/gcc_64/lib/libQt5Core.so.5.9.2)
==13969==    by 0x68B5327: QTimer::timerEvent(QTimerEvent*) (in /home/tim/Qt/5.9.2/gcc_64/lib/libQt5Core.so.5.9.2)
==13969==    by 0x68A977A: QObject::event(QEvent*) (in /home/tim/Qt/5.9.2/gcc_64/lib/libQt5Core.so.5.9.2)
==13969==    by 0x57775EB: QApplicationPrivate::notify_helper(QObject*, QEvent*) (in /home/tim/Qt/5.9.2/gcc_64/lib/libQt5Widgets.so.5.9.2)
==13969==    by 0x577EA16: QApplication::notify(QObject*, QEvent*) (in /home/tim/Qt/5.9.2/gcc_64/lib/libQt5Widgets.so.5.9.2)
==13969==    by 0x687E2B7: QCoreApplication::notifyInternal2(QObject*, QEvent*) (in /home/tim/Qt/5.9.2/gcc_64/lib/libQt5Core.so.5.9.2)
==13969==    by 0x68D008D: QTimerInfoList::activateTimers() (in /home/tim/Qt/5.9.2/gcc_64/lib/libQt5Core.so.5.9.2)
==13969==  If you believe this happened as a result of a stack
==13969==  overflow in your program's main thread (unlikely but
==13969==  possible), you can try to increase the size of the
==13969==  main thread stack using the --main-stacksize= flag.
==13969==  The main thread stack size used in this run was 8388608.
==13969== 
==13969== Events    : Ir
==13969== Collected : 1403356087
==13969== 
==13969== I   refs:      1,403,356,087
`

2TallTim, 2017-10-22:
Sure, I'll give that a go. The output from valgrind might actually be useful so here it is:
```tim@7559:~/Code/ao/build/gui$ valgrind --tool=callgrind ./Studio
==13969== Callgrind, a call-graph generating cache profiler
==13969== Copyright (C) 2002-2017, and GNU GPL'd, by Josef Weidendorfer et al.
==13969== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
==13969== Command: ./Studio
==13969== 
==13969== For interactive control, run 'callgrind_control -h'.
==13969== brk segment overflow in thread #7: can't grow to 0x4a51000
==13969== (see section Limitations in user manual)
==13969== NOTE: further instances of this message will not be shown
QWidget::setWindowModified: The window title does not contain a '[*]' placeholder
==13969== Warning: noted but unhandled ioctl 0x6458 with no size/direction hints.
==13969==    This could cause spurious value errors to appear.
==13969==    See README_MISSING_SYSCALL_OR_IOCTL for guidance on writing a proper wrapper.
==13969== 
==13969== Process terminating with default action of signal 11 (SIGSEGV)
==13969==  Access not within mapped region at address 0x8
==13969==    at 0x5FBB730: QOpenGLContext::shareGroup() const (in /home/tim/Qt/5.9.2/gcc_64/lib/libQt5Gui.so.5.9.2)
==13969==    by 0x6247CE3: ??? (in /home/tim/Qt/5.9.2/gcc_64/lib/libQt5Gui.so.5.9.2)
==13969==    by 0x6247E86: ??? (in /home/tim/Qt/5.9.2/gcc_64/lib/libQt5Gui.so.5.9.2)
==13969==    by 0x153CE5: View::redrawPicker() (qopenglfunctions.h:1244)
==13969==    by 0x68A8B95: QMetaObject::activate(QObject*, int, int, void**) (in /home/tim/Qt/5.9.2/gcc_64/lib/libQt5Core.so.5.9.2)
==13969==    by 0x68B5056: QTimer::timeout(QTimer::QPrivateSignal) (in /home/tim/Qt/5.9.2/gcc_64/lib/libQt5Core.so.5.9.2)
==13969==    by 0x68B5327: QTimer::timerEvent(QTimerEvent*) (in /home/tim/Qt/5.9.2/gcc_64/lib/libQt5Core.so.5.9.2)
==13969==    by 0x68A977A: QObject::event(QEvent*) (in /home/tim/Qt/5.9.2/gcc_64/lib/libQt5Core.so.5.9.2)
==13969==    by 0x57775EB: QApplicationPrivate::notify_helper(QObject*, QEvent*) (in /home/tim/Qt/5.9.2/gcc_64/lib/libQt5Widgets.so.5.9.2)
==13969==    by 0x577EA16: QApplication::notify(QObject*, QEvent*) (in /home/tim/Qt/5.9.2/gcc_64/lib/libQt5Widgets.so.5.9.2)
==13969==    by 0x687E2B7: QCoreApplication::notifyInternal2(QObject*, QEvent*) (in /home/tim/Qt/5.9.2/gcc_64/lib/libQt5Core.so.5.9.2)
==13969==    by 0x68D008D: QTimerInfoList::activateTimers() (in /home/tim/Qt/5.9.2/gcc_64/lib/libQt5Core.so.5.9.2)
==13969==  If you believe this happened as a result of a stack
==13969==  overflow in your program's main thread (unlikely but
==13969==  possible), you can try to increase the size of the
==13969==  main thread stack using the --main-stacksize= flag.
==13969==  The main thread stack size used in this run was 8388608.
==13969== 
==13969== Events    : Ir
==13969== Collected : 1403356087
==13969== 
==13969== I   refs:      1,403,356,087
```

2TallTim, 2017-10-24:
I just ran the tests with build 48f58c5 and I'm seeing one fail, here's the output. I'm not sure if this has to do with this issue, but perhaps it's symptomatic of a problem with my configuration?
```/home/tim/Code/ao/ao/test/feature.cpp:45: FAILED:
  REQUIRE( a.push({-1, -1, -1}) == false )
with expansion:
  true == false
```

mkeeter, 2017-10-24:
Nope, that's an unrelated bug that fails on Linux for unknown reasons – I'm intending to check it out, but it's far from the text editor / GUI experience.

Since this is a graphics-ish issue, have you tried using your vendor's binary GPU drivers?

2TallTim, 2017-10-25:
I'm using the latest nvidia-384 proprietary drivers. I've been poking through the code trying to figure it out, without any real success.

Niffler, 2017-10-27:
Hi, I'm having the same issue on Ubuntu 17.10. I worked around this problem by switching back to Xorg as display server. Everything seems to work fine now and I can no longer experience one of the described bugs.

2TallTim, 2017-10-27:
Fix confirmed.

closed 2017-10-27 by 2TallTim