libfive/libfive #60: Crash on exit

nickodell, 2017-10-23:
Reproduction steps:

1. Open Studio
2. Close Studio by clicking x in upper right corner.

This can trigger one of two errors.

Error message type 1:

```
QWidget::setWindowModified: The window title does not contain a '[*]' placeholder
Studio: /usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:365: Eigen::DenseCoeffsBase<Derived, 1>::Scalar& Eigen::DenseCoeffsBase<Derived, 1>::operator()(Eigen::Index, Eigen::Index) [with Derived = Eigen::Array<float, -1, 256, 1>; Eigen::DenseCoeffsBase<Derived, 1>::Scalar = float; Eigen::Index = long int]: Assertion `row >= 0 && row < rows() && col >= 0 && col < cols()' failed.
Aborted
```

Stack trace type 1:

```
Thread 46 "Thread (pooled)" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffb77fe700 (LWP 26198)]
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff43643fa in __GI_abort () at abort.c:89
#2  0x00007ffff435be37 in __assert_fail_base (fmt=<optimized out>, 
    assertion=assertion@entry=0x7ffff77b3cc0 "row >= 0 && row < rows() && col >= 0 && col < cols()", 
    file=file@entry=0x7ffff77b3c88 "/usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h", line=line@entry=365, 
    function=function@entry=0x7ffff77b50c0 <Eigen::DenseCoeffsBase<Eigen::Array<float, -1, 256, 1, -1, 256>, 1>::operator()(long, long)::__PRETTY_FUNCTION__> "Eigen::DenseCoeffsBase<Derived, 1>::Scalar& Eigen::DenseCoeffsBase<Derived, 1>::operator()(Eigen::Index, Eigen::Index) [with Derived = Eigen::Array<float, -1, 256, 1>; Eigen::DenseCoeffsBase<Derived, "...) at assert.c:92
#3  0x00007ffff435bee2 in __GI___assert_fail (assertion=0x7ffff77b3cc0 "row >= 0 && row < rows() && col >= 0 && col < cols()", 
    file=0x7ffff77b3c88 "/usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h", line=365, 
    function=0x7ffff77b50c0 <Eigen::DenseCoeffsBase<Eigen::Array<float, -1, 256, 1, -1, 256>, 1>::operator()(long, long)::__PRETTY_FUNCTION__> "Eigen::DenseCoeffsBase<Derived, 1>::Scalar& Eigen::DenseCoeffsBase<Derived, 1>::operator()(Eigen::Index, Eigen::Index) [with Derived = Eigen::Array<float, -1, 256, 1>; Eigen::DenseCoeffsBase<Derived, "...) at assert.c:101
#4  0x00007ffff75ca3bd in Eigen::DenseCoeffsBase<Eigen::Array<float, -1, 256, 1, -1, 256>, 1>::operator() (this=0x7fffc8015ef8, 
    row=3355523968, col=0) at /usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:364
#5  0x00007ffff75ca0a9 in Kernel::ArrayEvaluator::set (this=0x7fffc8015ee0, p=..., index=0)
    at /home/nick/ao/ao/src/../include/ao/eval/eval_array.hpp:46
#6  0x00007ffff765e8fe in Kernel::XTree<3u>::XTree(Kernel::XTreeEvaluator*, Kernel::Region<3u>, double, double, bool, std::atomic<bool>&)::{lambda(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, unsigned long)#8}::operator()(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, unsigned long) const (__closure=0x7fffb77eb6a0, v=..., i=0) at /home/nick/ao/ao/src/render/brep/xtree.cpp:341
#7  0x00007ffff765bf9c in Kernel::XTree<3u>::XTree (this=0x7fff78004360, eval=0x7fffc8015ec0, region=..., min_feature=0.10000000149011612, 
    max_err=1e-08, multithread=false, cancel=...) at /home/nick/ao/ao/src/render/brep/xtree.cpp:387
#8  0x00007ffff765ad14 in Kernel::XTree<3u>::XTree (this=0x7fff780034e0, eval=0x7fffc8015ec0, region=..., min_feature=0.10000000149011612, 
    max_err=1e-08, multithread=false, cancel=...) at /home/nick/ao/ao/src/render/brep/xtree.cpp:167
#9  0x00007ffff765ad14 in Kernel::XTree<3u>::XTree (this=0x7fff78006260, eval=0x7fffc8015ec0, region=..., min_feature=0.10000000149011612, 
    max_err=1e-08, multithread=false, cancel=...) at /home/nick/ao/ao/src/render/brep/xtree.cpp:167
#10 0x00007ffff765ad14 in Kernel::XTree<3u>::XTree (this=0x7fff78002ca0, eval=0x7fffc8015ec0, region=..., min_feature=0.10000000149011612, 
    max_err=1e-08, multithread=false, cancel=...) at /home/nick/ao/ao/src/render/brep/xtree.cpp:167
#11 0x00007ffff765ad14 in Kernel::XTree<3u>::XTree (this=0x7fff78002aa0, eval=0x7fffc8015ec0, region=..., min_feature=0.10000000149011612, 
    max_err=1e-08, multithread=false, cancel=...) at /home/nick/ao/ao/src/render/brep/xtree.cpp:167
#12 0x00007ffff765ad14 in Kernel::XTree<3u>::XTree (this=0x7fff78005b40, eval=0x7fffc8015ec0, region=..., min_feature=0.10000000149011612, 
    max_err=1e-08, multithread=false, cancel=...) at /home/nick/ao/ao/src/render/brep/xtree.cpp:167
#13 0x00007ffff765ad14 in Kernel::XTree<3u>::XTree (this=0x7fff78001a80, eval=0x7fffc8015ec0, region=..., min_feature=0.10000000149011612, 
    max_err=1e-08, multithread=false, cancel=...) at /home/nick/ao/ao/src/render/brep/xtree.cpp:167
#14 0x00007ffff765ad14 in Kernel::XTree<3u>::XTree (this=0x7fff780008e0, eval=0x7fffc8015ec0, region=..., min_feature=0.10000000149011612, 
    max_err=1e-08, multithread=false, cancel=...) at /home/nick/ao/ao/src/render/brep/xtree.cpp:167
#15 0x00007ffff765e654 in Kernel::XTree<3u>::XTree(Kernel::XTreeEvaluator*, Kernel::Region<3u>, double, double, bool, std::atomic<bool>&)::{lambda()#1}::operator()() const (__closure=0x7fffc0003198) at /home/nick/ao/ao/src/render/brep/xtree.cpp:152
#16 0x00007ffff771f3c8 in std::_Bind_simple<Kernel::XTree<3u>::XTree(Kernel::XTreeEvaluator*, Kernel::Region<3u>, double, double, bool, std::atomic<bool>&)::{lambda()#1} ()>::_M_invoke<>(std::_Index_tuple<>) (this=0x7fffc0003198) at /usr/include/c++/6/functional:1391
#17 0x00007ffff77184cd in std::_Bind_simple<Kernel::XTree<3u>::XTree(Kernel::XTreeEvaluator*, Kernel::Region<3u>, double, double, bool, std::atomic<bool>&)::{lambda()#1} ()>::operator()() (this=0x7fffc0003198) at /usr/include/c++/6/functional:1380
#18 0x00007ffff77072db in std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<Kernel::XTree<3u>*>, std::__future_base::_Result_base::_Deleter>, std::_Bind_simple<Kernel::XTree<3u>::_Bind_simple(Kernel::XTreeEvaluator*, Kernel::Region<3u>, double, double, bool, std::atomic<bool>&)::{lambda()#1} ()>, Kernel::XTree<3u>*>::operator()() const (this=0x7fffb77fddf0)
    at /usr/include/c++/6/future:1325
#19 0x00007ffff76fcdec in std::_Function_handler<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> (), std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<Kernel::XTree<3u>*>, std::__future_base::_Result_base::_Deleter>, std::_Bind_simple<Kernel::XTree<3u>::_Bind_simple(Kernel::XTreeEvaluator*, Kernel::Region<3u>, double, double, bool, std::atomic<bool>&)::{lambda()#1} ()>, Kernel::XTree<3u>*> >::_M_invoke(std::_Any_data const&) (__functor=...) at /usr/include/c++/6/functional:1717
#20 0x00007ffff7635f3f in std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>::operator()() const (this=0x7fffb77fddf0) at /usr/include/c++/6/functional:2127
#21 0x00007ffff7634b57 in std::__future_base::_State_baseV2::_M_do_set(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*) (this=0x7fffc0003160, __f=0x7fffb77fddf0, __did_set=0x7fffb77fdd77)
    at /usr/include/c++/6/future:533
#22 0x00007ffff763cb3d in std::__invoke_impl<void, void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*>(std::__invoke_memfun_deref, void (std::__future_base::_State_baseV2::*&&)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*&&, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*&&, bool*&&) (__f=<unknown type in /home/nick/ao/build/ao/src/libao-kernel.so, CU 0x581789, DIE 0x65a9f7>, 
    __t=<unknown type in /home/nick/ao/build/ao/src/libao-kernel.so, CU 0x581789, DIE 0x5a45ef>, 
    __args#0=<unknown type in /home/nick/ao/build/ao/src/libao-kernel.so, CU 0x581789, DIE 0x65aa03>, 
    __args#1=<unknown type in /home/nick/ao/build/ao/src/libao-kernel.so, CU 0x581789, DIE 0x65aa0f>) at /usr/include/c++/6/functional:227
[... cut ...] ::XTree<3u>*>::_Async_state_impl(Kernel::XTree<3u>::XTree(Kernel::XTreeEvaluator*, Kernel::Region<3u>, double, double, bool, std::atomic<bool>&)::{lambda()#1} (&&)())::{lambda()#1} ()> >::_M_run() (this=0x7fffb80028c0) at /usr/include/c++/6/thread:197
#34 0x00007ffff4ca3e6f in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#35 0x00007ffff6578494 in start_thread (arg=0x7fffb37fe700) at pthread_create.c:333
#36 0x00007ffff4418aff in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:97
```

mkeeter, 2017-10-23:
Did this just start happening in that revision?  If you go back to 9d57a23d8951aea5588cc791e1518bc0a064985b, does it still crash?

Does this only happen when rendering is still in progress when you quit?

This was to fix a crash that I was noticing where if you quit while a render task was still in progress, it would hold off on deleting the `Shape` until after the OpenGL context was destroyed, so it would crash when it tried to delete the buffers, etc.

nickodell, 2017-10-23:
9d57a2 still crashes, with both kinds of errors. I think it segfaults more.

nickodell, 2017-10-23:
There is definitely a connection between rendering and the crash. It crashed 5 out of 5 times when I interrupted the render, and 0 out of 5 when I didn't.

mkeeter, 2017-10-23:
How about if you wait for rendering to finish (when the spinner disappears from the lower right)?

These crashes seem to happen in the meshing thread, so I'm curious how it's getting shut down... 

Also, can you see if 48f58c53034ea6bd32eb85f6366c91f828a7b3cb helps?

nickodell, 2017-10-23:
Fix confirmed.

closed 2017-10-23 by nickodell