libfive/libfive #61: Progress meter

omgitsraven, 2017-10-24:
I've currently got a pretty complex model (lots of smooth-minimum use) with lots of fiddly details that call for me to render it at a very high resolution (before I decimate it back down to something reasonable elsewhere).

To be specific, once I crank the quality up, I'm staring at that spinning wheel for something like eight hours before it's ready. And, that's fine if that's how long it takes! But, given that mesh generation can take that long, it would be nice if there could be some kind of progress meter instead of just a spinning wheel—even if not a time estimate, just a percentage indication of how far through the bounds it's calculated (and/or how many iterations it's gone through, I'm not sure exactly how dual contouring works to be honest :) ).

(While I'm at it, just as a quality-of-life thing, it might be nice to be able to opt into a confirmation dialog box if you're about to drop something on the keyboard and make the process have to start over because it responded automatically to your changes.... It hasn't happened to me yet, but I live in terror of the day it does ;) )

mkeeter, 2017-10-25:
Wow!  I guess smooth-minimums are expensive; I wonder if there's a way to wrap them in a standard min/max block some distance out from the blend to take advantage of tree pruning.

Anyways, I think that the right place to put a progress meter is in the exporting dialog, since that blocks until meshing is complete (right now, it just shows a blank progress bar).  Using the export dialog will also alleviate your fears of accidentally restarting, because it ignores keyboard presses (other than escape).

omgitsraven, 2017-10-25:
Oh, fantastic! I didn't even bother trying exporting before it was ready because I assumed it would just export the last mesh… That's good to know, then :)