libfive/libfive #62: regression: ‘QMutex’ does not name a type

nickodell, 2017-11-09:
Revision 5f2aadf does not compile for me. It fails with this error:

```
[ 74%] Building CXX object gui/CMakeFiles/Studio.dir/Studio_automoc.cpp.o
In file included from /home/nick/ao/gui/src/interpreter.cpp:21:0:
/home/nick/ao/gui/include/gui/interpreter.hpp:90:5: error: ‘QMutex’ does not name a type
     QMutex mutex;
     ^~~~~~
In file included from /home/nick/ao/gui/src/window.cpp:30:0:
/home/nick/ao/gui/include/gui/interpreter.hpp:90:5: error: ‘QMutex’ does not name a type
     QMutex mutex;
     ^~~~~~
In file included from /home/nick/ao/build/gui/Studio_automoc.dir/moc_interpreter_JL3ZW3PCB4US6T.cpp:9:0,
                 from /home/nick/ao/build/gui/Studio_automoc.cpp:7:
/home/nick/ao/build/gui/Studio_automoc.dir/../../../gui/include/gui/interpreter.hpp:90:5: error: ‘QMutex’ does not name a type
     QMutex mutex;
     ^~~~~~
/home/nick/ao/gui/src/interpreter.cpp: In member function ‘void _Interpreter::eval()’:
/home/nick/ao/gui/src/interpreter.cpp:107:28: error: ‘mutex’ was not declared in this scope
         QMutexLocker lock(&mutex);
                            ^~~~~
/home/nick/ao/gui/src/interpreter.cpp: In member function ‘void Interpreter::onScriptChanged(QString)’:
/home/nick/ao/gui/src/interpreter.cpp:270:40: error: ‘class _Interpreter’ has no member named ‘mutex’
         QMutexLocker lock(&interpreter.mutex);
                                        ^~~~~
gui/CMakeFiles/Studio.dir/build.make:542: recipe for target 'gui/CMakeFiles/Studio.dir/src/window.cpp.o' failed
```
However, 7dac3c7 compiles and runs successfully.

mkeeter, 2017-11-10:
Probably just a missing `#include <QMutex>`; see if 7d6ba9ae9155ccd6eed6a79bf2bf7cc2bd2c4079 fixes this issue.

nickodell, 2017-11-10:
Fix confirmed.

closed 2017-11-10 by nickodell