libfive/libfive #67: Changing render parameters doesn't cancel renders in progress

omgitsraven, 2017-12-06:
In short: if I open the render options window, and punch in a resolution value that will take 60 seconds to render, then immediately change it to something that should take 1 second to render, I will still have to wait 61 seconds to see it.

Changing any of the parameter values should immediately cancel any renders in progress, to start the new render as soon as possible (since the user has already indicated they're not interested in whatever the previous render was). In the case of renders that take hours, I didn't realise I was making myself wait for far longer than I needed to, since I was switching from my 'preview' settings to my 'final' settings by changing both the resolution value and the quality value, which I didn't realise was essentially starting TWO rendering jobs, back to back, each taking several hours to complete.

I can work around this by changing the values, saving, quitting, and relaunching Studio, but it would be nice not to have to bother -- and extra nice for people who haven't realised that this is what Studio is doing, yet :)

mkeeter, 2017-12-07:
Good idea!  I already cancel rendering in a few other cases, and I think the commit above should make this work.

omgitsraven, 2017-12-08:
That did it :) Thanks!

closed 2017-12-08 by omgitsraven