libfive/libfive #72: Longer example/demo?

omgitsraven, 2017-12-28:
I keep stumbling across screenshots on your twitter that mention enormous, important features that I had no idea that Ao included :) Particularly things like define-shape, or prefacing a value with # so you can change it with the mouse...

There's also plenty of things that I see listed in the shape reference, that I have no idea how to actually use in practice, like what I'm supposed to be using for the "shape" argument in extrude-z... (That part might just be because I'm a beginner at Scheme?)

I know development is still early, but a simple example scene (or a collection of them?) that just shows the most important of these actually put into practice somehow (even just on cubes or whatever) would be extremely helpful for figuring out what can be done with this program!

omgitsraven, 2017-12-30:
Oh, wow—I only just now noticed that there's several more examples in ao/gui/examples. That might be worth mentioning in the readme, or in the comments on the tutorial file :)

yunfan, 2018-01-16:
agree, also need full examples

i found that high level example in the official site `example` dont work, need to first use the under low level example to define `sphere` first

i'd use openscad, i dislike its language, but i am not familliar with guile too. so i really need some full example to gain the passion to learn such cool tool :D

mkeeter, 2018-01-16:
@yunfan thanks for the hint – I updated the website to work with the new `sphere` argument order (it used to be `sphere center r`, and is now `sphere r [optional center]`).

yunfan, 2018-01-17:
@mkeeter also another feature that openSCAD attracted me is its tiny size and portable . i hope Studio could achieve that too. maybe you could try using forth to act as the embeding language ?  (just a thinking)

zoe1337, 2019-05-01:
I use OpenSCAD regularly and want to change to libfive, because it is way more powerful. I wish there would be a longer tutorial like OpenSCAD has on wikibooks - that makes it easier to start learning.

I would like to help, where should we start?