libfive/libfive #73: Argument order for shape-taking (nesting) functions

omgitsraven, 2017-12-28:
So, I'll be honest: I only recently discovered the "Shape reference" in the help menu (was it there all along?), and prior to that I had written a lot of my own versions of those same functions, not knowing they already existed.

I'm sure there's a lot wrong with my approach (I know barely anything about Scheme), but there's one thing about my version that I miss, and it's that the values came before the shape, which lent itself to writing things like:

```
(translate x y z 0.1 0.2 0.3 (lambda (x y z)
(rot3dY x y z 0.1 (lambda (x y z)
(scale x y z 1.0 1.5 1.0 (lambda (x y z)
    (mysphere x y z 0.2)
)) )) ))
```

I won't miss all those verbose 'x y z' and 'lambda's that I couldn't figure out how to get rid of, but writing it this way let me keep the values on the same line as the function that they're referring to, whereas with things like 'move' I'd have to write it this way:

```
(move
(rotate-y
(scale-y
    (mysphere x y z 0.2)
1.5)
0.1)
#[0.1 0.2 0.3])
```

where the values are a pretty big distance away from the name describing what they are—and also in reverse order.

I understand that this would be a breaking change, but I figured I'd bring it up anyway (if only to also implicitly ask if there's some more reasonable way of approaching this that I'm unaware of?)

mkeeter, 2017-12-28:
That's an interesting idea!

The main reason why I chose the current API structure is for functions that take optional arguments.  For example, `rotate-z` takes an optional center to rotate about, which needs to be at the back of the argument list.

Since this is Scheme, I think that that the right solution is some kind of rewriting macro.  Consider the following hypothetical code:

```
(sequence (mysphere x y z 0.2)
          (scale-y 1.5)
          (rotate-y 0.1)
          (move #[0.1 0.2 0.3]))
```
which both has arguments near the transformations, *and* puts things in the actual order of execution!

How does that sound?

omgitsraven, 2017-12-28:
That seems great! (I'm still not quite grasping exactly how the flow of information works in Scheme if something like that is possible, apparently...)

Also, if that turns out to be too much work, and optional arguments are the only issue, you could always just have multiple differently-named functions instead, like `(rotate-z 0.1 myshape)` vs (`rotate-z-about 0.1 #[0.1 0 0] myshape)`

mkeeter, 2017-12-29:
Okay, `sequence` is in, and works just like I showed above.

Rewriting syntax on the fly is amazing 😁

omgitsraven, 2017-12-29:
Fantastic! That works beautifully. Thank you :)

closed 2017-12-29 by omgitsraven