libfive/libfive #78: Something weird with expt

omgitsraven, 2017-12-29:
Sorry I can't be more specific; to be honest I'm only about 50% sure that this is a bug in Ao and not a transcription error on my part?

I was comparing the look of various blending operations (the `blend` built into Ao, and the three `smin`s found [here](http://iquilezles.org/www/articles/smin/smin.htm)), but something very strange is happening with the 'power smooth' one (the fourth one). I don't know if this is something I've done wrong, or if there's just something about it that's incompatible with how Ao works, but it seemed enough like a bug that I thought I'd run it by you just in case...

[smintests.ao.zip](https://github.com/mkeeter/ao/files/1593837/smintests.ao.zip)

mkeeter, 2017-12-29:
Looking at the code, I think the issue is that you're turning a two-sided distance function (where negative values are inside and positive values are outside) into a one-sided distance function, because `expt(expt(i, 16), 1/16)` loses sign (in the same way that `sqrt(x^2)` maps everything to positive values).

Experimenting with odd exponents did produce crashes, though, so I'll look into this a little more.

mkeeter, 2017-12-31:
Okay, I fixed the odd exponent issue (turns out `pow` doesn't work on negative values with fractional exponents), but the fundamental issue is still signed vs unsigned distance fields – I don't think that particular `smin` is going to work, just because of how it's constructed.

closed 2017-12-31 by mkeeter

omgitsraven, 2017-12-31:
Fair enough, thanks!