libfive/libfive #79: Make origin optional for shape-generation functions

omgitsraven, 2017-12-30:
Typically, when I'm working with shapes in Ao, their transformations are so nested in scaling, rotating, etc, that I usually wind up passing `#[0 0 0]` for their positions, and translating them elsewhere in the `sequence`. So personally, I think it would be more convenient if the typical argument order for the shape-generating functions were more like `(sphere 0.5 #[0 0 0])` rather than `(sphere #[0 0 0] 0.5)`, so that the `#[0 0 0]` could be an optional argument, like it is in the transform functions.

I'm willing to do the pull request myself, I just wanted to bring it up here first and hear what you think before I bothered with it, since I know it's a breaking change unfortunately...

mkeeter, 2017-12-30:
Okay, I think this makes sense – if you make a PR, I'll look it over and bring it in.

omgitsraven, 2017-12-30:
Minor question: how attached to your code formatting are you? Personally I find it easier to write Scheme with lots of linebreaks to make the nesting parentheses clearer; would it be annoying for you if I just committed it that way, rather than cleaning it up to match your style?

mkeeter, 2017-12-30:
For these changes, it would be good to match the existing formatting so that `git diff` isn't full of spurious lines (since the only real change will be to the argument order and docstrings).  For new functions, format them however you're most comfortable.

omgitsraven, 2017-12-30:
OK, I'm sorry to trouble you after I said I could take care of it, but -- it's crashing on launch and I have no idea why. https://github.com/omgitsraven/ao/commits/patch-2 Are there any obvious mistakes that leap out to you?

mkeeter, 2017-12-31:
The issue is that to use `#:optional`, you have to use `define*` (because it's a [Guile extension](https://www.gnu.org/software/guile/manual/html_node/lambda_002a-and-define_002a.html#lambda_002a-and-define_002a)).  I fixed this and merged everything in, so we should be all set.

closed 2017-12-31 by mkeeter

omgitsraven, 2017-12-31:
Ahhh, thank you! :)