libfive/libfive #81: Return multiple shapes without using union?

omgitsraven, 2017-12-30:
So I'm starting to get the idea of why putting shapes into as many separate `lambda-shape`s as possible is a good approach—it only has to recalculate the field for whatever shape you're making changes in, which makes iterating so much faster, which was one of the biggest problems I had with my old approach. But one thing I miss about my old approach was, I could 'nest' parts of the scene, just purely for readability's sake, like:

(read this from the bottom up...)
```
(define (myBody x y z)
  ;etc...
)

(define (mySkull x y z)
  (scale x y z 1.3 1 1 (lambda (x y z)
    (mysphere x y z 0.5)
  ))
)
;etc...

(define (myHead x y z)
  (min
    (mySkull x y z)
    (myFace x y z)
    (myHair x y z)
  )
)

(lambda-shape (x y z)
  (min
    (myHead x y z)
    (myBody x y z)
  )
)
```

And of course I can still do this with unions, but then I lose the benefit of rebuilding simpler shapes on each change. And I know I can put it all inline and describe each section with comments, but that spreads out all the different parts so far, that it can be hard to keep track of the flow of how things are connected...

I guess I'm just trying to brainstorm a method for things to be written nested in some way -- especially if they can literally be nested inside transformations too, instead of having to make a transform-producing function and apply it to them all individually -- without losing the beneficial separation of having them all be separate calls happening on the root? Like, some kind of macro that will call a `sequence` on all the `lambda-shape`s that are passed to its subsequent arguments, and another that will let you return several `lambda-shape`s within a single function (or pass it a function to provide them as arguments to), that will make them into a list for the first macro?

I'm really just spitballing here, but your solution in coming up with `sequence` worked so well that I thought I'd at least run it by you, heh :)

mkeeter, 2017-12-30:
The standard Scheme strategy for multiple return values is the [`values` keyword](https://www.gnu.org/software/guile/manual/html_node/Multiple-Values.html).  I've added support for multiple return values, so if you do something like
```scheme
(values
  (sphere #[0 0 0] 1)
  (sphere #[1 0 0] 1))
```
both spheres will be rendered.

I also added a new operator, `values-from`, which flattens out returned values, so you could do something like

```scheme
(define (head)
  (values (sphere #[0 0 0] 1)
          (sphere #[0.5 0.8 0.5] 0.2)
          (sphere #[-0.5 0.8 0.5] 0.2)))

(define (model)
  (values-from (head) ;flattens values from head
               (cylinder-z #[0 0 -1] 1 1)))

(model)
```

It's a bit finicky, since you can only have **functions** that return multiple values (e.g. `head` is a zero-argument function above), but seems to work.

I'm not sure about multi-value transforms – do you have any suggested syntax for that kind of operation?

mkeeter, 2017-12-30:
On second thought, I just made `sequence` work with multiple values as well, so now you can do

```
(define (q)
  (values (sphere #[0 0 0] 1)
          (sphere #[1 0 0] 0.5)))

(sequence (q)
  (move #[0 0 1])
)
```

Can you try this out and see if it works for you?

omgitsraven, 2017-12-30:
Brilliant. `sequence` accepting `values` being passed multiple `shape`s works perfectly, thank you so much :)

I'm not sure I understand what `values-from` is for, though? Where would I use `(values-from (head))` rather than just `(head)`? (Or is there no longer a reason to now that `sequence` can take a `values`?)

closed 2017-12-30 by omgitsraven

mkeeter, 2017-12-31:
`values-from` is like Python's `yield from` (vs regular `yield`).

For example, `(values (values 1 2 3) 4 5 6)` will just return `1 4 5 6`, because the nested values aren't flattened out; `(values-from (values 1 2 3) 4 5 6)` will return `1 2 3 4 5 6`

omgitsraven, 2017-12-31:
Is there ever any reason to use `values` instead of `values-from`, or is it safe to just always use the latter rather than having to remember to step up to it if I'm nesting another `values`?

mkeeter, 2017-12-31:
You can always use `values-from`.  The only reason not to use it is performance, which probably isn't an issue for any reasonable-size model.