libfive/libfive #85: Strange results with autobounds

omgitsraven, 2018-01-07:
I'm liking the new autobounds feature! But I keep accidentally running into situations where it makes the bounds enormous for no clear reason...

I actually came here a few minutes ago with an example, only to find that you'd already committed a fix for that one -- but just now I've found another, that even the new verison doesn't accomodate, so, here goes ;)

```
(sequence
  (sphere 1)
  (rotate-x (/ pi 2))
  (rotate-x 1.15)
  (scale-y 0.625)
  (rotate-y -0.1)
)
```

Commenting out any one of the transformations is enough to give it reasonable bounds, but somehow the combination of them produces an enormous bounding box instead. (Never mind that I'm rotating a sphere; this is the stripped-down version of how I found it, that still produces the bug.)

mkeeter, 2018-01-08:
Hah, it's worse than you think – the large bounding box is actually the system falling back to the default bounds of ±10 on each axis, because `findBounds` estimates the bounds to be approximately ±3.4e38.

I'll take a look and see if I can track this down, though it may need a smarter f-rep tree pruning system (e.g. something that automatically collapses affine shapes; I had something like this [earlier](http://www.mattkeeter.com/blog/2016-03-20-affine/) but it needed too much manual hand-holding).

mkeeter, 2018-01-15:
Okay, this is now fixed:
```(shape-bounds
  (sequence
    (sphere 1)
    (rotate-x (/ pi 2))
    (rotate-x 1.15)
    (scale-y 0.625)
    (rotate-y -0.1)
))
```
gives
```
(#[-1.01629638671875 -0.664031982421875 -1.07098388671875] .
 #[1.01629638671875 0.664031982421875 1.07098388671875])
```
which looks reasonable.

This kind of affine rewriting isn't magical, and I suspect you'll still be able to find cases that it fails on, but it's a step in the right direction.

closed 2018-01-15 by mkeeter