libfive/libfive #88: Build failure on OpenBSD 6.2

alexshendi, 2018-01-15:
The build fails on OpenBSD 6.2 / amd64 with the following error message:
#error Boost.Numeric.Interval: Please specify rounding control mechanism.

I have tried to add "-D__USE_ISOC99=1" to CFLAGS, but had no success.

I had installed the following packages:
- Boost 1.58
- Qt 5.9.1
- Eigen 3.2.2
- Guile 2.2.3
- CMake 3.9.3
- C/C++ compiler clang 4.0.0

I would be grateful for any help. Many thanks in advance.

mkeeter, 2018-01-16:
I'm not sure exactly what you tried, but I'd recommend adding `-D__USE_ISOC99` to [this line](https://github.com/libfive/libfive/blob/master/CMakeLists.txt#L12) and see if that works; otherwise, you'll have to debug it yourself, as I don't have the resources to support OpenBSD.

alexshendi, 2018-01-17:
Hi Matt,

I did as you suggested and modified "CMake Lists.txt" accordingly. Then I had some issues with eigen3 3.2.2 from packages. I removed the eigen3 package and installed 3.4.3 from source. After that I was able to build libfive, libfive-guile and Studio.

Thank you for your help and for creating libfive.

Best Regards,

Alexander 

Am 16. Januar 2018 01:16:39 MEZ schrieb Matt Keeter <notifications@github.com>:
>I'm not sure exactly what you tried, but I'd recommend adding
>`-D__USE_ISOC99` to [this
>line](https://github.com/libfive/libfive/blob/master/CMakeLists.txt#L12)
>and see if that works; otherwise, you'll have to debug it yourself, as
>I don't have the resources to support OpenBSD.

-- 
Diese Nachricht wurde von meinem Android-Mobiltelefon mit K-9 Mail gesendet.

mkeeter, 2018-01-17:
Great, I've tweaked the top-level CMakeLists to add this flag if `CMAKE_SYSTEM_NAME` is `OpenBSD`.

closed 2018-01-17 by mkeeter