libfive/libfive #92: Studio - black screen on startup.

mjgordon, 2018-01-17:
Window with 3d axes and I assume the editor appears for a half second on startup, then the window turns black. The first keypress after this shows "QWidget::setWindowModified: The window title does not contain a '[*]' placeholder" in the terminal. Saving and loading dialog windows will still appear, but using them has no effect on the black window.
"QOpenGLFramebufferObject::toImage() called without a current context" was also printed sometimes after the turning black.

mkeeter, 2018-01-17:
What OS are you using?  Have you updated your graphics drivers (and/or switched to the binary drivers, if applicable)?

mjgordon, 2018-01-17:
Using Ubuntu 17.04, I believe the nvidia provided graphics drivers are being used. After updating, (which did update that driver), running now shows:
```
QOpenGLWidget: Failed to create context
QOpenGLWidget: Failed to create context
QOpenGLWidget: Failed to create context
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed... (repeated)
```
and segfaults upon clicking in the window:
```
QImage::pixel: coordinate (17,396) out of range
Segmentation fault (core dumped)
```

mkeeter, 2018-01-17:
From 15 seconds of Googling, it could be [this issue](https://obsproject.com/forum/threads/failed-to-create-opengl-context-on-ubuntu-14-04.25286/).

Can you run `glxinfo | grep OpenGL` and see what it says?

mkeeter, 2018-01-19:
I've added a startup check to ensure that the application has gotten a sufficiently new OpenGL context.

If this theory is correct, then the error message will be shown when you start the application, giving you an informative message rather than just crashing.  Please re-open this issue if that's not the case!

3dGrabber, 2018-01-23:
Same issue here!   
2e027a1 on Ubuntu, so including the startup check.
I suspect it is a problem with qt (which I have very little experience with).  
Will investigate.  
Reopen?

### Studio output

    QFactoryLoader::QFactoryLoader() checking directory path "/opt/qt5.10/plugins/platforms" ...
    QFactoryLoader::QFactoryLoader() looking at "/opt/qt5.10/plugins/platforms/libqeglfs.so"
    Found metadata in lib /opt/qt5.10/plugins/platforms/libqeglfs.so, metadata=
    {
        "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
        "MetaData": {
            "Keys": [
                "eglfs"
            ]
        },
        "className": "QEglFSIntegrationPlugin",
        "debug": false,
        "version": 330241
    }
    
    
    Got keys from plugin meta data ("eglfs")
    QFactoryLoader::QFactoryLoader() looking at "/opt/qt5.10/plugins/platforms/libqlinuxfb.so"
    Found metadata in lib /opt/qt5.10/plugins/platforms/libqlinuxfb.so, metadata=
    {
        "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
        "MetaData": {
            "Keys": [
                "linuxfb"
            ]
        },
        "className": "QLinuxFbIntegrationPlugin",
        "debug": false,
        "version": 330241
    }
    
    
    Got keys from plugin meta data ("linuxfb")
    QFactoryLoader::QFactoryLoader() looking at "/opt/qt5.10/plugins/platforms/libqminimal.so"
    Found metadata in lib /opt/qt5.10/plugins/platforms/libqminimal.so, metadata=
    {
        "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
        "MetaData": {
            "Keys": [
                "minimal"
            ]
        },
        "className": "QMinimalIntegrationPlugin",
        "debug": false,
        "version": 330241
    }
    
    
    Got keys from plugin meta data ("minimal")
    QFactoryLoader::QFactoryLoader() looking at "/opt/qt5.10/plugins/platforms/libqminimalegl.so"
    Found metadata in lib /opt/qt5.10/plugins/platforms/libqminimalegl.so, metadata=
    {
        "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
        "MetaData": {
            "Keys": [
                "minimalegl"
            ]
        },
        "className": "QMinimalEglIntegrationPlugin",
        "debug": false,
        "version": 330241
    }
[... 128 lines cut ...]
    loaded library "/opt/qt5.10/plugins/platforminputcontexts/libcomposeplatforminputcontextplugin.so"
    QFactoryLoader::QFactoryLoader() checking directory path "/opt/qt5.10/plugins/styles" ...
    QFactoryLoader::QFactoryLoader() checking directory path "/home/eef/build/libfive/build/studio/styles" ...
    QFactoryLoader::QFactoryLoader() checking directory path "/opt/qt5.10/plugins/iconengines" ...
    QFactoryLoader::QFactoryLoader() checking directory path "/home/eef/build/libfive/build/studio/iconengines" ...
    QFactoryLoader::QFactoryLoader() checking directory path "/opt/qt5.10/plugins/accessible" ...
    QFactoryLoader::QFactoryLoader() checking directory path "/home/eef/build/libfive/build/studio/accessible" ...
    
### glxinfo | grep OpenGL

    OpenGL vendor string: NVIDIA Corporation
    OpenGL renderer string: GeForce GTX 660/PCIe/SSE2
    OpenGL core profile version string: 4.5.0 NVIDIA 384.111
    OpenGL core profile shading language version string: 4.50 NVIDIA
    OpenGL core profile context flags: (none)
    OpenGL core profile profile mask: core profile
    OpenGL core profile extensions:
    OpenGL version string: 4.5.0 NVIDIA 384.111
    OpenGL shading language version string: 4.50 NVIDIA
    OpenGL context flags: (none)
    OpenGL profile mask: (none)
    OpenGL extensions:
    OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 384.111
    OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
    OpenGL ES profile extensions:

mkeeter, 2018-01-23:
Could this be a Wayland issue?  [Other folks have seen weird graphic glitches](https://github.com/libfive/libfive/issues/59), and had them go away when they switch to Xorg.

mjgordon, 2018-01-28:
I'm experiencing this problem with i3, which I believe runs on top of xorg (apologies if that's incorrect) so in this case I don't think that's the crux. 
Issue was the same when just logging into normal 'Ubuntu with xorg' as well. 

Also, the 'failed to create context' errors I reported in my first reply appear to have been a fluke on my end, sorry, the original post is still an accurate description of the situation.

simeonveldstra, 2018-02-02:
I too am experiencing this issue.
Commit ce0a221118854c39f48f57aa910fea07dca36014

The window draws at startup, and on a window resize, then goes black almost instantly. 
The menus do work, if you click the correct area of black screen. 
The about box and the Shape Reference window will open and display correctly.
The save changes dialog displays correctly on window destroy sometimes (can't tell if I've edited anything). When the save changes box appears, I also get 
`QWidget::setWindowModified: The window title does not contain a '[*]' placeholder` 
in the console.

I'm running up to date Debian testing, with some packages from unstable hanging  around from when I needed the Nvidia driver from there (card works with >v384 only), notably GCC v7.3
I use the stumpwm window manager over Xorg.
Other OpenGL programs work correctly.

```
? glxinfo | grep OpenGL
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: GeForce GT 1030/PCIe/SSE2
OpenGL core profile version string: 4.5.0 NVIDIA 384.111
OpenGL core profile shading language version string: 4.50 NVIDIA
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 4.5.0 NVIDIA 384.111
OpenGL shading language version string: 4.50 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 384.111
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions:
```

mkeeter, 2018-02-02:
@simeonveldstra Thanks for the detailed report!

The fact that it draws correctly *once* makes me suspicious.  Can you try commenting out `redrawPicker` ([here](https://github.com/libfive/libfive/blob/master/studio/src/view.cpp#L215-L268)) and see if you can resize the window?  (Don't click in the window, because it will just crash 😃 ).

simeonveldstra, 2018-02-02:
@mkeeter No problem.

With the function body commented out, the window draws properly and does not go black.
Typing into the focused window appears in the editor box. Resizing the window works.
Clicking the menus and menu items works. Clicking the editor pane moves the cursor.
Clicking the preview pane does indeed crash.

mkeeter, 2018-02-03:
@simeonveldstra Can you check to see whether the latest commit (970c6f70180181edb9a992afdc0dd43b09d4b048) changes anything?  It's a minor tweak to how we claim and release the OpenGL context.

If that doesn't help, you should do a commenting-out-blocks exercise `redrawPicker` and see if you can isolate a particular action that's cause the redraw issue.

simeonveldstra, 2018-02-03:
Hi Matt,
970c6f70180181edb9a992afdc0dd43b09d4b048 does not solve the problem.

I have narrowed it down to one [line](https://github.com/libfive/libfive/blob/master/studio/src/view.cpp#L244):
`glEnable(GL_DEPTH_TEST);`
With that line commented out, Studio works as expected; no black screen, resize works, spinning an object works, nothing appears broken.

In fact, commenting out the entire stanza has the same effect,
```
    glClearColor(0, 0, 0, 1);
    glClearDepthf(1);
    glClearStencil(0);
    glClear(GL_COLOR_BUFFER_BIT|GL_DEPTH_BUFFER_BIT|GL_STENCIL_BUFFER_BIT);
    glEnable(GL_DEPTH_TEST);
    glViewport(0, 0, camera.size.width(), camera.size.height());
```
with no apparent problems, but the issue appears to be caused by the one line.

mkeeter, 2018-02-03:
That depth test matters, though you wouldn't have seen it in the test you describe.  Consider this design:
```
(sphere #2 #[0 0  0])
(sphere #2 #[0 0  2])
(sphere #2 #[0 0 -2])
```
The depth test when rendering the pick buffer ensures that when you mouse over the three spheres, the correct one gets highlighted; otherwise, it's done based on draw order.

How about disabling `GL_DEPTH_TEST` after the rendering is done?
Can you try dropping `glDisable(GL_DEPTH_TEST)` around [here](https://github.com/libfive/libfive/blob/master/studio/src/view.cpp#L253) and see if that makes a difference?

Otherwise, maybe we need to clean the buffers before running the usual rendering operation, e.g. copying [this chunk of code](https://github.com/libfive/libfive/blob/master/studio/src/view.cpp#L240-L243) to [`paintGL`](https://github.com/libfive/libfive/blob/master/studio/src/view.cpp#L303).

simeonveldstra, 2018-02-03:
Okay,
Adding the glDisable call works as well as commenting out the glEnable call -- no black screen. 
Copying the cleanup code to paintGL makes no difference.

Here's the thing, I'm unable to grab an object and resize it. No matter where the mouse is when I click, all it does is rotate the view. I didn't realize that this was missing at first, until I had another look at the [demo animations](https://libfive.com/studio/).

So it looks like disabling the depth test after drawing the shapes is not the fix either.
I'm slowly getting familiar with the code, this is some sweet software.

mkeeter, 2018-02-04:
Okay, I'll add that `glDisable` call to get rendering working.  It sounds like the pick buffer still isn't working; I've given some advice on debugging in [this thread](https://github.com/libfive/libfive/issues/93), which you may find a good starting point.

I'm glad you're getting familiar with the codebase!  If you're interested in contributing, the [Help wanted tag](https://github.com/libfive/libfive/issues?q=is%3Aissue+is%3Aopen+label%3A%22help+wanted%22) has issues that I think could be good starting points.

closed 2018-02-04 by mkeeter

simeonveldstra, 2018-02-04:
That you are adding the glDisable call implies that it does not affect the pick buffer. I had assumed the two problems were related. That is good news.
I will continue investigating my pick buffer issue.
Thanks Matt.