libfive/libfive #93: issue with ui manipulation of vars

jobez, 2018-01-17:
### Description

[Description of the issue]

I am unable to manipulate vars from the UI interface.
[Steps to reproduce]
Load tutorial project, prefixing a number with a '#',  and attempt to interact with shape
[What did you expect to happen?  What happened?  All the time?]
Looking at the logic, I expected the cursor to be responsive and change to the `Qt::OpenHandCursor` when the system has a hover target. This never happens.

[Are there any error messages printed to the console?]
No
### Versions

- Operating system: [NixOS]
- `libfive` commit hash: [9687c9d]
- Qt version (if applicable): [5.9]

### Additional Information
I have a 4k display. I'm thinking this might be relevant when you move from mouse coords space to grabbing a pixel from the `pick_img`? I didn't find much info to go on when I explored qt+hi-dpi monitors so I thought I'd reach out and see if you had any thoughts.

follower, 2018-01-18:
Does the [`charm.io`](https://github.com/libfive/libfive/blob/master/studio/examples/charm.io) example work for you?

mkeeter, 2018-01-18:
Since I can't reproduce this locally, let me suggest a debugging workflow.

After [this line](https://github.com/libfive/libfive/blob/master/studio/src/view.cpp#L249), add `pick_img.save("/tmp/pick_img.png")`.  Rebuild, open `charm.io`, then open up the resulting image file and look at it.  If the image is being rendered correctly, it should be entirely black, except for a 1-bit blue shadow of the grabbable shape; you will need to adjust brightness / contrast levels in your image viewer to see it:

[image]

If no such file ever appears, then `redrawPicker` isn't being called, which is strange.  If the image is purely black, then either the shape isn't being recognized as having a free variable, or something is going wrong in the FBO loading.  If the image looks correct, then your DPI theory is likely, and I'd recommend adding debug print statements [here](https://github.com/libfive/libfive/blob/master/studio/src/view.cpp#L517) or [here](https://github.com/libfive/libfive/blob/master/studio/src/view.cpp#L448).

jobez, 2018-01-18:
Update: I am using the charm.io example as a sanity check for debugging. I am not seeing the 1-bit blue shadow on inspecting the `pick_img` and adjusting its levels as you specified. Will continue to poke at the dynamic.

simeonveldstra, 2018-02-07:
I can confirm this bug.

commit d2235e32f3108de50f93f4a38703f6e73caf867d

Applying the debugging instructions above, I get a correctly sized all-black png.
redrawPicker is being called.
pick_fbo.isValid() returns true.
drawMonochrome gets called, and calls glDrawElements.
pick_depth.data() is all zeros.

Debian testing (Buster)
Qt version 5.9

For testing, if imagemagick is installed, the dark blue can be brightened up with convert:
`convert /tmp/pick_img.png -auto-level /tmp/auto-pick.png && xli /tmp/auto-pick.png`

simeonveldstra, 2018-02-07:
I've made a small discovery,
in drawMonochrome, both the vao.bind() call and the call to glDrawElements set the error flag to
`GL_INVALID_OPERATION`

At the top of the function, vao.isCreated() returns true.
I'm out of time, but I will investigate this further later.

simeonveldstra, 2018-02-07:
@mkeeter : Please change [line](https://github.com/libfive/libfive/blob/4cac328fa1dfc943e383712bda7cf35cdf0913eb/studio/src/view.cpp#L225) 225 of view.cpp from:
`const bool needs_gl = (context() == QOpenGLContext::currentContext());`
to 
`const bool needs_gl = !(context() == QOpenGLContext::currentContext());`

Thanks,
Simeon

(If you would prefer a pull request, let me know)

mkeeter, 2018-02-08:
Cool, let's hope PR #119 fixes this for everyone.  Feel free to re-open if this occurs again.

closed 2018-02-08 by mkeeter