libfive/libfive #96: Enable single threaded support at compile/runtime?

follower, 2018-01-22:
### Context

I wanted to generate an STL file from a libfive model and the C++/C API using single-threaded code. As far as I could tell, the only way to select this option was to change the hard-coded value for `multithread` from `true` to `false`.

The good news is that it then generated the file successfully but it seems like a less than ideal configuration approach. :)

### Description

From looking at the implementation of `XTree<N>::build()`, I realized that it supported both single- and multi-threaded implementations:

https://github.com/libfive/libfive/blob/668629f6513d5ee6c8af142f882705483dae4618/libfive/src/render/brep/xtree.cpp#L59-L73

However, a call to `libfive_tree_save_mesh()`, like this:

https://github.com/libfive/libfive/blob/56d624f648a2e3c0ff751f045174d2be63e5a7af/libfive/src/libfive.cpp#L273-L279

Uses an implementation of `Mesh::render()` where the use of multi-threading is hard-coded via the value `true`:

https://github.com/libfive/libfive/blob/cda862db31b84bc9b3fe0e18934e02908bf43ae6/libfive/src/render/brep/mesh.cpp#L109-L134

Did you have any plans to expose the option to select single- or multi-threaded either at compilation time or runtime?

(I notice `Heightmap::render()` seems to support multi-threading but as it seems to use a different mechanism I'm not sure what the implications of using single-threading there would be.)

### Implementation options

Seems like single thread support could be either a compile- or run-time option but FWIW I'd lean toward making it a compile time option.

Do you have any particular thoughts on the topic?

mkeeter, 2018-01-22:
Let's go with the simplest, most flexible option: in the commit above, I've added a `bool multithread` argument to `Mesh::render`, defaulting to `true`.

closed 2018-01-22 by mkeeter

follower, 2018-02-05:
Thanks for adding this.

I had been using the C API `libfive_tree_save_mesh()` function but once I switched to using `Kernel::Mesh::render()->saveSTL()` the new way of specifying single-threaded mode worked for me, thanks.

The only downside was that--due to C++'s lack of named arguments--I had to specify/repeat the default values for minimum error etc. There's probably ways around that but it's not really a major issue.

Appreciate the addition!