libfive/libfive #97: Optional "dumb" mouse control of values?

omgitsraven, 2018-01-25:
I'm experimenting with the `repel` transform, and it's (understandably) tripping up the function that tries to change values based on mouse dragging, since it isn't really consistent in direction or anything. Could there be the option of mapping values to just a blind "dragging up vs dragging down" mouse input, to just linearly change the value, for use with values that don't really make sense for the "smart" mouse control?

Obviously the one issue with this is knowing what scale of change to make; my thoughts were either:
- a pixel of movement would correspond to 0.001x of whatever the value was when the drag started (with something hardcoded for zero)
- or, the user could indicate the order of magnitude they want for the dragging, maybe with the number of symbols typed (like `$$500` to drag from 490 to 510, and `%%%1` to drag from 0.999 to 1.001).
- drag horizontally to change the scale at which your vertical dragging is applied (maybe this is a bit too abstract...)
- specify a second value when marking the value, and dragging the length of the screen will have that radius (i.e. taking a value of `10`, and prepending so it says `2#10`, will make dragging the length of the screen change the value between 8 and 12)

This got a lot messier than I pictured it being when I opened this issue, heh :) But hopefully one of these ideas sounds reasonable? I love the 'smooth' style of dialing in a value, to see how its effects change clearly, but it would be great to be able to apply it to more than just things that produce a vaguely-linear result!

mkeeter, 2018-01-25:
From what I've done in the past, logarithmic dragging (your first idea) works pretty well.

I'm limited by the Guile reader in terms of what syntax I can overload (this is why arrays are `#[...]` and free variables are `#...`, because `#` is the only custom syntax object), so I'm not sure whether it's feasible to add another type of syntax to mark "simply draggable free variables".

How about putting this kind of interaction in the script window, e.g. shift + clicking a `#value` would let you do logarithmic dragging?

omgitsraven, 2018-01-25:
That sounds even better, yeah! (If it's going to work that way, would we even need to add the `#`?)

mkeeter, 2018-01-25:
I think we'd still need the `#`, for internal reasons: script evaluation can be expensive, so when you drag a free variable, it doesn't re-evaluate the entire script, just starts re-rendering the shapes with those new values.  Automatically promoting all numbers to variables would work around this, but would kill off a bunch of benefits (like constant folding).