libfive/libfive #109: Studio window is black. No GUI.

doug-moen, 2018-02-04:
### Description

When I start Studio, a window opens, the GUI is briefly visible in the window, then the window turns black. No error messages are printed to the console.

After trying this a few times, I memorized the location of the "File" menu button in the menu bar, before the window turns black. Clicking on this causes a File menu to pop up, and I can select New, but then the window turns black again. So mouse input is working.

### Versions

- Operating system: Ubuntu 16.04 LTS
- `libfive` commit hash: 970c6f70180181edb9a992afdc0dd43b09d4b048
- Qt version: 5.10.0, downloaded from qt.org, since there is no ubuntu package for Qt 5.7 or later on Ubuntu 16.04, and I only run LTS releases.
- GPU and driver: Nvidia GTX 1050, 'nvidia-378' driver (made by nvidia)

It's probably a Qt problem. What version of Qt do you use for your development? I'll try switching versions.

mkeeter, 2018-02-04:
I'm on Qt 5.9.2, but shouldn't be doing anything too exciting.  I've been tweaking a bunch of OpenGL things recently – can you try going back to `dd7cef9f091ca0be51f2cbe78c9d5baa664be92e` and using `git bisect` to identify if there's a bad commit in the recent set of commits?

doug-moen, 2018-02-04:
The bug also exists in commit https://github.com/libfive/libfive/commit/dd7cef9f091ca0be51f2cbe78c9d5baa664be92e

mkeeter, 2018-02-04:
Okay, then it's not a recent bug.  There are certainly folks on Linux that have things working.  Can you try out any of the Qt OpenGL sample apps and see if they work?

doug-moen, 2018-02-05:
I'm still having the problem with Qt 5.9.2.

qtcreator works fine. Ditto Qt Assistant, Qt Designer. These are GUI programs I'm executing directly out of the Qt 5.9.2 tree.

doug-moen, 2018-02-05:
I built one of the gui examples, "elasticnodes", using qmake, and it seems to work fine.

I built the opengl "cube" example, that works too.

mkeeter, 2018-02-05:
Weird!  Are there any debug messages in the console?

doug-moen, 2018-02-05:
No debug messages on the console.

Although the window is black, I can still operate the menu bar. If I execute  the "Load Tutorial" command from the menu bar, then I get a crash.

```
#0  0x00007fe8362722f7 in ?? ()
   from /usr/lib/nvidia-378/libnvidia-glcore.so.378.13
#1  0x0000000000440d1e in QOpenGLFunctions::glDrawElements (indices=0x0, 
    type=5125, count=<optimized out>, mode=4, this=0x7fe83c047960)
    at /usr/local/Qt5.9.2/5.9.2/gcc_64/include/QtGui/qopenglfunctions.h:752
#2  Shape::drawMonochrome (this=0x7fe83c047950, M=..., color=...)
    at /home/doug/res/src/libfive/studio/src/shape.cpp:199
#3  0x00000000004566ff in View::redrawPicker (this=0x101fbd0)
    at /home/doug/res/src/libfive/studio/src/view.cpp:255
#4  0x00007fe865b66b96 in QMetaObject::activate(QObject*, int, int, void**) ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Core.so.5
#5  0x00007fe865b73057 in QTimer::timeout(QTimer::QPrivateSignal) ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Core.so.5
#6  0x00007fe865b73328 in QTimer::timerEvent(QTimerEvent*) ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Core.so.5
#7  0x00007fe865b6777b in QObject::event(QEvent*) ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Core.so.5
#8  0x00007fe8669005ec in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    () from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Widgets.so.5
#9  0x00007fe866907a17 in QApplication::notify(QObject*, QEvent*) ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Widgets.so.5
#10 0x00007fe865b3c2b8 in QCoreApplication::notifyInternal2(QObject*, QEvent*)
    () from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Core.so.5
#11 0x00007fe865b8e08e in QTimerInfoList::activateTimers() ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Core.so.5
#12 0x00007fe865b8e871 in ?? ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Core.so.5
#13 0x00007fe860c19197 in g_main_context_dispatch ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#14 0x00007fe860c193f0 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#15 0x00007fe860c1949c in g_main_context_iteration ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#16 0x00007fe865b8ebaf in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Core.so.5
#17 0x00007fe865b3a8aa in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Core.so.5
#18 0x00007fe865b43134 in QCoreApplication::exec() ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Core.so.5
#19 0x0000000000420a91 in main (argc=1, argv=0x7ffdd40794d8)
    at /home/doug/res/src/libfive/studio/src/main.cpp:46
```

doug-moen, 2018-02-05:
It's possible that the current thread does not have an active OpenGL context.
https://forum.qt.io/topic/69804/opengl-and-qt-crash-at-gldrawelements

How do I diagnose that using gdb, I wonder?

mkeeter, 2018-02-05:
Can you try commenting out `redrawPicker` and see if the background is correctly drawn?

doug-moen, 2018-02-05:
I added a return statement to the top of `View::redrawPicker`. This is much better, the window appears to be drawn correctly now.

doug-moen, 2018-02-05:
Load Tutorial no longer crashes. The source code appears on the left, the graphics (a cube intersected with a sphere) appear on the right.

doug-moen, 2018-02-05:
But clicking in the graphics window of the tutorial now causes a crash.
```
#0  View::mousePressEvent (this=0x1c88ac0, event=0x7ffee1d25110)
    at /home/doug/res/src/libfive/studio/src/view.cpp:484
#1  0x00007f8ad6359291 in QWidget::event(QEvent*) ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Widgets.so.5
#2  0x00007f8ad631c5ec in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    () from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Widgets.so.5
#3  0x00007f8ad63243f3 in QApplication::notify(QObject*, QEvent*) ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Widgets.so.5
#4  0x00007f8ad55582b8 in QCoreApplication::notifyInternal2(QObject*, QEvent*)
    () from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Core.so.5
#5  0x00007f8ad632305f in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Widgets.so.5
#6  0x00007f8ad6371f56 in ?? ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Widgets.so.5
#7  0x00007f8ad63748ab in ?? ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Widgets.so.5
#8  0x00007f8ad631c5ec in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    () from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Widgets.so.5
#9  0x00007f8ad6323a17 in QApplication::notify(QObject*, QEvent*) ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Widgets.so.5
#10 0x00007f8ad55582b8 in QCoreApplication::notifyInternal2(QObject*, QEvent*)
    () from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Core.so.5
#11 0x00007f8ad5b47940 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Gui.so.5
#12 0x00007f8ad5b49735 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Gui.so.5
#13 0x00007f8ad5b24f7b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Gui.so.5
#14 0x00007f8accf0ead0 in ?? ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/plugins/platforms/../../lib/libQt5XcbQpa.so.5
#15 0x00007f8ad0583197 in g_main_context_dispatch ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#16 0x00007f8ad05833f0 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#17 0x00007f8ad058349c in g_main_context_iteration ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#18 0x00007f8ad55aabaf in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Core.so.5
#19 0x00007f8ad55568aa in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Core.so.5
#20 0x00007f8ad555f134 in QCoreApplication::exec() ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Core.so.5
#21 0x00000000004207b1 in main (argc=1, argv=0x7ffee1d25a58)
    at /home/doug/res/src/libfive/studio/src/main.cpp:46
```

mkeeter, 2018-02-05:
Yup, that's to be expected if the pick buffer isn't being drawn.

Can you scatter some debugging statements around `redrawPicker`?  I'm particular, I'm curious whether the FBO is correctly constructed, bound, and released (both of which return `bool`, which I'm not checking right now).  You could also check the return value of `fbo.isValid()` after it is constructed.

doug-moen, 2018-02-05:
pick_fbo is valid after being constructed.
bind and release both return true.

mkeeter, 2018-02-05:
Very strange.  Can you see if there's a subset of `redrawPicker` that make things work if you comment them out?

doug-moen, 2018-02-05:
Adding assertions to redrawPicker prevents Load Tutorial from crashing. This may mean there are references to uninitialized variables. So I ran `valgrind Studio`, and I got a lot of messages about uninitialized data references. Most are in libgc, the garbage collector (could be spurious?). One is in Guile. More interestingly, one of the assertions now fails inside the valgrind environment:
       assert(pick_fbo->bind());
This change in behaviour could be due to relying on the contents of uninitialized variables?

Recompiling with the clang address sanitizer might reveal something interesting.

doug-moen, 2018-02-05:
With the following change, the window does not turn black and Load Tutorial does not crash.
Also, I can now rotate the model in the graphics pane. Superficially, everything seems to work.
```
#if 0
    glClearColor(0, 0, 0, 1);
    glClearDepthf(1);
    glClearStencil(0);
    glClear(GL_COLOR_BUFFER_BIT|GL_DEPTH_BUFFER_BIT|GL_STENCIL_BUFFER_BIT);
    glEnable(GL_DEPTH_TEST);
    glViewport(0, 0, camera.size.width(), camera.size.height());
#endif
```

mkeeter, 2018-02-05:
Oh, this could be the same issue as #92, and you're running a version that predates those fixes.  Can you update all the way to `master` (41eb8c59c0959694094d0829393b81090839ce37) and see if that works (or 08de9db4bec30021cfd4653910b36251657a026a, which is specifically to fix #92).

doug-moen, 2018-02-05:
fixed in master.

closed 2018-02-05 by doug-moen