libfive/libfive #111: Non-integer values with expt?

omgitsraven, 2018-02-04:
I forget if this has already been discussed somewhere else, but is it a bug or an inherent limitation that you can't use either decimals or fractions as the power in an `expt` call inside a `lambda-shape` call?

Decimals (i.e. `8.1`) give
```
goops-error: No applicable method for #<<generic> expt (1)> in call (expt #<<shape> 11ab02e80> 8.1)
```

and fractions (i.e. `(/ 8)`) give
```
unbound-variable: Unbound variable: make-tree
```

mkeeter, 2018-02-04:
A little of both!

It's an inherent limitation (in Boost::Interval) that you can't use decimals, but now the error message is better:
```
wrong-type-arg: RHS of exponentiation must be rational, not 8.1

Stack trace:
0: (lambda-shape (x y z) (expt x 8.1))
```

It was a bug that you couldn't use fractions, and that's now fixed!

closed 2018-02-04 by mkeeter

omgitsraven, 2018-02-04:
Oh, fantastic! I was almost positive it used to work with fractions, but I just couldn't remember...

So, forgive me if this is a stupid question since I'm not sure exactly how much power macros have, but, is there not some way to take a value like `8.2` and translate it to `(+ 8 (/ 2 10))`?

mkeeter, 2018-02-04:
You probably don't need a macro for this; try something like
```
scheme@(guile-user)> (define (tenths a) (round (* 10 (- a (floor a)))))
scheme@(guile-user)> (tenths 8.1)
$10 = 1.0
```
to extract the tenths digit, then build up your expression from there.

As an aside, Guile will happily tell you the fraction representation for numbers like 8.1:
```
scheme@(guile-user)> (numerator 8.1)
$12 = 4559894622712627.0
scheme@(guile-user)> (denominator 8.1)
$13 = 562949953421312.0
```
The reason I don't rely on this is because even though `x^8.1` looks reasonable, breaking it into`(x^4559894622712627)^(1/562949953421312)` causes things to blow up in Boost::Interval.

omgitsraven, 2018-02-04:
I'm not sure that works; even just doing `(lambda shape (x y z) (expt x (round 8.1)))` gives "wrong type argument in position 1" for some reason.

mkeeter, 2018-02-04:
Ah, that's because `round` still returns an inexact.

Here's a better version of `tenths`:
```
scheme@(guile-user)> (define (tenths a) (/ (inexact->exact (round (* 10 (- a (floor a))))) 10))
scheme@(guile-user)> (tenths 8.1)
$3 = 1/10
scheme@(guile-user)> (tenths 8.3)
$4 = 3/10
scheme@(guile-user)> (tenths 8.4)
$5 = 2/5
```

omgitsraven, 2018-02-05:
I get "unbound variable" when I try to use `inexact->exact`...

mkeeter, 2018-02-05:
Oh, it's not in the list of safe functions exposed to the sandbox.

I've added it now; hopefully, there's not an clever sandbox escape that relies on it 😛

omgitsraven, 2018-02-05:
...hahah, I just realised I was thinking about this too literally and I don't need to separate out the ones place like that at all. I just got too deep into how I was thinking about the problem and it didn't come to me until five minutes after closing my laptop.

mkeeter, 2018-02-05:
Glad you got things working!

I don't think that special-casing decimal exponents is worth the confusing break in behavior – I could see someone asking "why does 8.1 work but 8.25 not work?", vs always seeing a sensible error message when you pass in decimals.