libfive/libfive #112: Crash occurs if `min_feature` value is 0

follower, 2018-02-05:
A crash occurs if the `min_feature` argument of `Kernel::Mesh::render()` is 0.

Why would anyone ever specify such a value? Well, maybe they weren't paying attention and used integer division when specifying a `min_feature` value of `1/15`. Maybe. :)

AFAICT the issue is due to this test:

https://github.com/libfive/libfive/blob/366880317e4405d9f757a66fb582cc363ea18f6a/libfive/src/render/brep/xtree.cpp#L139

So, I think an initial check to ensure that `min_feature` is not equal to 0 might be required?

Although, the crash possibly also indicates that some other code isn't checking the return value of an memory allocation request?

mkeeter, 2018-02-05:
Alright, that should fix it in most cases.

This is a probably a good old-fashioned stack overflow, so it's may be possible to detect in the `XTree` constructor, but the sanity-checking is good enough for now.

closed 2018-02-05 by mkeeter

follower, 2018-02-06:
There's a part of me that wonders about just using the default value if an invalid value is supplied but I guess it's probably better to explicitly return a `nullptr` given there's other scenarios where `nullptr` is returned that also need to be handled.

(Plus that hard won modern logic of "be conservative in what you accept because otherwise you'll probably get a bunch of unintended consequences..." :D )

Thanks!