libfive/libfive #113: Recently added OpenGL version check prevents Studio running

follower, 2018-02-05:
Apparently it's been a while since I compiled Studio because now it refuses to run and shows this error message: 

```
Error:

OpenGL context is too old
(got 3.2, need 3.3)

The application will now exit.
```

The associated OpenGL version check was added in cda862db31b84bc9b3fe0e18934e02908bf43ae6:

https://github.com/libfive/libfive/blob/cda862db31b84bc9b3fe0e18934e02908bf43ae6/studio/src/view.cpp#L274-L296

If I edit out the version check the application still appears to run successfully.

This is on Mac OS X 10.8.5 with Qt 5.6.2 which I realise is not officially supported but thought I'd at least mention it. :)

follower, 2018-02-05:
Ah, I see that this is because version 3.3 is being selected as the default format here:

https://github.com/libfive/libfive/blob/master/studio/src/main.cpp#L30-L35

I guess the fact that it still works with a 3.2 context is because the code currently doesn't use any of the [new OpenGL 3.3 features](https://www.khronos.org/opengl/wiki/History_of_OpenGL#OpenGL_3.3_.282010.29)?

Would you have an objection to changing the requested version to 3.2 rather than 3.3?

mkeeter, 2018-02-05:
Go for it!  Remember to change the shader versions from `#version 330` to `#version 150` as well (I usually target OpenGL 3.3 because it's the first version that has matching GL and GLSL version numbers 😛 ).

follower, 2018-02-06:
Cool, I have a patch in the PR above.

Your logic for choosing 3.3 seems pretty sensible. :)

Of course, as mentioned in the PR, it turns out that the "real world" wants to complicate things and apparently Mac OS X 10.8 supports OpenGL 3.2 (and presumably GLSL 1.5) but *also* supports some large part of GLSL 3.3. And the existing libfive use `location()` which *isn't* supported under 1.5.

So, I just went ahead and changed the OpenGL version requested but didn't change the GLSL version. Hopefully this doesn't cause issues for other systems...

(Oh, and, I discovered--when I remembered I could get logs by running the Studio binary directly--that at least some of Qt's standard shaders don't compile by default anyway so spew a bunch of error messages...)

If you'd rather not have the version mismatch I can try to come up with a different approach...

mkeeter, 2018-02-08:
Fixed in PR #116

closed 2018-02-08 by mkeeter