libfive/libfive #117: Proof of concept port of libfive to WebAssembly

follower, 2018-02-06:
A client of mine (@Omniblox) asked me to investigate porting libfive to WebAssembly (via Emscripten) in order to run it in the browser.

This work resulted in a proof-of-concept initial port of the C/C++ layer and they were open to me making the result public here: <https://github.com/Omniblox/libfive/tree/add-web-assembly-support>

There are a couple of demos viewable here: <https://omniblox.github.io/libfive/>

I've updated the README with build instructions (and how to interact with the second demo that displays a libfive-generated STL file via three.js): <https://github.com/Omniblox/libfive/blob/add-web-assembly-support/README.md>

### Current state

The primary "hacky" thing about the proof of concept is that I haven't yet actually exposed libfive functionality via bindings to JavaScript--the second demo actually runs a (Emscripten-compiled) standalone C++ program that generates a .stl file onto a "virtual filesystem" the content of which is then supplied to three.js via JavaScript.

### CMake configuration

Most of the effort required was in configuring CMake to build things correctly and the libfive source itself compiled pretty much "as is" (with the previous exception of single thread support). 

I tried to isolate the CMake configuration changes as much as possible and implement them in a way that might be useful in other contexts (e.g. edge-cases for other platforms). (Although I've since realised there's potentially a simpler way to handle the dependency configuration in some cases.)

Most of the configuration changes have some explanatory documentation as to why the change is required for use with Emscripten/WebAssembly: <https://github.com/Omniblox/libfive/blob/add-web-assembly-support/CMakeLists.txt>

### Test suite

I discovered that at present the test suite will compile but not run--which I suspect might have something to do with the test framework itself. I saw some slightly odd behaviour with really simple text output that seemed to generate similar errors. I didn't investigate this further as wanted to focus on a more visual proof-of-concept. :D

### Outro...

Anyway, I thought you might be interested to check things out and I'd be interested in any thoughts/feedback you might have.

mkeeter, 2018-02-08:
That's incredibly cool!  I'm going to leave this issue open, so feel free to post progress; we can eventually consider merging into `master` if that ends up making sense.

I'd be particularly interested in seeing if you can get the test suite running to profile rendering speed against the native library.

tir-kaval, 2018-02-08:
This is an interesting idea!

Another idea I had related to the web browser would be to adapt [Shader Toy](https://en.wikipedia.org/wiki/Shadertoy) to use [Varjo](https://github.com/cbaggers/varjo) to render libfive models.  Varjo could be used stand-alone as well.  No proof here, just the concept.

tir-kaval, 2018-02-09:
To clarify, is there a reason it wouldn't be feasible to render a libfive tree as a pixel shader on the GPU rather than to triangles, as is done in this port to WA?  Seems like it would be faster if so, and could work in the browser or on the desktop for viewing.  See:

http://www.iquilezles.org/www/articles/distfunctions/distfunctions.htm

tir-kaval, 2018-02-11:
[Edited for clarity]

This is an interesting approach to bringing libfive to the browser.  My concern would be the performance--even for just exporting meshes--and a speed comparison against native would be useful, as Matt said.

An alternative I thought of would be to use GLSL/WebGL to implement a separate viewer for libfive models.  There is a project using pixel shaders to render in the browser:

[Shader Toy](https://www.shadertoy.com/)

and signed distance fields can be rendered this way:

http://www.iquilezles.org/www/articles/distfunctions/distfunctions.htm

This would obviously be a project in itself, but would be another potential way to have libfive in the browser.

mkeeter, 2018-02-11:
There's a ton of great work coming out of the demoscene on distance fields!  Doug Moen is doing a great job documenting the field and building a raytrace-based modeling system named [Curv](https://github.com/doug-moen/curv).

I've been paranoid about ray/sphere tracing because I want the infrastructure-level libraries to work no matter what the shape of the underlying distance fields – even if they're non-Euclidean, non-Lipshitz-continuous, etc, I'd like to be using algorithms that remain robust.

follower, 2018-03-04:
FYI for anyone reading this issue at a later date: there were some posts related to rendering signed distance fields in GLSL which appear to have been deleted by their author.

The links mentioned included:

 * <https://github.com/cbaggers/varjo> -- "Lisp to GLSL Language Translator"

 * <http://www.iquilezles.org/www/articles/distfunctions/distfunctions.htm>

 * ShaderToy

follower, 2018-03-04:
With regard to performance: yeah, the initial proof-of-concept has had no specific attention applied to performance.

It would probably be worth investigating where the best boundaries are between WebAssembly / JavaScript / WebGL in terms of performance with libfive & whatever is used to render the output.

omgitsraven, 2019-12-27:
I know I'm bumping a real old thread, but out of curiosity, what's the state of this? (I guess at this point it's way too out of date to be merged successfully?) I was just starting to think about trying something like this myself, when I remembered that there was already this earlier attempt... (For the record, I haven't touched WebAssembly at all myself yet, but I was considering this as a possible place to get my feet wet with it...)

diclophis, 2020-01-08:
@omgitsraven did you have any luck reviving the fork or getting it updated with a master branch merge?

omgitsraven, 2020-01-13:
I didn't get around to trying in the end, sorry.

follower, 2020-03-03:
Thanks for your interest in this, from the "better late than never" file :) here's a response:

### Current state

As it was 2 years ago--no additional development since then.

### Revival potential

> "Most of the effort required was in configuring CMake to build things correctly and the libfive source itself compiled pretty much "as is" (with the previous exception of single thread support)."

Due to this, the changes for WebAssembly *may* be relatively independent of subsequent source changes in libfive (outside of dependencies, project file structure changes etc).

See here for the changes I made at the time: https://github.com/libfive/libfive/compare/master...Omniblox:add-web-assembly-support#files_bucket

smilberger, 2020-12-16:
For inspiration about what could be done with Libfive in the browser, the author of the Unity binding for Libfive has done a scripted browser CAD system using OpenCascade compiled to JS/WebAssembly:

https://github.com/zalo/CascadeStudio
https://github.com/donalffons/opencascade.js#FAQ

I was thinking if they could do this with that giant C++ OpenCascade library, why not Libfive?  There are a number of Lisp/Scheme implementations that run in JS, I'm not sure about Guile.

P.S. I'm not volunteering for this task, but I have been experimenting with a Blender binding and came across this work.

bkrmendy, 2023-12-09:
I tried reviving the fork linked in the issue description, however build fails for me with this message (building on an M1 Mac with emsdk version 3.1.48):
```
[  1%] Generating version.c, _version.c
[  3%] Building CXX object libfive/src/CMakeFiles/five.dir/eval/base.cpp.o
In file included from /Users/***/Developer/libfive/libfive/src/eval/base.cpp:21:
In file included from /Users/***/Developer/libfive/libfive/src/../include/libfive/eval/base.hpp:20:
In file included from /Users/***/Developer/libfive/libfive/src/../include/libfive/eval/tape.hpp:27:
In file included from /Users/***/Developer/libfive/libfive/src/../include/libfive/eval/interval.hpp:21:
In file included from /Users/***/Developer/emsdk/upstream/emscripten/cache/sysroot/include/boost/numeric/interval.hpp:18:
/Users/***/Developer/emsdk/upstream/emscripten/cache/sysroot/include/boost/numeric/interval/hw_rounding.hpp:43:4: error: Boost.Numeric.Interval: Please specify rounding control mechanism.
   43 | #  error Boost.Numeric.Interval: Please specify rounding control mechanism.
      |    ^
1 error generated.
em++: error: '/Users/***/Developer/emsdk/upstream/bin/clang++ -target wasm32-unknown-emscripten -fignore-exceptions -fvisibility=default -mllvm -combiner-global-alias-analysis=false -mllvm -enable-emscripten-sjlj -mllvm -disable-lsr -DEMSCRIPTEN --sysroot=/Users/***/Developer/emsdk/upstream/emscripten/cache/sysroot -Xclang -iwithsysroot/include/fakesdl -Xclang -iwithsysroot/include/compat -I/Users/***/Developer/libfive/libfive/src/../include -isystem /opt/homebrew/Cellar/eigen/3.4.0_1/include/eigen3 -Wall -Wextra -g3 -fPIC -pedantic -Werror=switch -D__USE_ISOC99 -O3 -DRELEASE -DEIGEN_NO_DEBUG -std=gnu++11 -MD -MT libfive/src/CMakeFiles/five.dir/eval/base.cpp.o -MF CMakeFiles/five.dir/eval/base.cpp.o.d -c /Users/***/Developer/libfive/libfive/src/eval/base.cpp -o CMakeFiles/five.dir/eval/base.cpp.o' failed (returned 1)
make[2]: *** [libfive/src/CMakeFiles/five.dir/eval/base.cpp.o] Error 1
make[1]: *** [libfive/src/CMakeFiles/five.dir/all] Error 2
make: *** [all] Error 2
emmake: error: 'make' failed (returned 2)
```

I haven't investigated this in depth yet, maybe a rookie mistake. Any pointers would be appreciated

bkrmendy, 2023-12-09:
I worked around the problem by reverting to emsdk version `2.0.23-lto` and updating
```
add_compile_options(-s USE_LIBPNG=1)
```
to
```
add_compile_options("SHELL:-s USE_LIBPNG=1" "SHELL:-s USE_BOOST_HEADERS=1")
```

Ammarkhat, 2023-12-25:
@follower I tested this branch on my Mac mini (macOS Sonoma 14.1). For some reason, I am getting very large .wasm files (~72mb) for the added examples compared to ~400kb in your published github page. Any idea what I might be doing wrong?

@bkrmendy since you tried this recently, are you getting large wasm files?

Ammarkhat, 2023-12-27:
FYI, I think emscripten just includes debug symbols in the resulting WASM file. Removing the -g flag removes the extra debug symbols and returns the size to ~400kb. Not sure how it was running before.