libfive/libfive #124: libfive-test does not run correctly on Windows

omgitsraven, 2018-02-12:
I'm trying out the Windows support, and the compilation isn't giving me any errors, but when I actually launch libfive-test.exe, it tells me "five.dll was not found" -- and when I go and get five.dll out of build\libfive\src\Debug and copy it to libfive-test.exe's directory, it still fails in several different ways:
[image]

I'm sorry to show up with a question as vague as "does anyone have any suggestions", but I really know virtually nothing about using Visual Studio, so I'm not sure if there's anything in these errors that suggests I've followed some of the instructions incorrectly... It isn't the end of the world if I can't run this on Windows, since I can still run it on my Mac, but I thought I'd bring it up in case there's some obvious solution, or in case this is a useful test case for people wanting to solidify the Windows instructions.

speps, 2018-02-12:
I have a fix for the missing DLL. I still need to investigate the error message. The one in the screenshot I have a fix for but other ones after it are std::async related it seems.

From: RavenWorks
Sent: 12 February 2018 20:14
To: libfive/libfive
Cc: Subscribed
Subject: [libfive/libfive] libfive-test does not run correctly on Windows(#124)

I'm trying out the Windows support, and the compilation isn't giving me any errors, but when I actually launch libfive-test.exe, it tells me "five.dll was not found" -- and when I go and get five.dll out of build\libfive\src\Debug and copy it to libfive-test.exe's directory, it still fails in several different ways:

I'm sorry to show up with a question as vague as "does anyone have any suggestions", but I really know virtually nothing about using Visual Studio, so I'm not sure if there's anything in these errors that suggests I've followed some of the instructions incorrectly... It isn't the end of the world if I can't run this on Windows, since I can still run it on my Mac, but I thought I'd bring it up in case there's some obvious solution, or in case this is a useful test case for people wanting to solidify the Windows instructions.

speps, 2018-02-14:
The DLL fix is now in 7cbd6bc. Looking at the other issue (the REQUIRE error).