libfive/libfive #126: Possible copy/paste error

speps, 2018-02-13:
https://github.com/libfive/libfive/blob/9f56b68e94a10bf8f781ee9e012e45f3afbff305/libfive/src/render/brep/xtree.cpp#L163

It seems on that line it should be `eval` and not `eval + i`. Am I reading this wrong? Adding an int to the pointer will just point to invalid memory.

mkeeter, 2018-02-13:
Nope, this is intentional!

When doing multithreaded evaluation, `eval` is an array of evaluators (constructed [here](https://github.com/libfive/libfive/blob/9f56b68e94a10bf8f781ee9e012e45f3afbff305/libfive/src/render/brep/xtree.cpp#L62-L66)); this lets us assign each thread a separate `XTreeEvaluator`.

This is (curtly) documented [here](https://github.com/libfive/libfive/blob/9f56b68e94a10bf8f781ee9e012e45f3afbff305/libfive/include/libfive/render/brep/xtree.hpp#L60).

closed 2018-02-13 by mkeeter

speps, 2018-02-14:
Thanks, I didn't see that when I looked :)

speps, 2018-02-14:
I was only looking at this because it seems in VS2017 `std::async` has issues in the part I highlighted. The result is different is multithread or not and the result is correct when I replace `std::async` with `std::thread`.

mkeeter, 2018-02-14:
Interesting!  I was using `async` to get `futures` out, but could probably just use threads and `join` them manually, since I'm not taking advantage of the `std::future` API (just calling `get` below).

On the other hand, other folks on MSVC (e.g. @igfraga) haven't reported this issue to me, so I'm not sure what's different in your setup..